JMRI/JMRI #4930: Help with a jmri.jar please.

Splitpin258, 2018-02-22:
I'm dipping my toes into JMRI development for the first time and need some help getting my tinkerings testing in a working environment.
I'm all setup with Eclipse Oxygen, github etc. I have 2 computers. My development machine not connected to the layout is Win10 and I can now build/run jmri without error in Eclipse and with Run/Run Configurations I can run PanelPro and see my code changes present and working. So thats all good to that point, now the rub.
My 'train computer' is Win7. I 'think I understand' what I have to do next is compile the project using 'ant dist' into a single jmri.jar file and I'm doing that apparently. The file is produced with 2 warnings but no error. I think I then just have to copy that new jmri.jar into the 'train computer's' \JMRI folder (after backing up the original of course) and all should be well. But no, after a number of attempts and some wailing and gnashing of teeth I don't seem to be getting my changes to the working machine.
What am I missing ? Is jmri.jar not the whole story ? Or is Eclipse (etc) doing some versioning I don't understand and faithfully recompiling the git/downloaded master without my changes ?

TIA
JohnOD

bobjacobsen, 2018-02-23:
You could try

ant package-windows

which should create an installer (you’ll need NSIS on that machine) which you can copy over & run. 

The compilation steps just make the jar file, but there’s a bunch of other files have to have consistent contents. 

Bob

Splitpin258, 2018-02-23:
Thanks Bob.

closed 2018-02-23 by Splitpin258