JMRI/JMRI #4931: 4.11.3 cannot change state of Sensors from Sensor Table

nigelcliffe, 2018-02-22:
4.11.3, on both Windows and Raspberry PI.   Cannot change state of a sensor in the sensor table by clicking the button in the table.    Applies to both System (LocoNet in my case) sensors and Internal sensors.

Not sure when this came in as a problem.

bobjacobsen, 2018-02-22:
I can recreate this, will track it down.

bobjacobsen, 2018-02-22:
I was able to recreate it with a LocoNet simulator connection.  Git bisect between v4.11.2 and v4.11.3 tracked it down to commit 21afbd01dec33b387ae86ddad5e804cf770450a1 where @rhwood was fixing the column ID problem with saving table preferences.

At this point, I'd like to pass this one off to somebody else.  The commit has too many of those obnoxious reformatting changes than I'd like to spend time working through right now.

commit b4f248c0 (2018-02-23): Merge pull request #4933 from rhwood/hide-table-columns

Fix bugs in tables introduced in 4.11.3

Fix #4927
Fix #4931
closed 2018-02-23 by rhwood