JMRI/JMRI #4935: Should power managers extend AbstractPowerManager?

SteveRawlinson, 2018-02-23:
Currently only LnPowerManager extends AbstractPowerManager the rest implement PowerManager directly.

There is a fair amount of duplicated code and if you want to add a feature to power managers generally you'd have to write it in a lot of places.

If the consensus is that they should I'd be happy to do the refactoring (or indeed happy not to).

rhwood, 2018-02-23:
No they "should" not; however, there is likely no reason not to make (most) of the them extend AbstractPowerManager.

Note that nothing should be in AbstractPowerManager that is not required to implement a method defined in PowerManager or other API implemented by AbstractPowerManager.

bobjacobsen, 2018-02-23:
> Note that nothing should be in AbstractPowerManager that is not required to implement a method defined in PowerManager or other API implemented by AbstractPowerManager.

To amplify on this:

Sometimes interfaces are implemented by an AbstractFoo class, and sometimes by a DefaultFoo class (which might in turn be extending AbstractFoo).

The convention is that AbstractFoo is really about _just_ the visible parts of Foo; it’s still “Abstract” because it’s leaving things to subclasses.  DefaultFoo, on the other hand, includes a number of implementation choices so that you can use it without further subtyping, although people often do subtype them.

Consider the javax.swing.CellEditor interface. There’s an abstract AbstractCellEditor class:

"A base class for CellEditors, providing default implementations for the methods in the CellEditor interface except getCellEditorValue(). Like the other abstract implementations in Swing, also manages a list of listeners.”

There’s also a default implementation javax.swing.DefaultCellEditor which extends AbstractCellEditor to become "The default editor for table and tree cells”.  In the process, it’s decided that it’ll implement getCellEditorValus by referencing another JComponent that was provided in a constructor.  But that’s just an implementation choice, you could do it differently either by implementing your own subclass of AbstractCellEditor or DefaultCellEditor

closed 2018-02-27 by SteveRawlinson