JMRI/JMRI #4937: Problems getting started with JMRI

innovationgame, 2018-02-23:
I have installed JMRI and I wanted to make a start so that when I get a laptop, I will be able to transfer the files and get things moving.  I have been able to enter the layout editor and create a schematic of my layout, including all the loops and turnouts.  However, when I try to access any table, all that happens is that a hidden blank window is created.  The editor and PanelPro still work at this stage.  However, if I open the window via the PanelPro Window menu item, everything stops working.  I cannot even close PanelPro.  I have to resort to Taskmanager to end the Java Process. I went to the JMRI discussion forum on SouorceForge.net, but I need to register to use that.  I have tried to register with SourceForge, but I get no reply, so I am stuck.  Does anyone have any ideas?

KenC57, 2018-02-23:
This is the audio driver hang. It happens to some systems based on lower
level interaction with the audio subsystem on some computers. It seems to be
triggered by different changes in the Windows OS from time to time. We have
a workaround that is part of 4.11.1 and newer releases. We do have a patch
for older installs. The simpler way is updating.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

KenC57, 2018-02-23:
The patch is in #4599 if you want to do that.

innovationgame, 2018-02-23:
Thanks Ken.  I installed 4.11.3..... and it seems to hve done the trick.   :)

closed 2018-02-23 by innovationgame