JMRI/JMRI #4946: A tweak for Roster Speed Profiling

Splitpin258, 2018-02-24:
The current behavior of the speed profiling tool is such that for each pass, as the loco starts to exit the timed block it halves it's speed until it completes that exit at which point it stops goes through a stop, pause, reverse direction and go again cycle.
The problem I ran into was that 'the half current speed while leaving the block' simply stops on the spot any loco in the lower end of it's speed range.
Yes I learnt a little hand of God at the end of each slow pass to help the loco exit the block got me a good profile from bottom to top, but that meant baby sitting the process for an hour or more in some cases. The other alternative is start the profile at (minimum run throttle setting * 2) which misses profiling the low end of what's possible for this loco. Worse really, is the 'new user of the tool' experience. Yes me recently. The interface is clear enough, I entered a 'minimum reliable slow crawl speed for this loco' as the Profile Start, 126 for End, 2 for Increment all seemed reasonable, and Start.
Loco slow crawls to the end of the block and stops..... ok what happens now. Nothing - hmmm. Yes I worked it out in time, worked around it but I'm patient, persistent and know just enough about programming to be dangerous. How many users just say it's broken and move on ? And it is kinda, it's a pain to profile the lower end of a loco's speed range. 
So I'm motivated to fix this, and have for myself. The change I made was; create a variable to store a 'profileSpeedAtStart' then simply test if (profileSpeed/2) > profileSpeedAtStart, if so, good, current behavior, half speed. If not, yes we can still reduce speed but only to profileSpeedAtStart as a minimum.
The crux of the issue is in jmri.jmrit.roster.swing.speedprofile.speedprofilepanel.stopCurrentSpeedStep() which is fired by the Finish Sensor going active at the end of each lap.
Currently the test for 'should I halve speed ?' in this method is: if (profileStep >= 4) which is kind of protecting the lower end of the speed range but is pretty arbitrary, has a lot more meaning for 14/28 steps than 128 and ... causes locos to stop dead if myspeed/2 < as slow as I can go.
So what do you think ? Have I fundamentally missed something in speed profiling and/or is it worth fixing ? I'm also interested in learning my way though the whole pushme-pullyou git process which might take another little while.

JohnOD

bobjacobsen, 2018-03-04:
I think PR #4970, now merged for JMRI 4.11.4, fixes this.

Splitpin258, 2018-03-04:
Yes Bob thanks, both this and PR #4970 are mine. I should have closed this before.

closed 2018-03-04 by Splitpin258