JMRI/JMRI #4951: BadSystemNameException while creating an Entry/Exit Pick List.

dsand47, 2018-02-25:
The system name for the DestinationPoints bean is a random UUID which can start with a numeric character.

```
2018-02-24 22:47:28,835 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
jmri.NamedBean$BadSystemNameException
	at jmri.Manager.getSystemPrefixLength(Manager.java:323)
	at jmri.NamedBean.compareTo(NamedBean.java:370)
	at jmri.util.NamedBeanComparator.compare(NamedBeanComparator.java:30)
	at jmri.util.NamedBeanComparator.compare(NamedBeanComparator.java:23)
	at java.util.TreeMap.put(TreeMap.java:552)
	at java.util.TreeSet.add(TreeSet.java:255)
	at jmri.jmrit.picker.PickListModel.makePickList(PickListModel.java:137)
	at jmri.jmrit.picker.PickListModel.updateNameList(PickListModel.java:122)
	at jmri.jmrit.beantable.BeanTableDataModel.<init>(BeanTableDataModel.java:86)
	at jmri.jmrit.picker.PickListModel.<init>(PickListModel.java:77)
	at jmri.jmrit.picker.PickListModel$EntryExitPickModel.<init>(PickListModel.java:890)
	at jmri.jmrit.picker.PickListModel.entryExitPickModelInstance(PickListModel.java:497)
	at jmri.jmrit.conditional.ConditionalEditBase.createSinglePanelPickList(ConditionalEditBase.java:389)
	at jmri.jmrit.conditional.ConditionalListEdit.variableItemChanged(ConditionalListEdit.java:1998)
	at jmri.jmrit.conditional.ConditionalListEdit.access$200(ConditionalListEdit.java:86)
	at jmri.jmrit.conditional.ConditionalListEdit$24.actionPerformed(ConditionalListEdit.java:1829)
	at javax.swing.JComboBox.fireActionEvent(JComboBox.java:1258)
	at javax.swing.JComboBox.setSelectedItem(JComboBox.java:586)
	... snip ...

```

This affects both the traditional tabbed pick list and the single panel pick list.  In the meantime I can use the Logix/Conditional combo box option.

I am running 4.11.4ish+.  

Dave Sand

bobjacobsen, 2018-02-25:
Could you give a recipe for recreating this?  I'm not really familiar with Entry/Exit

dsand47, 2018-02-25:
Load the attached file in 4.11.3.

1. Select **Tools >> Tables >> Logix**.
2. Verify that **Use Traditional Pick Lists** is selected in **Options**.
3. Edit Logix **IX101**
4. Edit Conditional **IX101C1**
5. Select **Add State Variable**

The UUID is generally hidden.  This is a section of a panel XML file:
```
      <source type="sensor" item="NX-BE">
        <destination type="sensor" item="NX-BW-Side" nxType="fullinterlocking" uniqueid="bf95efc9-d5ff-416a-8a98-a8e6e677951f" />
        <destination type="sensor" item="NX-BW-Main" nxType="fullinterlocking" uniqueid="48c8a342-dbb9-43f6-944d-a620f2ba4bcf" />
      </source>
```

Dave Sand

[NX-Test-TO.xml.txt](https://github.com/JMRI/JMRI/files/1755758/NX-Test-TO.xml.txt)

dsand47, 2018-02-25:
You also need the two attached icons at preference:resources/icons.

[image]
[image]

dsand47, 2018-02-25:
Here is my temporary work around:

```
--- a/java/src/jmri/Manager.java
+++ b/java/src/jmri/Manager.java
@@ -6,6 +6,7 @@ import java.util.Arrays;
 import java.util.LinkedList;
 import java.util.List;
 import java.util.TreeSet;
+import java.util.regex.Pattern;
 import javax.annotation.CheckForNull;
 import javax.annotation.CheckReturnValue;
 import javax.annotation.Nonnull;
@@ -320,7 +321,11 @@ public interface Manager<E extends NamedBean> {
             throw new NamedBean.BadSystemNameException();
         }
         if (!Character.isLetter(inputName.charAt(0))) {
-            throw new NamedBean.BadSystemNameException();
+            if (!Pattern.matches(
+                    "^[0-9a-f]{8}-[0-9a-f]{4}-[0-9a-f]{4}-[0-9a-f]{4}-[0-9a-f]{12}$",   // NOI18N
+                    inputName.toLowerCase())) {
+                throw new NamedBean.BadSystemNameException();
+            }
         }
 
         // As a very special case, check for legacy prefixs - to be removed
```

Dave Sand

bobjacobsen, 2018-02-25:
I think the fundamental question here is whether a `jmri.jmrit.entryexit.DestinationPoints` should be a NamedBean at all.  

Clearly it's convenient, sort of, because it sort-of gets you manager-like code and pick-list like code almost automatically.  Except DestinationPoints really aren't _Named_Beans, they don't have all of the behaviors:  Their system names really aren't system names for instance.  

So we're likely to keep hitting more and more special cases for them.

Right now, we should probably put a bypass in place so that this continues to work.  But I think we should think pretty hard before putting more work into better bypasses, etc, without also really making these NamedBeans with all that implies. Or would it be better to just not have DestinationPoints as a NamedBean?

dsand47, 2018-02-26:
For the immediate problem, I see a number of possibilities:

- Enable UUID system names.
- Remove the Entry/Exit **Destination Points** Pick List table.
- Prefix the UUID with something like **NX**.

Since I have only been doing Java programming for a year, I don't know the expected NamedBean behaviors or what might be missing.

Dave Sand

closed 2018-04-06 by dsand47