JMRI/JMRI #4955: SPROG light-it

AntonMezger, 2018-02-25:
hi,

using the latest version of jrmi/DecoderPro 4.11.3 I succeeded to set a correct loco address as described by Dave Heap. However when I try to do the same procedure for signal mode then DecoderPro does not do anything and blocks. What can be wrong there?

best regards

Anton Mezger (a-ch.mezger@bluewin.ch)

dheap, 2018-02-25:
@AntonMezger What SPROG version are you using?

I have a SPROG IIv3 so I'll give it a try.

AntonMezger, 2018-02-25:
I use a SPROG IIv4, jmri 4.11.3 on windows 7. The interesting thing is that when I program the address in loco mode, prodecoder writes in the window below, the cv's that are being written. However in signal mode nothing is writen except idl and the program has to be killed while it does not respond any more

Thanks for taking a look at it.

dheap, 2018-02-25:
Confirmed this with my SPROG IIv3 and both V4.11.3 and current Head of Master on a Mac Yosemite. Any attempt to issue Signal aspect or programming packets causes DecoderPro to hang irrecoverably and has to be killed. Also with Master, very slow response to keyboard and mouse actions.

We'll see what Andrew Crosland has to say...

dheap, 2018-02-25:
@AntonMezger  By there way, it's not a Light-It issue but a SPROG issue so I've changed the title of this issue.

dheap, 2018-02-25:
@crosland Watching with `jconsole` shows number of threads drops a little but Heap Memory Usage is not happy:
[image]

bobjacobsen, 2018-02-25:
Does the "Detect Deadlock" button on the Threads pane come up with anything?

Also, the "jvisualvm" tool can give you a more detailed view of the JVM. Its Threads tab is particularly good when the "Threads visualization" box is checked.

dheap, 2018-02-26:
> Does the "Detect Deadlock" button on the Threads pane come up with anything?

"No deadlock detected"

> Also, the "jvisualvm" tool can give you a more detailed view of the JVM. Its Threads tab is particularly good when the "Threads visualization" box is checked.

[image]
(the NCE Simulator connection  is running so I had access to the Packet Analyser)
Hang looks like this:
[image]
```
Log is streaming:
     [java] 2018-02-26 11:55:46,970 sprog.SprogCommandStation             WARN  - Slot thread wait timeout [SPROG slot thread]
     [java] 2018-02-26 11:55:47,002 sprog.SprogCommandStation             WARN  - Timeout Waiting for reply [AWT-EventQueue-0]
     [java] 2018-02-26 11:55:47,103 sprog.SprogCommandStation             WARN  - Timeout Waiting for reply [AWT-EventQueue-0]
     [java] 2018-02-26 11:55:47,206 sprog.SprogCommandStation             WARN  - Timeout Waiting for reply [AWT-EventQueue-0]
     [java] 2018-02-26 11:55:47,309 sprog.SprogCommandStation             WARN  - Timeout Waiting for reply [AWT-EventQueue-0]
     [java] 2018-02-26 11:55:47,415 sprog.SprogCommandStation             WARN  - Timeout Waiting for reply [AWT-EventQueue-0]
     [java] 2018-02-26 11:55:47,519 sprog.SprogCommandStation             WARN  - Timeout Waiting for reply [AWT-EventQueue-0]
     [java] 2018-02-26 11:55:47,624 sprog.SprogCommandStation             WARN  - Timeout Waiting for reply [AWT-EventQueue-0]
     [java] 2018-02-26 11:55:47,729 sprog.SprogCommandStation             WARN  - Timeout Waiting for reply [AWT-EventQueue-0]
     [java] 2018-02-26 11:55:47,829 sprog.SprogCommandStation             WARN  - Timeout Waiting for reply [AWT-EventQueue-0]
     [java] 2018-02-26 11:55:47,930 sprog.SprogCommandStation             WARN  - Timeout Waiting for reply [AWT-EventQueue-0]
     [java] 2018-02-26 11:55:47,971 sprog.SprogCommandStation             WARN  - Slot thread wait timeout [SPROG slot thread]
     [java] 2018-02-26 11:55:48,031 sprog.SprogCommandStation             WARN  - Timeout Waiting for reply [AWT-EventQueue-0]
     [java] 2018-02-26 11:55:48,135 sprog.SprogCommandStation             WARN  - Timeout Waiting for reply [AWT-EventQueue-0]
     [java] 2018-02-26 11:55:48,239 sprog.SprogCommandStation             WARN  - Timeout Waiting for reply [AWT-EventQueue-0]
     [java] 2018-02-26 11:55:48,341 sprog.SprogCommandStation             WARN  - Timeout Waiting for reply [AWT-EventQueue-0]
     [java] 2018-02-26 11:55:48,502 sprog.SprogCommandStation             WARN  - Timeout Waiting for reply [AWT-EventQueue-0]

```

dheap, 2018-02-26:
@bobjacobsen @crosland I was sure I'd tested all this successfully before I released the Light-It definitions and associated code changes, hence the SPROG Turnouts and Signals table I have set to auto-load at startup with this profile.

So I went back and checked V4.10. All is well with that so the breaking change is recent.

dheap, 2018-02-26:
@crosland I'm also seeing the hang problem with Accessory Ops Packets. As well, Mobile Ops Packets and Mobile speed and function commands are not making it to the DCC track packets.

Some test files:
[SPROG Turnouts.zip](https://github.com/JMRI/JMRI/files/1757852/SPROG.Turnouts.zip)
[Archive.zip](https://github.com/JMRI/JMRI/files/1757858/Archive.zip)

dheap, 2018-02-26:
@crosland Edited my response. No problem with loco  packets. I'd turned off monitoring them in the Packet Analyser hardware config to stop the Pseudo-idle packets being sent, mea culpa.

Do we really need the pseudo-idle packets now the SPROG CS code is sending true IDLE packets (according to my Packet Analyser and your SPROG Command Monitor)?

dheap, 2018-02-26:
*"being seen", not "being sent"

dheap, 2018-02-26:
@crosland Realised the pseudo-idle generator is in firmware, not CS code so we are stuck with it...

crosland, 2018-02-26:
@dheap @AntonMezger Unfortunately I messed up in 4.11.2 and introduced an infinite wait if the command station does not see a reply from the hardware. I have some local changes, that I am almost ready to commit, that fix this and underlying issues (see below). I hope to get these in for 4.11.4 but I need to do more thorough testing.

Do you have the log before the constant stream of timeouts?

I have only seen this complete hang on a Pi-SPROG system that suffered a short circuit. The power retry timing causes the Pi-SPROG to miss a command (UART overrun) and then the command station code hangs waiting for a reply. USB SPROGs should be OK as the USB stack in IRQ driven and buffers the command.

I'm struggling to know why a IIv3 would hang in this way, but I have seen some pathalogical delays in JMRI due to the swing thread doing things. It's a lot worse on the RPi (over 4 s between packets being sent during startup).

My new code makes a couple of other changes: The slot thread is started at MAX_PRIORITY - 1; The command station reply handling was using a runnable and invokeLater() so it ran on the swing thread. I think this was the cause of delays due to other gui activity. I now despatch the replies directly from the traffic controller thread.

dheap, 2018-02-26:
Not much happens before trying to set an aspect except one or two `Slot thread wait timeout [SPROG slot thread]` messages.
 
[session.log](https://github.com/JMRI/JMRI/files/1760657/session.log)

dheap, 2018-02-26:
Realised I didn't have the Track Power switched on. Here's a log with that change. No change to behaviour.
[session.log](https://github.com/JMRI/JMRI/files/1760674/session.log)

crosland, 2018-02-27:
I am hoping that PR 4977 will fix this. I still have some issues on the R-Pi with some pathalogical delays.

AntonMezger, 2018-03-01:
Hi,

May I hope that in the next test release this will be solved ?

Thanks for all the work you have done to follow this problem up

dheap, 2018-03-02:
@AntonMezger Follow  progress on #4977 .

I've tested the code changes to date and was able to program Signal Mode. But  @crosland is working on some code improvements.

crosland, 2018-03-03:
@AntonMezger I have just submitted further code improvements so the fix should soon be ready merge and will, hopefully, appear in 4.11.4

closed 2021-04-14 by stale[bot]

AntonMezger, 2021-04-19:
I am using the last version I downloaded last friday and programming the light-it decoder as signal decoder using aspects as well as a loco decoder works fine. Problems solved!! Thanks everybody