JMRI/JMRI #4973: WiThrottle doesn't advertise connection address

KenNinomiya, 2018-02-27:
Macbook Pro (Late 2008)
OS 10.11.6

Last known working version: 4.11.2+Rb400a47

Known non-working versions: JMRI version 4.11.4ish+jenkins+20180226T1648Z+R4b60758; 4.11.4ish+jenkins+20180226T2247Z+R5d52dfd; 4.11.4ish+jenkins+20180227T0248Z+R2889638

When starting WiThrottle, the WiThrottle panel will display "Pending" instead of the ip address to connect a throttle to command station. This behavior has been repeatable on a Windows 10 machine and intermittently on a Linux machine.

The only oddity in log is that the working version shows statement #1 below and the non-working versions shows statement #2.

#1: 2018-02-26 21:52:05,904 withrottle.UserInterface              INFO  - Creating new WiThrottle DeviceServer(socket) on port 12090, waiting for incoming connection... [WiThrottleServer]

#2: 2018-02-26 21:38:44,420 withrottle.FacelessServer             INFO  - Published ZeroConf service for 'my jmri railroad._withrottle._tcp.local.' on 10.0.1.3:12090 [WiThrottleServer]
2018-02-26 21:38:44,421 withrottle.FacelessServer             INFO  - Creating new WiThrottle DeviceServer(socket) on port 12090, waiting for incoming connection... [WiThrottleServer]

rhwood, 2018-02-27:
Although the window shows “Pending...” a WiThrottle or EngineDriver app should still connect. Are those failing also?

KenNinomiya, 2018-02-27:
The WiThrottle app and Engine Driver apps connect to the WiThrottle server fine.

> 
> Although the window shows “Pending...” a WiThrottle or EngineDriver app should still connect. Are those failing also?
____________________________________________________________
Constant Fatigue Is A Warning Sign窶 Here's The Simple Fix
gundrymd.com
http://thirdpartyoffers.juno.com/TGL3131/5a95a009b0deb200846a1st01duc

mstevetodd, 2018-03-13:
I upgraded my RPi image to the latest Jenkins build, and this issue occurs consistently there. My image creates its own network using hostapd and assigns a static address to the server.
More problematic than the address not showing, clients are not picking up the zeroconf broadcast either. Not sure where this was broken, but 4.8 works properly, showing the address and broadcasting to the clients.

KenNinomiya, 2018-03-21:
Just updated JMRI to 4.11.4.R4534304 and this issue still present on:

Mac running 10.11.6
ESU CabControl command station (ICU) connected to AirPort Extreme over Ethernet

JMRI connects to CabControl ICU properly according to logs, but when WiThrottle server started, the "advertising IP address" dialog never populated the address to connect the android or iOS throttle to. Just says "pending."

Ken

Although DecoderPro connects to the

mstevetodd, 2018-03-21:
The issue seems to be that the current code only sets the UI address when the zeroconf broadcast is successfully received by the server. I'm adding logic to set the address directly when the zeroconf broadcast fails (which is of course when the address is most useful).

mstevetodd, 2018-03-22:
Found one issue, which was causing UserInterface's `facelessServer.getZeroConfService().addEventListener(this)` to be skipped, with warning `Unable to register for ZeroConf events`, which was why the UI didn't update before #5126 .
The creation of the FacelessServer and the UserInterface were on different threads, and this call was made before FacelessServer had created the ZeroConfService. This happened consistently on slower systems, for example, the RPi. For a test, I rearranged the calls to give FacelessServer more time, and now the listener is added and subsequent servicePublished(...) calls performed.
Unfortunately, the broadcast is still not being received by the clients. On to the next issue.....

mstevetodd, 2018-03-22:
On the RPi, I replaced jmdns-3.5.3.jar with the one from 4.8 (jmdns 3.5.1), restarted, and now the broadcast works properly to all my clients. Now to test with my other networks.

mstevetodd, 2018-03-23:
I have created PR #5138 to address several items with the WiThrottle server

mstevetodd, 2018-03-28:
This is working well on my the networks I have access to, and I've received a couple of positive responses on jmriusers. Marking complete.

closed 2018-03-28 by mstevetodd

KenNinomiya, 2018-04-18:
So, I just downloaded and installed JMRI test version 4.11.5 for Mac.

I see where WiThrottle server is now broadcasting the IP address to connect a wireless throttle.

However, my ESU CabControl command station connected by Ethernet cable to my AirPort Extreme wifi router is connected on 10.0.1.2. But WiThrottle server says the connection IP is 10.0.1.3:xxxxx

My Engine Driver connects fine on the broadcasted address. It controls a loco fine.

Is this as it should be?

Ken

Ken Ninomiya
Oxnard, CA

dheap, 2018-04-18:
That would be correct. Every device connected to your router (wired or wireless) will be assigned a unique IP address.

Your WiThrottle server is running on your Mac (JMRI) and that is what Engine Driver needs to talk to. Your Mac then talks to your ESU command station.

> However, my ESU CabControl command station connected by Ethernet cable to my AirPort Extreme wifi router is connected on 10.0.1.2. But WiThrottle server says the connection IP is 10.0.1.3:xxxxx

BHoffman351, 2018-04-18:
> On Apr 18, 2018, at 6:51 PM, KenNinomiya <notifications@github.com> wrote:
> However, my ESU CabControl command station connected by Ethernet cable to my AirPort Extreme wifi router is connected on 10.0.1.2. But WiThrottle server says the connection IP is 10.0.1.3:xxxxx
> 
> My Engine Driver connects fine on the broadcasted address. It controls a loco fine.
> 
Hi Ken,

Sounds fine to me.  The ESU and the JMRI computer exist at separate addresses on the AirPort network.  It would be a problem if the addresses were the same.

Regards,
Brett