JMRI/JMRI #4978: Control Panel Editor Issues

SkissorsWolf, 2018-02-27:
Two issues here: 
Currently running JMRI Version 4.10+R419243e.
1.) In the Control Panel Editor, when you change any of the icons in the Icon Sets all icons will disappear in the interface after clicking the 'Done' button. PanelPro has to be completely restarted in order for the interface to be fixed.

2.) I just had an instance where I was adding track sections to a control panel and every new section showed up off screen. When I got frustrated (due to the fact that I couldn't see them), I reloaded PanelPro and reopened the panels.  I had to delete the panel and start over due to where it placed the sections and it screwed up the scaling of the display. I suspect this issue might have to do with the fact that I was zoomed in while I was trying to place sections.

petecressman, 2018-03-02:
Several bugs were introduced late in 4.9. x.  An enhancement has a few issues. A fix is in process and will be submitted soon after more testing is done.
Please assign this issue to pete_cressman

rhwood, 2018-03-02:
@petecressman, I cannot assign the issue to you, but will able to assign the pull request to you. Include the text (on its own line) `Fixes #4978` in the text of the initial comment on the PR, and this issue will be tied to it.

mattharris, 2018-03-02:
@rhwood I've just added @petecressman to the Developers team which then allows assignment to issues & PR's, once the invitation is accepted.

closed 2021-04-14 by stale[bot]