JMRI/JMRI #4980: Font selection

bobjacobsen, 2018-02-28:
From jmri-developers @janboen 
> Would suggest that in the various panel editors it is possible to also select which font is used.
> At this moment it looks like Tahoma is being used in JMRI 4.10.
> In older panels this seems to be Tahoma Bold.
> Is to silly to have to revert to editing the XML file, which a normal user will not want or be able to do anyway.

dsand47, 2018-02-28:
Layout Editor and Panel Editor provide font selection in a context menu.
Control Panel Editor does not.  It is possible to switch from CPE view to PE view, make the font selection and switch back.

petecressman, 2018-03-02:
I will add this to DecoratorPanel after the PaletteTextFix branch is merged.  Plwase assign this issue to pet cressman

rhwood, 2018-03-02:
@petecressman, I cannot assign the issue to you, but will able to assign the pull request to you. Include the text (on its own line) `Fixes #4980` in the text of the initial comment on the PR, and this issue will be tied to it.

rhwood, 2020-04-18:
This appears to have been fixed by #5089

closed 2020-04-18 by rhwood