JMRI/JMRI #4986: Monitor Loconet parsing of OPC_SW_STATE messages

n3ix, 2018-03-02:
The parsing of OPC_SW_STATE (op code 0xBC) for the Monitor Loconet window fails if either the DIR or ON bits in SW2 are equal to 1.  During testing using a Digitrax DT100 throttle, 0xBC messages with the DIR bit either 0 or 1 were observed.  The DCS240 command station responded to all the 0xBC messages from the DT100 regardless of the value of the DIR bit.  Further testing with a JMRI script showed that the command station also responds whether the ON bit is 0 or 1 and also confirmed that the command station responds whether the DIR bit is 0 or 1.

The Loconet PE is silent on the DIR and ON bits in SW2 for an OPC_SW_STATE message.  Based on the observed behavior of Digitrax products reported above, it appears that the Monitor Loconet message can ignore those bits when parsing OPC_SW_STATE messages.  

Thanks.

devel-bobm, 2018-03-02:
@n3ix , 

I do not have a DT100 throttle, so I cannot try the example you are reporting.  If you can provide a few specific examples (i.e. LocoNet message hex data) of throttle and command station behavior, we can address the LocoNet Monitor problem and add testcases to help prevent behavioral-reversion once it has been fixed.

n3ix, 2018-03-02:
Below are 2 consecutive requests from a DT100.  The first request is for LT319 and the second is LT320.  DIR=1 in the first request and gets the parse error.  DIR=0 in the second request and it parses ok.  The command station responds to both requests.

14:13:23.154: [BC 3E 22 5F]  Unable to parse LocoNet message.
contents: BC 3E 22 5F
14:13:23.170: [B4 3C 10 67]  LONG_ACK: Command station response to switch state request 0x10 (Thrown).
14:13:23.420: [BC 3F 02 7E]  Request status of switch LT320 ().
14:13:23.436: [B4 3C 00 77]  LONG_ACK: Command station response to switch state request 0x00 (Thrown).

Here are two requests from a JMRI script for LT10.  In the first ON=1 and in the second ON=0.  The first requests gets the parse error and the second parses ok.  The command station responds to both requests.

14:16:57.717: [BC 09 10 5A]  Unable to parse LocoNet message.
contents: BC 09 10 5A
14:16:57.733: [B4 3C 51 26]  LONG_ACK: Command station response to switch state request 0x51 (Thrown).
14:17:04.842: [BC 09 00 4A]  Request status of switch LT10 ().
14:17:04.842: [B4 3C 51 26]  LONG_ACK: Command station response to switch state request 0x51 (Thrown).

devel-bobm, 2018-03-02:
@n3ix,

What JMRI version are you working with?

This issue was previously reported (issue #4697) and has already been addressed in the JMRI source code and appropriate testcases were added (issue #4698) in January of this year. 

So the problem exists in JMRI releases before and including JMRI "production" release 4.10 and "test" release 4.11.1.  It is resolved as of JMRI "test" release 4.11.2.

I double-checked the current code's behavior for the <BC> messages you listed.  I used a DB150 command station and sent the <BC> messages via the JMRI "Send LocoNet Packet" tool.  The LocoNet Monitor decoding of <BC> messages hasn't changed since the January 2018 fix.  Here is the result from LocoNet Monitor:

```
[BC 3E 22 5F]  Request status of switch LT319 ().
[B4 3C 00 77]  LONG_ACK: Command station response to switch state request 0x00 (Thrown).
[BC 3F 02 7E]  Request status of switch LT320 ().
[B4 3C 00 77]  LONG_ACK: Command station response to switch state request 0x00 (Thrown).
[BC 09 10 5A]  Request status of switch LT10 ().
[B4 3C 00 77]  LONG_ACK: Command station response to switch state request 0x00 (Thrown).
[BC 09 00 4A]  Request status of switch LT10 ().
[B4 3C 00 77]  LONG_ACK: Command station response to switch state request 0x00 (Thrown).
```

Please confirm that your problem is fixed in a recent version of JMRI (i.e. 4.11.2 or 4.11.3).  I will close this issue when you confirm proper behavior in one of these versions.

n3ix, 2018-03-02:
@devel-bobm  I have been testing production code.  Since I knew this wasn't current I attempted to do a search of the tickets before reporting this issue.  I guess I did the search improperly because I didn't find it.  Sorry for the trouble.  Please go ahead and close this, and also thanks for your quick responses.

devel-bobm, 2018-03-03:
@n3ix, you are welcome.

closed 2018-03-03 by devel-bobm