JMRI/JMRI #5001: NCE Tests Unreliable

rhwood, 2018-03-03:
We are beginning to see failures like:
```
[ERROR] Errors: 
[ERROR]   NceBinaryCommandTest.testCTor:17 NoClassDefFound Could not initialize class jm...
[ERROR]   NceLightTest.testCTor:19 Â» NoClassDefFound Could not initialize class jmri.jmr...
[ERROR]   NceThrottleTest.testSetF0:63 Â» NoClassDefFound Could not initialize class jmri...
[ERROR]   NceThrottleTest.testSetF1:73 Â» NoClassDefFound Could not initialize class jmri...
[ERROR]   NceThrottleTest.testSetF10:163 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF11:173 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF12:183 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF13:193 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF14:203 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF15:213 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF16:223 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF17:233 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF18:243 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF19:253 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF2:83 Â» NoClassDefFound Could not initialize class jmri...
[ERROR]   NceThrottleTest.testSetF20:263 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF21:273 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF22:283 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF23:293 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF24:303 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF25:313 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF26:323 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF27:333 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF28:343 Â» NoClassDefFound Could not initialize class jm...
[ERROR]   NceThrottleTest.testSetF3:93 Â» NoClassDefFound Could not initialize class jmri...
[ERROR]   NceThrottleTest.testSetF4:103 Â» NoClassDefFound Could not initialize class jmr...
[ERROR]   NceThrottleTest.testSetF5:113 Â» NoClassDefFound Could not initialize class jmr...
[ERROR]   NceThrottleTest.testSetF6:123 Â» NoClassDefFound Could not initialize class jmr...
[ERROR]   NceThrottleTest.testSetF7:133 Â» NoClassDefFound Could not initialize class jmr...
[ERROR]   NceThrottleTest.testSetF8:143 Â» NoClassDefFound Could not initialize class jmr...
[ERROR]   NceThrottleTest.testSetF9:153 Â» NoClassDefFound Could not initialize class jmr...
[ERROR]   NceThrottleTest>AbstractThrottleTest.testSetIsForward:105 Â» NoClassDefFound Co...
[ERROR]   NceThrottleTest>AbstractThrottleTest.testSetSpeedSetting:85 Â» NoClassDefFound ...
[ERROR]   NceTurnoutTest>AbstractTurnoutTestBase.testAddListener:68 Â» NoClassDefFound Co...
[ERROR]   NceTurnoutTest>AbstractTurnoutTestBase.testCommandClosed:93 Â» NoClassDefFound ...
[ERROR]   NceTurnoutTest>AbstractTurnoutTestBase.testCommandThrown:104 Â» NoClassDefFound
[ERROR]   NceTurnoutTest>AbstractTurnoutTestBase.testDispose:85 Â» NoClassDefFound Could ...
[ERROR]   NceTurnoutTest>AbstractTurnoutTestBase.testInvertedCommandClosed:168 Â» NoClassDefFound
[ERROR]   NceTurnoutTest>AbstractTurnoutTestBase.testInvertedCommandThrown:181 Â» NoClassDefFound
```
(From #4998)

These tests had been disabled because they are unreliable for reasons I do not understand; enabling them without fixing the underlying problems simply makes unit testing painful and undesirable.

pabender, 2018-03-03:
Sent from my iPhone

> On Mar 3, 2018, at 5:47 PM, Randall Wood <notifications@github.com> wrote:
> We are beginning to see failures like:
> 

Show me where.  I have not seem those on any of the CI servers.

rhwood, 2018-03-04:
Logs are lost when tests are restarted. Should we now stop all restarting for you?

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

bobjacobsen, 2018-03-04:
> Logs are lost when tests are restarted.

Will keep an eye out for this and capture a log to attach.  Sorry, didn’t realize there wasn’t a way to refer back to them (I thought that the URL was good for a while)

pabender, 2018-03-04:
@rhwood 

What I am saying is that I can't replicate this, and I haven't seen it happen since those tests were re-enabled.

WE need to know the circumstances that cause this.

It's impossible to fix if you don't understand that, and we can't just ignore our way out of the problem (which is what YOU choose to do previously).

From what you have posted here, I don't even have any clue what class can't be found.  Without that information, I have a clue as to how to start debugging this.

dheap, 2018-03-04:
Something is very broken with ci-test. If it try ant ci-test from Netbeans, partway through the number of threads and ports blows out dramatically. It starts throwing errors and going non-responding when my Mac Activity Monitor shows around 2,048 threads and in excess of 4,000 ports open!

Jconsole shows well over 1,000 Timer-nnn threads, hundreds of LocoNetThrottledTransmitter and around 1,000 openLCB timer and queue threads.

Whats the best/easiest way to get a thread dump?

dheap, 2018-03-04:
jstack pid produced:
[jstack.txt](https://github.com/JMRI/JMRI/files/1778229/jstack.txt)

pabender, 2018-03-04:
@dheap see https://github.com/JMRI/JMRI/pull/4818 and https://github.com/JMRI/JMRI/issues/4962

The biggest issue is the OpenLCB stuff ( https://github.com/openlcb/OpenLCB_Java/issues/119 ).

dheap, 2018-03-04:
Thanks. I get the idea. So no solution in sight as yet.

[jstack-thread-summary.txt](https://github.com/JMRI/JMRI/files/1778231/jstack-thread-summary.txt)

pabender, 2018-03-04:
Well, half of the battle is finding where the problem is.

I might be able to take a look at the OpenLCB side of this in a week or so, and then the JMRI side should be easier to fix.

rhwood, 2018-03-04:
I didn't ignore the failures; I simply could never get them to work reliably. See #3340 for the aborted history of attempting to get them working without crapping on other's ability to do their own work.

pabender, 2018-03-04:
> On Mar 4, 2018, at 7:07 AM, Randall Wood <notifications@github.com> wrote:
> I didn't ignore the failures; I simply could never get them to work reliably. See #3340 for the aborted history of attempting to get them working without crapping on other's ability to do their own work.
> 
I understand you attempted to make these work.   But you DID tell Maven to ignore those tests with a very broad blanket exclusion.

Perhaps the solution is to go back to using ant on the Travis tests, since that actually works.  In other words, perhaps we should revert the original cause of the problem, which was https://github.com/JMRI/JMRI/pull/3213

bobjacobsen, 2018-03-04:
I agree that moving back to Ant might be the right next step.

But another question:  Why now?  What's changed that's causing this to fail?

pabender, 2018-03-04:
@bobjacobsen 

Nothing changed other than In #4913 I re-enabled the NCE tests that @rhwood disabled as part of #3213.

So, in other words, these tests just have not been running on Travis since #3213 was merged in April 2017.

rhwood, 2018-03-05:
I used to be able to consistently get this to fail on macOS, but do not seem to be able to anymore.

rhwood, 2018-03-16:
OP is no longer concerned.

closed 2018-03-16 by rhwood