JMRI/JMRI #5005: Roster hangs while opening with 600+ entries, with 5Mb size pictures

KenC57, 2018-03-04:
Net result is the roster folder is about 2Gig. Time is 5-10 minutes at fastest noticed. No indication shows and swing seems locked during this. Test case was using PanelPro launch and then selecting roster. A thread name of Image Fetcher is spawned. Meanwhile the AWT thread is tied up with the following:
Thread [AWT-EventQueue-0] (Suspended)	
	waiting for: MediaTracker  (id=234)	
	Object.wait(long) line: not available [native method]	
	MediaTracker.waitForID(int, long) line: not available	
	ImageIcon.loadImage(Image) line: not available	
	ImageIcon.setImage(Image) line: not available	
	RosterIconFactory.getIcon(RosterEntry) line: 61	
	RosterTableModel.getIcon(RosterEntry) line: 210	
	RosterTableModel.getValueAt(int, int) line: 243	
	RosterTableModel.getPreferredWidth(int) line: 337	
	RosterTable.resetColumnWidths() line: 178	
	RosterTable.<init>(boolean, int) line: 128	
	RosterFrame.createTop() line: 476	
	RosterFrame.buildWindow() line: 340	
	RosterFrame.<init>(String, String, String) line: 143	
	RosterFrame.<init>(String) line: 134	
	RosterFrame.<init>() line: 130	
	RosterFrameAction.actionPerformed(ActionEvent) line: 55	
	JMenu$1(AbstractButton).fireActionPerformed(ActionEvent) line: not available	
	AbstractButton$Handler.actionPerformed(ActionEvent) line: not available	
	DefaultButtonModel.fireActionPerformed(ActionEvent) line: not available	
	DefaultButtonModel.setPressed(boolean) line: not available	
	JMenu$1(AbstractButton).doClick(int) line: not available	
	BasicMenuItemUI.doClick(MenuSelectionManager) line: not available	
	BasicMenuItemUI$Handler.mouseReleased(MouseEvent) line: not available	
	JMenu$1(Component).processMouseEvent(MouseEvent) line: not available	
	JMenu$1(JComponent).processMouseEvent(MouseEvent) line: not available	
	JMenu$1(Component).processEvent(AWTEvent) line: not available	
	JMenu$1(Container).processEvent(AWTEvent) line: not available	
	JMenu$1(Component).dispatchEventImpl(AWTEvent) line: not available	
	JMenu$1(Container).dispatchEventImpl(AWTEvent) line: not available	
	JMenu$1(Component).dispatchEvent(AWTEvent) line: not available	
	LightweightDispatcher.retargetMouseEvent(Component, int, MouseEvent) line: not available	
	LightweightDispatcher.processMouseEvent(MouseEvent) line: not available	
	LightweightDispatcher.dispatchEvent(AWTEvent) line: not available	
	Popup$HeavyWeightWindow(Container).dispatchEventImpl(AWTEvent) line: not available	
	Popup$HeavyWeightWindow(Window).dispatchEventImpl(AWTEvent) line: not available	
	Popup$HeavyWeightWindow(Component).dispatchEvent(AWTEvent) line: not available	
	EventQueue.dispatchEventImpl(AWTEvent, Object) line: not available	
	EventQueue.access$500(EventQueue, AWTEvent, Object) line: not available	
	EventQueue$3.run() line: not available	
	EventQueue$3.run() line: not available	
	AccessController.doPrivileged(PrivilegedAction<T>, AccessControlContext) line: not available [native method]	
	ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(PrivilegedAction<T>, AccessControlContext, AccessControlContext) line: not available	
	ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(PrivilegedAction<T>, AccessControlContext) line: not available	
	EventQueue$4.run() line: not available	
	EventQueue$4.run() line: not available	
	AccessController.doPrivileged(PrivilegedAction<T>, AccessControlContext) line: not available [native method]	
	ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(PrivilegedAction<T>, AccessControlContext, AccessControlContext) line: not available	
	EventQueue.dispatchEvent(AWTEvent) line: not available	
	EventDispatchThread.pumpOneEventForFilters(int) line: not available	
	EventDispatchThread.pumpEventsForFilter(int, Conditional, EventFilter) line: not available	
	EventDispatchThread.pumpEventsForHierarchy(int, Conditional, Component) line: not available	
	EventDispatchThread.pumpEvents(int, Conditional) line: not available	
	EventDispatchThread.pumpEvents(Conditional) line: not available	
	EventDispatchThread.run() line: not available	

So I'm figuring it is trying to read all 600+ image files to make the thumbnails for the roster page. Yet we can only show 20-30 lines at a time. We don't have limits on the images used within a roster entry and most users the concept of knowing how to resize to a 'sensible' value isn't a good concept.

So should we be caching the thumbnails? Deferring the making of those until you show those lines?  Filling in the image thumbnail later when we've had the chance to read it? Give a progress bar so they know why everything is locked up? Spawn a thread to read the roster so the main screen doesn't lock up?

I suspect we should be doing more than one of these. But this is a club layout that got into this without understanding why the system kept locking up for longer and longer times. I have a full copy of their profile and replicated on my system.

-ken c

bobjacobsen, 2018-03-04:
There are lots of things that could be done in this area to improve the code, but they're all a bit tricky.  The immediate thing to do is to reduce the size of the images; if there are 500 of them, it's worth a bit of time to do a batch-reduce to something better matched to the use than 5MB each.  (On macOS, GraphicConvertor is a _great_ tool for this)

KenC57, 2018-03-04:
I finished my test and resized them to around 30k each. The startup time for
the roster is now a reasonable 10-15 seconds. So this points the time sucker
rather clearly being the image processing.

Those downsized files did not seem to impact the views in the icon view in
the roster listing. The larger picture as shown in the roster media page is
only slightly impacted. Due to the size of that display, it might clear a
little bit with a 50-100k sized file. But do to the size of that window,
anything greater is going to be lost anyway. Going back to the full size
files, I still can't read most of the loco numbers anyway. So not being able
to read them on smaller file sizes does not make a difference.

I can see taking the files as whatever they give is reasonable. But to
process them every single time is a waste of cycles. Given we are seeing a
rise of systems like RasPi's for exactly this sort of use (supporting
decoders and phones), figuring how to optimize this will have some real
benefits. Personal view, the icon in the roster list is so small it only
gives the outline and color and very little else. So I'm not sure that is
really any use to users.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

KenC57, 2021-03-30:
A few different changes (back then) seems to minimize this to a tolerable level. Primary was processing the images in a different thread to not interfere with the rest of JMRI.

closed 2021-03-30 by KenC57