JMRI/JMRI #5016: Migrate Mailing Lists

rhwood, 2018-03-04:
The JMRI mailing lists hosted by SourceForge need to be migrated to another service. Ideally such a migration would entail moving the complete archives to the other service as well.

mattharris, 2018-03-06:
To facilitate this, I've created a jmri-developers group on groups.io

This is awaiting approval but, once approved, can then have sub-groups defined for the various sub-lists needed.

bobjacobsen, 2018-03-06:
There seems to be a [groups.io "JMRI" group](https://jmri.groups.io/g/JMRI), with a couple of subgroups, already existing at https://jmri.groups.io/g/JMRI 

Anybody know about that?

mattharris, 2018-03-06:
That one I don't know about, aside from finding about it the other day when attempting to set it up!

There is a jmriusers on groups.io that I created a while back last time Yahoo was having issues.

In the meantime, jmri-developers has been approved.

mattharris, 2018-03-08:
Ownership of the JMRI list on groups.io has now been determined.

Now to think about how we should proceed.

bobjacobsen, 2018-03-08:
The moderators on the existing Yahoo group can be reached at jmriusers-owner@yahoogroups.com

It would be good to include them in any discussions, as they're the ones with the most day-to-day experience.

mattharris, 2018-03-09:
Have reached out to groups.io support to find out what (if any) options we have with respect to transfer of mailing list archives from SourceForge. An extract of most list archives has already been performed.

Assuming that there is a solution forthcoming, we need to consider _which_ archives we would like to keep.

mattharris, 2018-03-09:
Just heard back from groups.io support and they are able to import the list archives.

mattharris, 2018-03-14:
An additional point with the way things are structured on groups.io is that a person cannot be solely subscribed to a subgroup - it is mandated that they are subscribed to the 'main' group as well.

Whilst I don't think this would be a major issue for most of our lists, we should consider the impact on some of the announcement-only ones.

For the time being, I think that we should concentrate on transferring the development-focused lists, leaving user-focused with Yahoo for now.

Subscription to the main group(s) should be open to anyone, but membership of certain subgroups will be restricted.

Bearing in mind the sub-group structure and constraints, I'd like to propose the following:

* jmri-developers - overall group for development discussions with subgroups
  * jmri@jmri-developers - main group for general development discussions
  * developers@jmri-developers - subgroup for code committers (restricted membership)
  * commits@jmri-developers - subgroup for commit reports from GitHub
  * builds@jmri-developers - subgroup for build reports from Jenkins
  * monitoring@jmri-developers - subgroup for infra availability reports (restricted membership)
* jmri-announce - separate list for project announcements (moderator/owner only posting)

User-focused list(s) would be separated from the developer lists.

One list not mentioned so far is the current 'jmri-reports' list. I'd suggest that this one would be better grouped under a subgroup of the user-focused list as opposed to the developer-focused ones - thoughts?

plocher, 2018-03-14:
Are there any downsides to having a unified presence, with developers and
users in the same hierarchy?  I'm thinking about the top level group being
an announcements one, with optional subscriptions to sub groups for
specialized stuff...  Additionally, some of the developer subgroups should
be connected to the git and Jenkins hooks, possibly via the groups.io
toolsets...

  -John

TOP LEVEL:

jmri@groups.io  - JMRI announcements (new production releases...) everyone
subscribed, both developer and user types, very low volume, restricted
posting ability

Sub Groups
(FUTURE) users@jmri.groups.io -  user stuff and all that entails whenever
that thread gets pulled
(NOW) developers@jmri.groups.io - overall group for development discussions
with subgroups

Sub-sub Groups

   - discussion@developers.jmri.groups.io - main group for general
   development discussions
   - maintainers@developers.jmri.groups.io - subgroup for decoder,
   localization, documentation & web "pull request" stuff (open membership)
   - committers@developers.jmri.groups.io - subgroup for code committers
   (restricted membership)
   - bugs@developers.jmri.groups.io - subgroup for bug reports
   - changes@developers.jmri.groups.io - subgroup for commit reports from
   GitHub
   - builds@developers.jmri.groups.io - subgroup for build reports from
   Jenkins
   - logs@developers.jmri.groups.io - subgroup for infra availability
   reports (restricted membership)

mattharris, 2018-03-14:
One issue is that the hierarchy structure is limited to a 'main' group and 'sub-groups'. It's not possible to have 'sub-sub-groups'.

Integration to Git and Jenkins is already investigated and configured for testing purposes.

Whilst the consolidation of everything under one unified umbrella group does seem attractive, there is the fact that storage limits are shared across all sub groups. Each individual top-level group has a new storage limit applied.

So, a consolidated view would give us 1GB total space for all attachments, files and photos from all sub-groups, whereas separate domain-specific groups would give 1GB space per group and related sub-groups.

Anyhow, thanks for the feedback on the proposal - gives me some food for thought.

bobjacobsen, 2018-03-15:
Might be worth a support request to ask if they can provide the subscriber list as a one-time service.

rhwood, 2018-03-16:
OP is no longer concerned.

mattharris, 2018-03-28:
Opened a support ticket with SourceForge for this - https://sourceforge.net/p/forge/site-support/17309/

mattharris, 2018-03-29:
Response from SourceForge:

> Hello Matt,
>
> Unfortunately due to new personal data regulations in Canada and the European Union we made changes to comply with last year, we are unable to export a raw list of mailing list subscriber e-mail addresses.
>
> What we recommend you do is send several e-mails at different times announcing the move and where to subscribe to the new list. Google Groups may be a good choice.
>
> Sincerely,
>
> SourceForge Support

So, looks like we cannot get the subscriber info...

rhwood, 2018-03-29:
We could just do the announce once a week for three weeks before the change and then respond to all new emails to the old list with an auto responder (unless SF’s mailman implementation doesn’t allow that) notifying users of the change.

That way no-longer interested people will be “automatically scrubbed” from the list by virtue of their not subscribing to the new list.

bobjacobsen, 2018-03-29:
>  then respond to all new emails to the old list with an auto responder (unless SF’s mailman implementation doesn’t allow that) notifying users of the change

Not sure if SF Mailman does it, but I can set up a local email filter here that’ll do that.

icklesteve, 2018-06-26:
I've  just submitted #5501 following full source code search to find remaining Yahoo links within help pages, have left a few ( eg. so people can find the Yahoo polls ).
There were a few references in some French swl ( shockwave? ) files which I've left alone, might be worth someone checking those over at some point?
Steve

mattharris, 2018-11-30:
Resurrecting this item as have started the process of migrating the developers list to groups.io - see [message posted on SF list ](https://sourceforge.net/p/jmri/mailman/message/36480330/) on 29-Nov.

rhwood, 2019-08-09:
Has the developer's archive been migrated?

rhwood, 2020-04-12:
Is this complete?

closed 2021-04-14 by stale[bot]