JMRI/JMRI #5017: Migrate Releases

rhwood, 2018-03-04:
Releases on SourceForge, if not already on GitHub need to be made available on GitHub, if a Git tag or GitHub release exists for that release.

We may want to retain SourceForge as a backup, but it needs to not be mentioned in our release notes, sites, or user groups except as a backup to GitHub or archive of older releases.

bobjacobsen, 2018-03-04:
JMRI 4.7.1 was the first release to have its download files on GitHub.  All since then have had the files on GitHub.

JMRI 4.6 and earlier have release tags on GitHub, c.f. this one for [JMRI 3.10.1](https://github.com/JMRI/JMRI/releases/tag/v3.10.1), but they don't have files attached nor descriptions beyond the release number. Most of them are dated Sept 9, 2017, too, which might be a bit confusing. 

The download files can be added to older if somebody wants to. Once that's done, their release notes can be updated to point to GitHub.

The only really old releases that probably need to have download files are:
- [JMRI 3.10.1](http://jmri.org/releasenotes/jmri3.10.1.shtml) for Java 1.6 users
- [JMRI 2.14.1](http://jmri.org/releasenotes/jmri2.14.1.shtml) for Java 1.5 users

rhwood, 2018-03-16:
OP is no longer concerned.

closed 2018-03-26 by bobjacobsen