JMRI/JMRI #5020: Add ReadMe to SourceForge files

rhwood, 2018-03-04:
We need to add a README file, in markdown syntax, at the root of the SourceForge files section, noting that the primary JMRI release site is GitHub. See [SF.net documentation](https://sourceforge.net/p/forge/documentation/Files-Readme/) on formats.

rhwood, 2018-03-04:
I added a basic README, but it needs to be updated once we have XtrkCadReader releases on GitHub.

bobjacobsen, 2018-03-05:
Putting the older XtrkCadReader files on GitHub is a bit tricky.  GitHub releases are really repository tags; I'm not sure we know exactly what to tag for those two XtrkCadReader files.

mattharris, 2018-03-05:
I really need to spend a bit of time finishing off the build improvements for XtrkCadReader I kicked off rather more months ago than I care to remember...

Once that's sorted out, it will give us a similar build mechanism to that used by the main JMRI project and enable the release process to be better streamlined and integrated.

rhwood, 2018-03-16:
OP is no longer concerned.

closed 2019-03-30 by bobjacobsen