JMRI/JMRI #5023: Debugging Info sent to SourceForge

rhwood, 2018-03-04:
Currently debugging info is sent to a mailing list on SourceForge. This needs to be sent elsewhere.

I can think of four possibilities for handling this:
- Send to a mailing list elsewhere and redirect the script at signer.jmri.org
- Create a bot at signer.jmri.org that creates "anonymous" submissions to GitHub Issues
- Create a proxy for JMRI applications that submits a GitHub Issue as an authenticated user
- Replace the code in JMRI applications with code that submits a GitHub Issue as an authenticated user

mattharris, 2018-03-09:
One issue I see with submitting via GitHub issues is that the GitHub API does not seem to support inclusion of attachments when creating and updating issues. It only seems possible to add attachments via the website UI.

bobjacobsen, 2018-03-09:
> GitHub API does not seem to support inclusion of attachments when creating and updating issues

The attachments (made manually) seem to be in a specific URL space:

https://github.com/JMRI/JMRI/files/1791539/session.log

Is there a part of the GitHub API that deals with files? Although a pain, the tool could upload new files and then include the URLs in the Issue

rhwood, 2018-03-16:
OP is no longer concerned.

closed 2021-04-14 by stale[bot]