JMRI/JMRI #5026: Conditional Tests fail on macOS

rhwood, 2018-03-05:
I get the following three failures consistently on macOS, using either ant or maven:
```
[java] There were 3 errors:
     [java] 1) comboBoxTest(jmri.jmrit.conditional.ConditionalTreeEditTest)org.netbeans.jemmy.TimeoutExpiredException: Component enabled: class javax.swing.JButton (ComponentOperator.WaitComponentEnabledTimeout)
     [java] 	at org.netbeans.jemmy.Waiter.waitAction(Waiter.java:213)
     [java] 	at org.netbeans.jemmy.operators.ComponentOperator.waitComponentEnabled(ComponentOperator.java:909)
     [java] 	at org.netbeans.jemmy.operators.AbstractButtonOperator.push(AbstractButtonOperator.java:352)
     [java] 	at jmri.jmrit.conditional.ConditionalTreeEditTest.comboBoxTest(ConditionalTreeEditTest.java:132)
     [java] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
     [java] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
     [java] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
     [java] 	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
     [java] 	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
     [java] 	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
     [java] 	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
     [java] 	at org.junit.internal.runners.statements.RunBefores.evaluate(RunBefores.java:26)
     [java] 	at org.junit.internal.runners.statements.RunAfters.evaluate(RunAfters.java:27)
     [java] 	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
     [java] 	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
     [java] 	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
     [java] 	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
     [java] 	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
     [java] 	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
     [java] 	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
     [java] 	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
     [java] 	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
     [java] 	at org.junit.runners.Suite.runChild(Suite.java:128)
     [java] 	at org.junit.runners.Suite.runChild(Suite.java:27)
     [java] 	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
     [java] 	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
     [java] 	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
     [java] 	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
     [java] 	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
     [java] 	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
     [java] 	at org.junit.runners.Suite.runChild(Suite.java:128)
     [java] 	at org.junit.runners.Suite.runChild(Suite.java:27)
     [java] 	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
     [java] 	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
     [java] 	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
     [java] 	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
     [java] 	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
     [java] 	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
     [java] 	at org.junit.runners.Suite.runChild(Suite.java:128)
     [java] 	at org.junit.runners.Suite.runChild(Suite.java:27)
     [java] 	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
     [java] 	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
     [java] 	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
     [java] 	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
     [java] 	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
     [java] 	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
     [java] 	at junit.framework.JUnit4TestAdapter.run(JUnit4TestAdapter.java:38)
     [java] 	at apps.tests.AllTest.main(AllTest.java:37)
     [java] 2) singlePickTest(jmri.jmrit.conditional.ConditionalTreeEditTest)org.netbeans.jemmy.TimeoutExpiredException: Frame with title "Single Pick List" (WindowWaiter.WaitWindowTimeout)
     [java] 	at org.netbeans.jemmy.Waiter.waitAction(Waiter.java:213)
     [java] 	at org.netbeans.jemmy.WindowWaiter.waitWindow(WindowWaiter.java:437)
     [java] 	at org.netbeans.jemmy.WindowWaiter.waitWindow(WindowWaiter.java:227)
     [java] 	at org.netbeans.jemmy.FrameWaiter.waitFrame(FrameWaiter.java:243)
     [java] 	at org.netbeans.jemmy.operators.FrameOperator.waitFrame(FrameOperator.java:431)
     [java] 	at org.netbeans.jemmy.operators.JFrameOperator.<init>(JFrameOperator.java:90)
     [java] 	at org.netbeans.jemmy.operators.JFrameOperator.<init>(JFrameOperator.java:126)
     [java] 	at org.netbeans.jemmy.operators.JFrameOperator.<init>(JFrameOperator.java:143)
     [java] 	at org.netbeans.jemmy.operators.JFrameOperator.<init>(JFrameOperator.java:158)
     [java] 	at jmri.jmrit.conditional.ConditionalTreeEditTest.singlePickTest(ConditionalTreeEditTest.java:100)
     [java] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
     [java] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
     [java] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
     [java] 	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
     [java] 	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
     [java] 	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
     [java] 	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
[... 62 lines cut ...]
     [java] 	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
     [java] 	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
     [java] 	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
     [java] 	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
     [java] 	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
     [java] 	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
     [java] 	at org.junit.runners.Suite.runChild(Suite.java:128)
     [java] 	at org.junit.runners.Suite.runChild(Suite.java:27)
     [java] 	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
     [java] 	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
     [java] 	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
     [java] 	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
     [java] 	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
     [java] 	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
     [java] 	at org.junit.runners.Suite.runChild(Suite.java:128)
     [java] 	at org.junit.runners.Suite.runChild(Suite.java:27)
     [java] 	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
     [java] 	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
     [java] 	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
     [java] 	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
     [java] 	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
     [java] 	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
     [java] 	at junit.framework.JUnit4TestAdapter.run(JUnit4TestAdapter.java:38)
     [java] 	at apps.tests.AllTest.main(AllTest.java:37)
```

dsand47, 2018-03-05:
The ConditionalTreeEditTest and jmri.jmrit.conditional.PackageTest work on macOS 10.13.3.

I run **ant tests** and then use the runtest.csh script.

rhwood, 2018-03-05:
Please verify you have no problems with running `ant alltest`. Tests that fail when run with other tests, but which work individually are bad tests that need to be fixed.

pabender, 2018-03-05:
Sent from my iPad

> On Mar 5, 2018, at 3:22 PM, Randall Wood <notifications@github.com> wrote:
> Please verify you have no problems with running ant alltest. Tests that fail when run with other tests, but which work individually are bad tests that need to be fixed.
> 
That is NOT always the case.

Sometimes the problem is the other tests are leaving a polluted test environment around.  You SHOULD write unit tests assuming that all previous tests have  cleaned the environment, but there are occasions when other tests leave garbage around, so hunting for issues with the failing test isn’t going to solve the problem.

rhwood, 2018-03-05:
lol I know is that code I don’t understand means I cannot run tests on my computer; and that means I have to disable the tests. If people who do understand this code; how it could or could not be polluted are only willing to say it works alone; this leaves me with no choice but to assume the entire premise of these tests is utter and complete wanking. We’ve disabled CI on macOS because it was “inconvenient” and since that was acceptable and, it seemed to me, directed, I have no choice but to make the assumption that testing on, or any development on, macOS should also be considered  “inconvenient”. I am just too goddamned tired of having people add tests to this code and then find it acceptable to leave us with situations where there is a high probability that any code submission, even one which does not touch a line of Java will just fail for reasons 100% unrelated to the submission. This is complete and utter Bullshit and had to stop, and stop soon, or I will begin a wholesale disablement of those tests that cause these failures.

pabender, 2018-03-05:
> On Mar 5, 2018, at 4:31 PM, Randall Wood <notifications@github.com> wrote:
> lol I know is that code I don’t understand means I cannot run tests on my computer; and that means I have to disable the tests. If people who do understand this code; how it could or could not be polluted are only willing to say it works alone; this leaves me with no choice but to assume the entire premise of these tests is utter and complete wanking. We’ve disabled CI on macOS because it was “inconvenient” and since that was
> 
No, we disabled tests CI tests on Travis because Travis had MacOS capacity issues.

I sm working through the process of setting up my Mac to be useable as a JMRI development machine again.   Once that process finishes, I will see if I can replicate what you are seeing.

dsand47, 2018-03-06:
I finally have a completed **ant alltest** run.  It took 65 minutes on my MBA.

I had to comment out a button push in ConditionalTreeEditTest because the push was occurring before the program had a chance to enable the button.  The other 2 errors no longer occur after moving jmrix.

I moved **jmri.jmrix.PackageTest.class** to the end of **jmri.PackageTest**.  the jmrix package creates over a thousand failures after the thread count hits 2048.  Running jmrix last allows the other tests to finish before the **OutOfMemoryError** failures start occurring. 

The attached zip file has 3 files. 

- **alltest-output.txt** &mdash; **57K** lines from the terminal window.
- **alltest-errors.txt** &mdash; **1097** error header lines.
- **tests.log** &mdash; The standard test log file.

There are some WARN messages from ConditionalTreeEditTest that I added to debug its problem.

[alltest.zip](https://github.com/JMRI/JMRI/files/1786147/alltest.zip)

pabender, 2018-03-06:
@dsand47 

I think the proper thing to do right now would be to submit a PR to disable these tests for now.

I'm stating that based on two things:
1) You had to disable a button push for one test to pass, so that test needs to be re-worked so the button push can happen.

2) You moved the package, so the conditions of the test changed.  That means the test is somehow dependent on things that happen before it does.  In other words, the setup isn't right for the test, and there is some other test that isn't cleaning up after itself.

My 2010 era Mac Mini doesn't appear to be up to this task any longer.  After quite a bit of frustration, I was able to get ant to compile, but it took longer than it should have.  Maven won't even compile the code, because the compiler runs out of heap space.  It might just be time to invest in a new Mac for me, though that isn't part of the budget right now.

On the threading issues, I started to go through the process today of cleaning up the OpenLCB library code so that it actually allows threads to be terminated.  I was pretty quickly able to get JMRI's tests to only see 1800ish total threads during a test run, but in the process of doing that, I broke the OpenLCB tests, so I have to fix those before I can continue.

bobjacobsen, 2018-03-06:
> You moved the package, so the conditions of the test changed. That means the test is somehow dependent on things that happen before it does. In other words, the setup isn't right for the test, and there is some other test that isn't cleaning up after itself.

I think that, in this case, it might just be the thread load of the jmrix tests. There might actually be anything wrong in jmrit. Or there might be, of course.

dsand47, 2018-03-06:
I have another **ant alltest** running right now.  The problematic button push is enabled and worked fine.

It appears that the root problem is the thread issue.  The jmrix package was before jmrit which left jmrit tests susceptible to thread issues.

pabender, 2018-03-07:
OK, it does sound like the threading is the primary issue.

I might make some progress on threads tomorrow, but I'm really going to get back to that thread issue in Ernest next week (It's spring break for me, so I'll have some time...).

rhwood, 2018-03-16:
OP is no longer concerned.

closed 2018-03-16 by rhwood