JMRI/JMRI #5030: Any known users of the Valid enum in jmri.jmrit.ussctc.Lock?

pabender, 2018-03-05:
I am trying to determine what the purpose of the Valid enum in jmri.jmrit.ussctc.Lock is.

https://github.com/JMRI/JMRI/blob/fea4d4b9d67674e8ba9407f6dfe1ff00ced42521/java/src/jmri/jmrit/ussctc/Lock.java#L14-L19

It doesn't appear to be referenced in any JMRI code.

@bobjacobsen, your name is associated with this, do you remember what it was for?

bobjacobsen, 2018-03-05:
It's used in the Python scripts that interact with this to drive a particular (real) CTC machine.  Someday, some of that code will probably move into Java, but not yet.

pabender, 2018-03-05:
Thanks @bobjacobsen 

I'm trying to figure out how to get that particular file to register as being either tested or to tell Jacoco to ignore it.  

Most of the interface class files that appear as not being tested (that is not having at least one line covered), I was able to put on the exclude list, but this one is still there.

I've determined the likely cause is the enum.  There was a second file that responded to the exclusion in the same way, but it dropped off the list when i used one of the enum values where it was used by the code.

I might just have to write a test that verifies the enum values exist....

closed 2018-03-11 by pabender