JMRI/JMRI #5045: Invalid test failure

rhwood, 2018-03-07:
Recent Travis CI test failure in the Cucumber code is attached.

Note that the JMRI web server *does not* use cookies, so this failure is *completely bogus*. I really have to ask, since these cucumber tests appear to be more about testing Firefox and Chrome browsers than anything else, why are we doing this? I'd understand if the person who created these tests put some interaction into them, but simply doing a single HTTP get could be done with a couple of lines of standard Java code and not launching browsers that really do not appear to be liking it, or are failing making non-JMRI related requests.
[log.txt](https://github.com/JMRI/JMRI/files/1788974/log.txt)

pabender, 2018-03-07:
@rhwood 

I don’t see an error message about cookies.  There is an informational message about cookies,but not an error.  

The only error I see is from the code that downloads new versions of the web drivers.  That was something you suggested we use.  I will see if I can isolate that error message so it doesn’t cause test failures when it occurs.  If I can’t, I will revert to manually managing the web drivers.

As far as what the tests do, they verify that the web server 
1) actually produces pages it says it should in response to the web browser request and
2) that those pages are at least partly visible by various web browsers.

In other words they look at the interaction between JMRI and a real web browser.  This is always going to be the first step in any web based test suite, and you always want to do that separately from a test of any further functionality.

pabender, 2018-03-07:
#5046 (now merged) should turn off the logging messages from the webdriver manager.

rhwood, 2018-03-16:
OP is no longer concerned.

closed 2018-03-16 by rhwood