JMRI/JMRI #5058: Layout Editor NPEs

Pugwash1, 2018-03-09:
@geowar1 @dsand47 
I believe you are the knowledgeable ones. 
Right click on track segment on a Layout Panel yields an NPE. Seems to be in display/layoutEditor/TrackSegment.java  due to this and a couple of similar lines failing.
```java
ImageIcon imageIcon = new ImageIcon(getClass().getResource("/resources/icons/decorations/ArrowStyle1.png"));
```
This worked for me, but not likely the correct fix.. 
```java
ImageIcon imageIcon = new ImageIcon(FileUtil.findURL("program:resources/icons/decorations/ArrowStyle1.png", FileUtil.Location.INSTALLED)); 
```

And right click on a blockcontents icon yields:
```
jmri.jmrit.display.PositionableLabel.setRotateOrthogonalMenu(PositionableLabel.java:509)
```
Which I have ~~knowledge~~ **nooooo clue** about. (once you start playing with this format stuff its *quite* **fun**).

dsand47, 2018-03-09:
Can you post a session.log that contains these errors?

Pugwash1, 2018-03-10:
@dsand47 
Not very interesting I'm afraid.

~~~
2018-03-09 19:23:15,932 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2018-03-09 19:23:16,000 util.Log4JUtil                        INFO  - This log is appended to file: /home/sg/.jmri/log/messages.log [main]
2018-03-09 19:23:16,001 util.Log4JUtil                        INFO  - This log is stored in file: /home/sg/.jmri/log/session.log [main]
2018-03-09 19:23:16,010 apps.Apps                             INFO  - PanelPro version 4.11.4ish+sg+20180310T0020Z+Ref974f7f7a starts under Java 1.8.0_144 on Linux amd64 v4.15.6-300.fc27.x86_64 at Fri Mar 09 19:23:16 EST 2018 [main]
2018-03-09 19:23:19,545 apps.Apps                             INFO  - Starting with profile SignalMastsClient.3dcb7e8b [main]
2018-03-09 19:23:19,731 node.NodeIdentity                     INFO  - Using jmri-485B39EF6C80-3dcb7e8b as the JMRI Node identity [main]
2018-03-09 19:23:20,190 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]
2018-03-09 19:23:21,894 server.WebServer                      INFO  - Starting Web Server on port 12080 [WebServer]
2018-03-09 19:23:22,040 util.FileUtilSupport                  INFO  - File path program: is /home/sg/devl/jmri/JMRI-Pugwash1/JMRI/ [main]
2018-03-09 19:23:22,041 util.FileUtilSupport                  INFO  - File path preference: is /home/sg/JMRI-Files/JMRI-Layouts/SignalMasts/ [main]
2018-03-09 19:23:22,041 util.FileUtilSupport                  INFO  - File path profile: is /home/sg/JMRI-Files/JMRI-Profiles/SignalMastsClient/ [main]
2018-03-09 19:23:22,042 util.FileUtilSupport                  INFO  - File path settings: is /home/sg/.jmri/ [main]
2018-03-09 19:23:22,042 util.FileUtilSupport                  INFO  - File path home: is /home/sg/ [main]
2018-03-09 19:23:22,043 util.FileUtilSupport                  INFO  - File path scripts: is /home/sg/devl/jmri/JMRI-Pugwash1/JMRI/jython/ [main]
2018-03-09 19:23:23,030 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2018-03-09 19:23:23,323 server.WebServer                      INFO  - Starting ZeroConfService _http._tcp.local for Web Server with properties {path=/, json=4.1} [WebServer]
2018-03-09 19:23:28,999 withrottle.FacelessServer             INFO  - Published ZeroConf service for 'bbabr-sm-client._withrottle._tcp.local.' on fd00:1:0:0:0:0:0:10%enp0s10:12090 [WiThrottleServer]
2018-03-09 19:23:31,725 withrottle.FacelessServer             INFO  - Published ZeroConf service for 'bbabr-sm-client._withrottle._tcp.local.' on 192.168.2.10:12090 [WiThrottleServer]
2018-03-09 19:23:31,726 withrottle.FacelessServer             INFO  - Creating new WiThrottle DeviceServer(socket) on port 12090, waiting for incoming connection... [WiThrottleServer]
2018-03-09 19:23:57,130 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at javax.swing.ImageIcon.<init>(ImageIcon.java:217)
	at jmri.jmrit.display.layoutEditor.TrackSegment.showPopup(TrackSegment.java:890)
	at jmri.jmrit.display.layoutEditor.LayoutEditor.showEditPopUps(LayoutEditor.java:5633)
	at jmri.jmrit.display.layoutEditor.LayoutEditor.mouseReleased(LayoutEditor.java:5470)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
[... cut ...]
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
~~~

geowar1, 2018-03-10:
This work on macOS… I suspect it doesn't on Windows?
Try this instead (TrackSegment.java:891):
               
`ImageIcon imageIcon = new ImageIcon(FileUtil.findURL("resources/icons/decorations/ArrowStyle1.png"));
`
There are four more immediately after this one.

geowar1, 2018-03-10:
It just occurred to me… there have been cases of the decorations directory not getting installed; you may want to check to see if the path (and icon files) is there (in your JMRI directory).

Pugwash1, 2018-03-10:
@dsand47 @geowar1 
Ok. Eclipse hasn't been told the new paradyme of using getclass().getresource() so doesn't copy everything to the hot fix area. Sorry for the trouble.

closed 2018-03-10 by Pugwash1

SteveRawlinson, 2018-03-10:
I'm seeing this error too (using Intellij) on Mac and Windows. The files are present. Can you explain that fix @Pugwash1 ?

Pugwash1, 2018-03-10:
@SteveRawlinson I realy dont have a fix. Eclipse will find getresources() if the resource  is in the same directory as the class file and have no leading "/" with the "/" infront I have no clue where its looking and wont tell me unless it finds a file. The copy resources doesn't work as the resources are not under the class. Im too sure why these 4 references dont just use the normal way.

bobjacobsen, 2018-03-10:
The `new ImageIcon(FileUtil.findURL(..))` idiom is the right approach to referencing a file, e.g. an image file, in JMRI.

The `new ImageIcon(getClass().getResource(..))` form isn't right for accessing a JMRI image file. That doesn't work with the various install and common IDE setups, it's not user friendly, etc.  There are only five occurrences of it in the current source code, all in `jmri.jmrit.display.layoutEditor.TrackSegment`, so the solution is to just change them over.

dsand47, 2018-03-10:
I can make the changes today.