JMRI/JMRI #5061: JsonHttpService.java  @Nonnull error

DanielBoudreau, 2018-03-10:
Eclipse IDE is reporting this error:
 
Description         Resource             Path       Location               Type
The nullness annotation @Nonnull is not applicable for the primitive type void    JsonHttpService.java                /jmri/java/src/jmri/server/json line 100 Java Problem
 
 
   /**
     * Respond to an HTTP DELETE request for the requested name.
     * <p>
     * Throw an HTTP 405 Method Not Allowed exception if the object is not
     * intended to be removable.
     * <p>
     * Do not throw an error if the requested object does not exist.
     *
     * @param type   the type of the deleted object
     * @param name   the name of the deleted object
     * @param locale the requesting client's Locale
     * @throws JsonException if this method is not allowed or other error occurs
     */
    @Nonnull
    public void doDelete(@Nonnull String type, @Nonnull String name, @Nonnull Locale locale) throws JsonException {
        throw new JsonException(HttpServletResponse.SC_METHOD_NOT_ALLOWED, Bundle.getMessage(locale, "DeleteNotAllowed", type));
    }

DanielBoudreau, 2018-03-15:
Randall,

Your latest commit has not fixed the problem:

Description	Resource	Path	Location	Type
The nullness annotation @Nonnull is not applicable for the primitive type void	JsonHttpService.java	/jmri/java/src/jmri/server/json	line 100	Java Problem

Looking at the history, it seems to me that you added the @Nonnull annotation.

rhwood, 2018-03-15:
I am fully aware:
1. that my last commit did not address some inconvenience to developers
2. that I added that annotation that causes that inconvenience
3. of how I need to fix it

My fixing *broken code that prevents users from using JMRI* does not and will not be held up by fixing problems unrelated to those.

You are just going to have to accept that I may have to fix someone else's issues before fixing issues you reported. If you are going to write harassing messages when user-facing problems have priority over yours, then I really don't know how or why to help.

bobjacobsen, 2018-03-15:
I've got a couple minutes before work, will PR this.

bobjacobsen, 2018-03-15:
PR #5099 will make the annotation correct with respect to the code defining the method and all uses of the routine, hence removing the syntax error.

If doDelete is better returning a feedback type, that's fine for a future update which can also change the annotation to be consistent with whatever is returned.

rhwood, 2018-03-15:
I have this on my schedule for this weekend; I have it on a branch that is mostly complete. My comments were more about getting pinged having dared do a commit that addresses someone else's problem first. But if we are going to be calling out users for each individual warning, not error, in eclipse, why isn't that the other [44-Eclispe compiler warnings](http://jmri.tagadab.com/jenkins/job/Development/job/Builds/2393/warnings20Result/NORMAL/) are not in the next commit a concern, or the 100s of FindBugs/SpotBugs a concern?

rhwood, 2018-03-16:
Assignee is no longer involved.

bobjacobsen, 2018-03-16:
Closed by #5099

closed 2018-03-16 by bobjacobsen