JMRI/JMRI #5068: Script panels are not available on DP3 menus

crosland, 2018-03-11:
On the web page http://jmri.org/help/en/html/tools/scripting/Start.shtml it says "There are several ways to use Python scripts with JMRI. The easiest is to use the built-in support in the standard JMRI applications: PanelPro, DecoderPro, etc."

It then goes on to talk about the script entry and output panels, which are not available from the DP3 menus.

bobjacobsen, 2018-03-12:
> the script entry and output panels, which are not available from the DP3 menus.

Should they be?

We could either change the page to not mention them in the context of DecoderPro, or add them and leave the page as is.

dsand47, 2018-03-12:
They are indirectly available using **File >> Open PanelPro Window**.

closed 2021-04-14 by stale[bot]