JMRI/JMRI #5069: Pi-SPROG One Command Station Issues

crosland, 2018-03-11:
Even with the latest code checked in for the 4.11.4 build, there are issues with the Pi-SPROG One command station operation. A number of workarounds should be applied:

- Do not turn on the track power until the power supply to the Pi-SPROG ONE is turned on
- Do not enable extra debugging through default.lcf, nor open the system console window
- Do not open the SPROG console window
- Disable power conversion (CV 29) in all locos
- If connecting via VNC (or similar), setup the screen layout before turning on the track power

Pi-SPROG One with version 2.4 or earlier firmware will cause a timeout after a track short circuit. The track power will be turned off and must be turned on again using the power control.

The issues are suspected (admittedly on little evidence, so far) to stem from the lack of real time performance of the R-Pi, especially when logging is enabled, causing frequent writes to the Flash file system.

I continue to look for a more robust solution.

bobjacobsen, 2018-03-11:
Should we put these bullets into the release note?

crosland, 2018-03-11:
@bobjacobsen Yes! I have submitted a change. Thank you for the reminder.

bobjacobsen, 2018-03-13:
I don't have hardware to test the following with, but perhaps these are worth trying:

- Raise JMRI's entire scheduler priority.  Not sure how the RPi linux handles this exactly, there are some differences between versions, but you usually can do it with the `nice` command when starting JMRI, or the `renice` command afterwards.  For most Linuxes, a value of -12 is about as aggressive as one should attempt.  

- Tell the Java garbage collector to try to limit the pauses due to garbage collection.  The option to include in the JVM options is
```
-XX:MaxGCPauseMillis=250
```
where the 250 is the (intended) max time in milliseconds.

bobjacobsen, 2018-03-14:
If it’s ok _except_ for while logging, we could try using a different logging configuration setup. The .lcf file has the simplest possible “write to file” configured. IIRC there are various other modes, one or more of which might coexist better. 

Is logging the primary remaining problem?

closed 2021-04-14 by stale[bot]