JMRI/JMRI #5083: PanelPro window positioning on startup

SteveRawlinson, 2018-03-13:
When you move windows around the screen they record their new position in a file. When you start PanelPro windows open in the position they were in when JMRI was last closed with the exception of the main PanelPro window which always opens in the middle of the screen.

Is there a reason for this?

silverailscolo, 2018-03-13:
The position of the main screen in the center of the screen is calculated upon start up of each of the apps. I noticed this while fixing the Help button.
You could make that placement into an option, but I would not recommend always just using the last location. Seems risky to lose it off screen.

bobjacobsen, 2018-03-13:
> main PanelPro window which always opens in the middle of the screen.

I dimly recall discussion of this from a UI perspective. I think there were technical issues balancing “get the final window up quick” (as opposed to a placeholder), vs “it’s disconcerting to have the window start in the center and then move around”. 

We can certainly change it. Might be good to have a discussion on jmri-developers or Jmriusers about how best to integrate it with startup.

mstevetodd, 2018-03-13:
If this is considered issue-hijacking, I apologize, but I've also noticed that the WiThrottle server window has fixed starting position (centered at the top).

mattharris, 2018-03-13:
@mstevetodd That would appear to be an explicit decision: https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/withrottle/UserInterface.java#L206

closed 2021-04-14 by stale[bot]