JMRI/JMRI #5084: JsonClientHandler no longer accepting "ping"

mstevetodd, 2018-03-13:
Loaded the latest Jenkins build, on Ubuntu, Raspbian Jessie and Windows 10. Started JMRI Web server. Opened initial "home" page of server. In the log, I now receive regular occurrences of the following message:

`json.JsonClientHandler                WARN  - Requested type 'ping' unknown. [WebServer-120]`

As far as I have tested, other functionality is unchanged.

mstevetodd, 2018-03-19:
I am still receiving the same error, in 4.11.4, and in the latest code from master.

rhwood, 2018-03-19:
Do you get this error if you do an `ant realclean debug` and then run from head of master?

mstevetodd, 2018-03-20:
Pulled again, realclean, debug, run, same result. 
Looking at the code, I see no jmri.server.json.ping package. I can add that, or is there a better way to handle ping?

rhwood, 2018-03-20:
Ping should be handled by the JsonUtilSocketService.

rhwood, 2018-03-20:
I am completely incapable of recreating this in head of master after ensuring it is clean (i.e., I can't get failure by running `ant debug panelpro` and selecting a profile with a panel when PanelPro starts. Where are you seeing this (what page), in what browser and OS, etc...

mstevetodd, 2018-03-20:
This occurs shortly after opening the JMRI home page (/) in Chrome, Edge and FireFox on a Windows 10 client. Same result from FireFox on my Ubuntu cloud machine desktop.

I was running the Run As Ant Build... in Eclipse (on Win10), but this morning I installed the Ant binaries 
and tried from there as well. "ant realclean debug PanelPro". Same result. This was after running "git pull https://github.com/JMRI/JMRI" which grabbed a few updates from last night.

I've also loaded the latest Jenkins 4.11.5ish+jenkins+20180320T0651Z+Rf62cef0 on my jmri.mstevetodd.com:1080 server. Same result. I'll leave this version running, with the console window available, if you'd like to browse to it.

pabender, 2018-03-20:
@mstevetodd,

What URL is being requested and what object is expected as a return value?

This seems like a good use for a cucumber test.

mstevetodd, 2018-03-20:
I believe it's coming from the websocket heartbeat message 
`{"type":"ping"}`

I can get the same result by entering that message in the json web console.

rhwood, 2018-03-20:
Opening any JSON socket (web or standard), should have a hello message transmitted from the JMRI server. That message contains a heartbeat interval that the client should send a ping message (`{"type":"ping"}`) more frequently than that interval and receive a pong message back (`{"type":"pong"}`).

rhwood, 2018-03-20:
I found the problem; my master branch has a commit that did not make it into JMRI/master, but I thought did. Will have a PR for it tonight.

PR #5092 (2018-03-15): Handle JSON Pings
A recent change caused all JSON message types from clients that were not also responses from the JMRI JSON server to be ignored. This fixes that as well as a couple of JSON schema.

Fixes #5084
PR #5124 (2018-03-21): Respond properly to JSON "ping"s.
Also contains correct schema for messages from server and pongs as well some adjustments to testing JSON messages.

Fixes #5084 (for real this time; I thought 62e9d82 was in #5092, but evidently I failed to push it before creating that PR)
commit e05cb233 (2018-03-21): Respond properly to JSON "ping"s. (#5124)

* Handle received as well bidirectional JSON types

Fixes #5084

* Prevent missing class errors in NetBeans

* Create a helper for getting schema from messages.

* Correct schemas to allow dataless heartbeats and goodbyes from server.

* Require that nodes tested for schema validation not be missing

* Allow verbose logging of outputs
commit 62e9d822 (2018-03-27): Handle received as well bidirectional JSON types

Fixes #5084
closed 2018-03-21 by mstevetodd