JMRI/JMRI #5087: Raspberry Pi Sound Crash

bunnshill, 2018-03-13:
This may be related to #4678.

My Raspberry pi will play wav sounds, such as bell.wav once only in a particular logix.  The second attempt to run the logic fails with the message as below.  The panel does not have this problem on an iMac or MacBookPro.  I am not sure if this problem is inherent in the Raspberry Pi. If so I dont know how to modify the settings of the Raspberry pi to allow the panel to work properly.  The Raspberry Pi will happily play YouTube sound videos. There is a warning below that no sounds will be available yet it plays sounds correctly once only for each logix.

The logix is there to warn the engineers that the signals have cleared for them when a new route is selected at an interlocking.  I have added a 7 second delay to allow all the switches to change before the signals clear.  i also use logic to play the crossing sound when trains are approaching a grade crossing.

2018-03-04 16:54:47,919 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2018-03-10 20:17:05,799 util.Log4JUtil                        INFO  - This log is appended to file: /home/pi/.jmri/log/messages.log [main]
2018-03-10 20:17:05,804 util.Log4JUtil                        INFO  - This log is stored in file: /home/pi/.jmri/log/session.log [main]
2018-03-10 20:17:05,899 apps.Apps                             INFO  - PanelPro version 4.11.3+R80b3ab3 starts under Java 1.8.0_65 on Linux arm v4.9.35-v7+ at Sat Mar 10 20:17:05 GMT 2018 [main]
2018-03-10 20:17:07,492 apps.Apps                             INFO  - Starting with profile Snake_Bend_Railroad.3f679432 [main]
2018-03-10 20:17:08,194 node.NodeIdentity                     INFO  - Using jmri-B827EBFBB2B5-3f679432 as the JMRI Node identity [main]
2018-03-10 20:17:08,769 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "LocoNet" [main]
2018-03-10 20:17:09,728 locobufferusb.LocoBufferUsbAdapter    INFO  - LocoBuffer-USB adapter set hardware flow control, mode=2 RTSCTS_OUT=2 RTSCTS_IN=1 [main]
2018-03-10 20:17:09,740 locobuffer.LocoBufferAdapter          INFO  - ttyUSB0 port opened at 57600 baud with DTR: true RTS: true DSR: true CTS: true  CD: false [main]
2018-03-10 20:17:13,754 util.FileUtilSupport                  INFO  - File path program: is /home/pi/.jmri/JMRI/ [main]
2018-03-10 20:17:13,757 util.FileUtilSupport                  INFO  - File path preference: is /home/pi/JMRI/ [main]
2018-03-10 20:17:13,759 util.FileUtilSupport                  INFO  - File path profile: is /home/pi/JMRI/Snake_Bend_Railroad/ [main]
2018-03-10 20:17:13,762 util.FileUtilSupport                  INFO  - File path settings: is /home/pi/.jmri/ [main]
2018-03-10 20:17:13,765 util.FileUtilSupport                  INFO  - File path home: is /home/pi/ [main]
2018-03-10 20:17:13,767 util.FileUtilSupport                  INFO  - File path scripts: is /home/pi/.jmri/JMRI/jython/ [main]
2018-03-10 20:17:14,811 server.WebServer                      INFO  - Starting Web Server on port 12080 [WebServer]
2018-03-10 20:17:15,970 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2018-03-10 20:17:18,354 server.WebServer                      INFO  - Starting ZeroConfService _http._tcp.local for Web Server with properties {path=/, json=4.0} [WebServer]
2018-03-10 20:17:27,370 withrottle.UserInterface              INFO  - Creating new WiThrottle DeviceServer(socket) on port 12090, waiting for incoming connection... [WiThrottleServer]
2018-03-10 20:18:55,898 withrottle.UserInterface              INFO  - Creating new WiThrottle DeviceServer(socket) on port 12090, waiting for incoming connection... [WiThrottleServer]
2018-03-10 20:18:55,928 withrottle.DeviceServer               INFO  - Received Name: David ’s 6+ [WiThrottleUIDeviceServer]
2018-03-10 20:20:05,933 audio.NullAudioFactory                WARN  - Initialised Null audio system - no sounds will be available. [AWT-EventQueue-0]
2018-03-10 20:20:18,831 script.JmriScriptEngineManager        INFO  - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 1.8.0_65 [AWT-EventQueue-0]
2018-03-10 20:20:18,851 script.JmriScriptEngineManager        INFO  - python 2.7 is provided by jython 2.7.0 [AWT-EventQueue-0]
Bunnshill.py
PM42 listener for shorts and autoreverse started.
Sound decoder address 1
A Approach B Stop Signal E Stop Signal 
= E Clear -> C .2018-03-10 20:24:19,045 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IllegalArgumentException: No line matching interface Clip supporting format PCM_SIGNED unknown sample rate, 16 bit, stereo, 4 bytes/frame, big-endian is supported.
	at javax.sound.sampled.AudioSystem.getLine(AudioSystem.java:479)
	at javax.sound.sampled.AudioSystem.getClip(AudioSystem.java:522)
	at jmri.jmrit.Sound.<init>(Sound.java:81)
	at jmri.jmrit.Sound.<init>(Sound.java:55)
	at jmri.implementation.DefaultConditional.takeActionIfNeeded(DefaultConditional.java:937)
	at jmri.implementation.DefaultConditional.calculate(DefaultConditional.java:322)
	at jmri.implementation.JmriSimplePropertyListener.propertyChange(JmriSimplePropertyListener.java:115)
	at jmri.implementation.JmriTwoStatePropertyListener.propertyChange(JmriTwoStatePropertyListener.java:47)
	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:327)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
	at jmri.implementation.AbstractNamedBean.firePropertyChange(AbstractNamedBean.java:227)
	at jmri.implementation.AbstractSensor.setOwnState(AbstractSensor.java:207)
	at jmri.jmrix.loconet.LnSensor.message(LnSensor.java:111)
	at jmri.jmrix.loconet.LnTrafficController.notify(LnTrafficController.java:98)
	at jmri.jmrix.loconet.LnPacketizer$RcvMemo.run(LnPacketizer.java:351)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

Many thanks for your efforts with JMRI.

David Matthews Worcester, UK

klk32003, 2018-03-14:
Here is an example, where a RasPi user initialized JMRI Audio successfully:
https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/145616

Please see the command ```$ sudo apt-get install libopenal-dev``` in the posting or [here](http://jmri.org/help/en/html/tools/Audio.shtml).

I hope this helps.

Klaus Killinger

bunnshill, 2018-03-15:
Klaus 

Many thanks for your suggestions.  libopenal solves the warning and error messages but I still can't get the sounds to work as I want yet using the Raspberry Pi. It works on a Mac.

I am not trying to move sound around the layout.  I am just trying to get sounds to play when  any loco passes through a grade crossing, or when there is a short, or when the signals have cleared and the loco is waiting.

Previously without libopenal I got these sound to play once through the various logix only the first time the logix was called but not subsequently.

David Matthews

klk32003, 2018-03-15:
David,

Is ```AudioExample.py``` working for you?

Klaus

klk32003, 2018-03-15:
Could be a path or format problem. Please check the path of your WAV file.

Klaus

bunnshill, 2018-03-25:
Yes.  At present I have no solution.  I can get sounds on the Raspberry Pi to play once and only once for each logic.

klk32003, 2018-03-25:
There are numerous hits on the web due to ```java.lang.IllegalArgumentException: No line matching interface Clip supporting format PCM_SIGNED unknown sample rate```.  It is the combination Linux and Javasound which causes some trouble. Hence I tried to suggest to use OpenAL instead of Javasound.

Of course this would require that the AudioBuffer is loading correctly.

bunnshill, 2018-04-08:
Klaus

Many thanks for you suggestions about OpenAl.  I have now got this work with a Raspberry Pi  with a buffer and source, but I cannot save it as the saved filed produces an audio parse error both on my Pi and an IMac. I have reported this as an error. 

The normal method of playing a wav file through logic still works once only on the Pi but is OK on my IMac.

klk32003, 2018-04-09:
Thank you for the response, David.

The underlying schema for the parsing has been changed 13 days ago.  So, an older version might be in place on your Raspberry Pi.

Klaus

bunnshill, 2018-04-11:
Klaus

I can now run the saved file by ignoring the parse error so I will close this issue.   Many thanks.

David Matthews

closed 2018-04-11 by bunnshill