JMRI/JMRI #5088: DCC++ lowest valid turnout number?

bobjacobsen, 2018-03-14:
A recent jmriusers discussion indicated that "0" (zero) is a valid DCC++ turnout index, i.e. with the usual defaults the system name DCCPPT0 is valid.  "1" and hence DCCPPT1 is acceptable to the code, but there's an explicit test that failed the "0" value at line 173 of `jmrix.dccpp.DCCppTurnoutManager`.

Should zero be valid?

KenC57, 2018-03-14:
I had only checked the system part of our website for DCC++ to see what it
said. I was slightly surprised it said zero is valid. I know a number of
other systems use a binary of zero but call it one. I have no idea what is
really true of the DCC++ stuff.
-ken c

pabender, 2018-03-14:
Sent from my iPad

> On Mar 13, 2018, at 8:23 PM, Bob Jacobsen <notifications@github.com> wrote:
> 
> A recent jmriusers discussion indicated that "0" (zero) is a valid DCC++ turnout index, i.e. with the usual defaults the system name DCCPPT0 is valid. "1" and hence DCCPPT1 is acceptable to the code, but there's an explicit test that failed the "0" value at line 173 of jmrix.dccpp.DCCppTurnoutManager.
> 
> Should zero be valid?
> 

I think the answer actually may be yes.

Two of the three output types DCC++ supports ( using the Io pins and stored DCC addresses ) are mapping’s in arrays stored in the arduino’s EEPROM. 0 is a valid table index.

 For the other output type, which sends accessory decoder commands to non-stored addresses, 0 is not valid.

Btw, sensors are also mapped in the EEPROM to IO pins.

Paul

PR #5162 (2018-03-27): Turnout number zero is OK in DCC++
- allow turnout T0 in DCC++
- add some tests to DCC++ Turnout Manager

Closes #5088
closed 2018-03-27 by bobjacobsen