JMRI/JMRI #5094: Layout Panel Memory Variable Center Justification Lost In Web Server after using [Home]

ModelTrainNut, 2018-03-14:
Center Justified Memory Variables display correctly on Layout Panels in Web Server initially and when using [List] to navigate between panels, but they display Left Justified after navigating using [Home].
Java 1.8.0_151, PanelPro 4.11.3, Win10 64 server, WiThrottle Full Version App on a newer iPad.

Randy

mstevetodd, 2018-03-14:
@ModelTrainNut please attach a panel file showing this problem.

ModelTrainNut, 2018-03-14:
[image]

Github won't let me attach the original .xml - is this what you wanted to see? Randy

ModelTrainNut, 2018-03-14:
[Ontario Southern Railway.zip](https://github.com/JMRI/JMRI/files/1811805/Ontario.Southern.Railway.zip)

Duh...wish I'd though of that...

mstevetodd, 2018-03-15:
I'm able to load your panel file (with a few missing icon messages), but I'm unable to open any of your panels via the web server. They show in the thumbnail list, but when I click the link, I get the attached message popup. This occurs with the latest dev build, and with 4.8. I get a WARN in the session.log file that says the same thing, no helpful detail. 
My panel loads fine in the web browser.
@ModelTrainNut are you able to open your panel in a regular web browser outside of the WiThrottle client? Chrome, FireFox, etc.
I'd also like to know if anyone else can view his panels via their JMRI webserver.
I'll look at it again when I'm over my current travel lag. Maybe I'm doing something dumb.
[image]

ModelTrainNut, 2018-03-15:
Will check when home tomorrow evening latest. Thanks! Randy

ModelTrainNut, 2018-03-15:
The problem is the same when I go directly to 192.168.0.199:12080 in my iPad Safari browser. The static pictures in the "My JMRI Railroad" link page (with Throttles and Open Windows and Utilities and File Access sections) show the Memories centered properly, but they shift to the right when the individual panels are opened.
Randy

mstevetodd, 2018-03-15:
Thanks, Randy.
Can you try it on something that's not Apple? I'm trying to figure out why I can't open the panel at all in my browsers. I can't look at the alignment problem if I can't open the page.
A long-shot, but can you post or email me your custom icons? Maybe that error is preventing the page from loading for me.
On the alignment issue, I made a note in the code that 
`alignment of memoryIcons without fixed width is very different. Recommend using fixed width.`
Can you change your panel to use a fixed width for those items? Or at least try one or two. That may provide a valid workaround.

ModelTrainNut, 2018-03-21:
I apologize for the delay - my Prusa 3d printer arrived and just HAD to be
assembled and tested right away, and of course I was scheduled to work long
shifts all weekend, and my wife got free tickets to a Beatles tribute show
in Toronto, and a dog ate my homework...
I just tested my panels on both Chrome and Microsoft Edge browsers on two
different Windows 64 PCs, and Chrome on my iPhone 7, and they all
demonstrate the same failure to center Memory Variables.
Randy

ModelTrainNut, 2018-03-21:
> Randy,
> If you have the time, it looks like you are slightly behind on Java (_161
> is current, you have _151). Also we've released 4.11.4, which came out over
> the weekend. I suspect this won't make a difference, but confirming that
> gives us two more things ruled out.
> -ken c

Will do.

mstevetodd, 2018-03-22:
@ModelTrainNut, did you try using the fixed width as I suggested? That should provide an immediate fix to your issue.

ModelTrainNut, 2018-03-22:
I just tried all your fixes on my day off, in this order:

- using a fixed width (and height) DOES fix the problem (yay!) on Windows
Edge and my iPhone 7
-
- updating to 4.11.4 DID NOT fix it, and now I am getting a recurring
'Requested Type 'ping' unknown' warnings and 'Blocking message pending
10000' python error dump and a different one referencing
'EatWhatYouKill.java'!? - these are probably just transients because I have
not rebooted; I'll open new issues if they persist because they are not
directly related to the centering issue
-
- updating to java 1.81 DID NOT fix the centering issue on any devices
-
Thank you so much for your patience on this, and for finding a simple
workaround. I can have my pages fixed up within an hour - just don't know
when that hour will happen!
Randy

mstevetodd, 2018-03-22:
@ModelTrainNut, 
Thanks for changing to the fixed width.
The 'ping' issue has been addressed, and will be in 4.11.5. It's in the Jenkins builds now.

ModelTrainNut, 2018-03-23:
Great! Thanks again. Randy

mstevetodd, 2021-03-30:
Looks like these issues were addressed or acceptable workarounds provided. Closing.

closed 2021-03-30 by mstevetodd