JMRI/JMRI #5104: LE based layout :  java.util.ConcurrentModificationException

dsand47, 2018-03-16:
Large LE based layouts with multiple panels are susceptible to the exception.

The error occurs in LayoutBlock at getNextPacketID().

```
        if (updateReferences.size() > 500) {
//             log.info("flush update references, size = {}", updateReferences.size());
            try {
>>>                updateReferences.subList(0, 250).clear();
            } catch (java.util.ConcurrentModificationException | java.lang.IndexOutOfBoundsException ex) {
                log.warn("getNextPacketID: flush update references exception: {} -- ", ex.toString(), ex);
            }
        }
```
When **updateReferences** gets above 500 rows it is trimmed back to 250.  After panel loading is done and the initial block processing completes, there are at least 3 threads making changes.  The layout has 4 panels, but only three are functional.  I suspect that there is a thread for each functional panel.

I added the log and try/catch statements to diagnose the problem.

The affected variable:
`    private static List<Integer> updateReferences = new ArrayList<>(500);
`

The commented log statement changes the timing just enough so that an index out bounds exception occurs instead of the modification exception. 

If the full package would still be useful, let me know.

Dave Sand

[session.log](https://github.com/JMRI/JMRI/files/1820591/session.log)

bobjacobsen, 2018-03-17:
A full .zip of the profile, sorry for the trouble.  

LayoutBlock is using a method of "tracking updates" to "avoid going into an update loop" that's going to be very hard to make properly thread-safe.  In that case, the two cases are to create some really good tests, and then try to make it thread-safe, or to reimplement it in a way that doesn't have the same issues. Not sure which one makes the most sense.  Having an entire profile, so I can get it to happen, even if only occasionally, would help a lot.

dsand47, 2018-03-17:
I have placed two zip files in my Dropbox account.  One is quite large.

- Profile directory &mdash; https://www.dropbox.com/s/0d3pyb4r3db8u6c/PeteB__Sim.zip?dl=0
- Preference directory &mdash; https://www.dropbox.com/s/gn7x0jo1mm0z2qg/PeteB.zip?dl=0

As a bonus, the **session.log** file is in the preference directory with a **_double_** **ConcurrentModificationException**.  This is the first one that I have seen.

bobjacobsen, 2020-07-25:
There's a fix for this and a test case derived from .zip files in the layout refactoring branch. Unfortunately (?) the fix takes advantage of some of the refactoring, so it's not really possible to back-port it.

closed 2021-04-14 by stale[bot]