JMRI/JMRI #5111: Create Test Release 4.11.5

bobjacobsen, 2018-03-18:
This is the fifth release of the 4.12 cycle. It's intended to be released around April 11th from HEAD of master.

bobjacobsen, 2018-04-14:
The release-4.11.5 branch has been created. 

From now on, please document your changes in the [jmri4.11.6.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.11.6.shtml) release note file.

Maintainers, please set the 4.11.6 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [CI server](http://builds.jmri.org/jenkins/job/TestReleases/job/4.11.5/)

silverailscolo, 2018-04-15:
Thanks @bobjacobsen. Downloads and installs fine on macOS High Sierra and Win7.

bobjacobsen, 2018-05-13:
jvisualvm summary of `ant alltest` run on final release:
[image]

closed 2018-05-16 by bobjacobsen