JMRI/JMRI #5114: RetryRule and Assume?

bobjacobsen, 2018-03-19:
The `jmri.jmrit.display.IconEditorWindowTest` test class defines a retry rule:
```
    @Rule
    public Timeout globalTimeout = Timeout.seconds(10); // 10 second timeout for methods in this test class.
    @Rule
    public RetryRule retryRule = new RetryRule(3);  // allow 3 retries
```

and then has some graphical tests that start with:
```
    @Test
    public void testSensorEditor() throws Exception {
        Assume.assumeFalse(GraphicsEnvironment.isHeadless());
        ...
```

These tests shouldn't be running when I do `ant headlesstest`, but the log shows that they're considered to be timing out:
```
ERROR - testSensorEditor(jmri.jmrit.display.IconEditorWindowTest) : giving up after 3 failures [main] util.junit.rules.RetryRule.evaluate()
```
(and six others, which together take a bit of time) Any ideas on what's going wrong here?

pabender, 2018-03-19:
Hmmm.. let me look into that.

The Travis build is currently using Maven for the tests, and I don’t recall seeing that there. (But I have not looked either.)

Paul

bobjacobsen, 2018-03-26:
This looks like it's resolved.  Can we close?

pabender, 2018-03-26:
I think so.

closed 2018-03-26 by pabender