JMRI/JMRI #5128: Byte Write (no feedback) on Main Track does Byte read?

ClockLoop, 2018-03-21:
Why is this happening when I choose to do ops mode writes?

I understand that verify reads happen when using the programming track, and I want this, but why is it happening when doing main track operations?

Is ops mode or main track byte reads, possible?  With any hardware?

I have a digitrax dcs240 and transponding decoders, and when I do main track programming the byte read always says that the command was accepted blind (no response will be sent)

Why does jmri do **reads** in ops or main track programming mode?

pabender, 2018-03-21:
Sent from my iPad

> On Mar 21, 2018, at 5:50 PM, ClockLoop <notifications@github.com> wrote:
> Is ops mode or main track byte reads, possible? With any hardware?
> 
Yes.  Loconet systems with transponding enabled can do reads in operations mode, as can systems with railcom installed.
> Why does jmri do reads in ops or main track programming mode?
> 
Because certain hardware configurations can do reads in ops mode and we have no idea if the hardware attached supports reads or not.  This is true even in the case when we know appropriate hardware is installed on the layout because not all decoders support the technology.

Paul

ClockLoop, 2018-03-21:
Oh, I didn't realize hardware can do reads in ops mode.

I wonder what im doing wrong.

bobjacobsen, 2018-03-21:
The last time we discussed this, there was an 80/20 consensus around leaving it at "assume ops can read by default".  The issue is around non-decoder devices, e.g. boards, that can reply, not so much around transponding.

Personally, although I don't like excessive options, I think having a by-connection "ops reads OK?" option in the connection preferences is the right solution, with (if needed) a default to "yes"

ClockLoop, 2018-03-21:
Not gunna work.  No idea why.   DCS240 with a dxp88 with a ps2012 power supply, DT500 throttle.
I have multiple dz123 decoders with transponding on, and multiple sdxn136ps decoders with transponding ON, and the readback fails 100%.   From what I am seeing the bxp88 has a issue with zone8 and detecting occupied when the power comes on, the leds on the bxp88 show transponding and occupancy for that zone, but the loconet and jmri don't report any transponding or occupied zone.   

Multiple issues with the bxp88 still, with latest firmware, and so I cannot verify if my issues are with the bxp88, i don't have any of the other digitrax transponder receivers to test with.

Now that I know this is probably a bxp88 problem, again, I will stop here, since I have already discovered that all 4 of my bxp88s are lemons, since they all do this.  I should have known this issue here was related
to my earlier issues.

For my situation, it would be nice to turn byte reads in ops mode to OFF, since it dosen't work even when I have all the proper stuff, and just adds time to the ops, operations.

Pugwash1, 2018-03-21:
Both of those decoders dont have built in resistors across the blue and white leads. If they have an led attched to the blue white and no resistor they may work, but only in one orientation. Ive never got a bulb to work unless it has a resistor in series.

On March 21, 2018 6:36:17 PM EDT, ClockLoop <notifications@github.com> wrote:
> Not gunna work.  No idea why.   DCS240 with a dxp88 with a ps2012 power
> supply, DT500 throttle.
> I have multiple dz123 decoders with transponding on, and multiple
> sdxn136ps decoders with transponding [...]

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

ClockLoop, 2018-03-21:
NO resistor with leds?     This is entering the funny farm zone.

Lets see 12v pwm to led, voltage drop of 3v or so.. means 9v of heat dissipation.   
Who wants a burnt out led?  Even with pwm, that junction is gonna fry.

I don't care, since I use leds that are high output anyway, the worst i will see is the led will desolder its self, or melt the engines plastic shell.

I will give it a try, led with no resistor.   65$ for a decoder, oh well, BURN BABY BURN!
Can't have reward with no risk of loss.

n3ix, 2018-03-21:
Guess this may be obvious but most/all Loconet stationary decoders can be read using Ops Mode.

 

Robin Becker

San Diego, CA

 

http://drgw.n3ix.com <http://n3ix.com/pers/drgw.html>

ClockLoop, 2018-03-22:
I have fried a sdxn136ps forward headlight output by connecting it to a led with no resistor.  The output no longer works, after swapping another sdxn136ps decoder in place, to verify wiring.

Its official, hook up a led with no resistor, and your decoder output will fry.

With a 100ohm resistor the led conducts 50ma of power, without 100ohm, decoder output dies.

Pugwash1, 2018-03-22:
The additional resistor goes between the blue and white leads. Yes the led has the current resistor in series with the led if the led or light board doesnt have a built in resistor, across the blue and white leads. See the Bxp88 instructions pg 13 sec 11.2

On March 21, 2018 8:11:21 PM EDT, ClockLoop <notifications@github.com> wrote:
> I have fried a sdxn136ps forward headlight output by connecting it to a
> led with no resistor.  The output no longer works, after swapping
> another sdxn136ps decoder in place, to verify wiring.
>
> Its [...]

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

ClockLoop, 2018-03-22:
I tried a 300ohm between blue and white leads, with my led also using 470ohm between blue and white leads.  (i never liked using 100 ohm, 50ma is a bit much for such a small decoder enclosed in the loco, plus 50ma thru a led is bad.. (unless you use the leds i use, that are rated for 500ma)

Transponding is working better now, but ops mode readback still gives me the same error.

I'll try putting this on all decoders.

Sure the bxp88 manual may say that, but the decoder manual says different.
I think they may need a bit of documentation overhaul.

[image]

Then different info in the decoder manual.
[image]
[image]

Thanks for the help, i will try putting the resistor on more decoders... and test them to see if they mix up, Or if addresses that are selected in the throttle but not on the layout still show up as located in the zone where a transponding loco is located, but hasn't been selected in the throttle.

ClockLoop, 2018-03-22:
Now when transponding is enabled amd the 300 ohm resistor with 470ohm with led, I cannot read or write cv's, unless i turn transponding off.   (it gives me read errors)

devel-bobm, 2018-03-22:
It seems that the JMRI portion of this issue is "closable".  Since "Ops mode readback" is usable on some mobile decoders with appropriate hardware, and is usable on some LocoNet-based stationary decoders, no JMRI change is appropriate.

The portion of this "issue" that is "transponding hardware debug" is not JMRI and therefore beyond the scope of an "issue" for the JMRI project.  How about moving this part of the issue to a topic-appropriate forum?

ClockLoop, 2018-03-22:
I would say that having two options would help me here.

1. turning off transponding on the programming track since it interferes.

2. allow users to NOT to readback in ops mode.

This thread was more to get info about the proper behavitor of reading CVs,

And I DO have the appropriate hardware to do ops mode cv reading.

devel-bobm, 2018-03-22:
@clockLoop, in regards to your initial question:
> Why is this happening when I choose to do ops mode writes?
> 
> I understand that verify reads happen when using the programming track, and I want this, but why is it happening when doing main track operations?

In JMRI "Preferences", under "Roster", on the "Programmer" tab, do you have the item 
`When possible, confirm CV writes by following them with a read`
checked?  If so, try unchecking it, and "read after write" ought to disappear.

bobjacobsen, 2018-03-22:
I opened #5130 in case people have ideas about how to better configure the read capability.  No hurry, just in case somebody has time to think through the various cases.

ClockLoop, 2018-03-22:
@devel-bobm 

> In JMRI "Preferences", under "Roster", on the "Programmer" tab, do you have the item
When possible, confirm CV writes by following them with a read
checked? If so, try unchecking it, and "read after write" ought to disappear.

I do both ops and programming track work within minutes of eachother, changing that option and restarting jmri, every few minutes...    
I guess i will just have to suffer through the read errors, because im not doing that.

bobjacobsen, 2018-03-26:
@ClockLoop I think there's a targeted fix for this in #5152 Are you able to build and test it?  If so, just set the "transponding" option to "no" and see if it behaves the way you want. If you're not set up to build and test, no problem, will just merge and we can sort that out later.

ClockLoop, 2018-03-28:
@bobjacobsen 

Another excellent fix verified.  Thanks.
I wish the planet was open source..   Then perhaps my garbage hardware would be worth what I paid, AND actually work.. or at least be fixable...
[image]

ClockLoop, 2018-03-29:
@bobjacobsen 

I did a full page write in "programming on main" and the speed at which the sequential writes are made, might be a bit TOO FAST for decoders? (theres no delay what so ever between writes)

The data seems to be getting through, but I am using pretty modern decoders.

Is it possible that a delay might be needed for some decoders?
Or is the max speed the com link goes at, ok for decoders?

I do a full page write using service mode(programming track) and it writes at different speeds.
(the verify read is OFF in both tests)

Matching the DELAY that service mode(programming track) has between writes, might be best?

Heres a video of a full page write, you can see its super fast.
The first write page, is ops mode, the second slower writes are service mode.
https://www.youtube.com/watch?v=Kr888LtpZr4

dheap, 2018-03-30:
> @ClockLoop Is it possible that a delay might be needed for some decoders?
> Or is the max speed the com link goes at, ok for decoders?

@ClockLoop This facility is already available (on a per-decoder basis) for both:
Ops-Mode Loco:
http://jmri.org/help/en/html/apps/DecoderPro/CreateDecoderAdvanced.shtml#opsdelayed
and Ops-Mode Accessory/Signal:
http://jmri.org/help/en/html/apps/DecoderPro/CreateDecoderAdvanced.shtml#accessoryops

closed 2018-04-11 by bobjacobsen