JMRI/JMRI #5129: Move IPv4 and IPv6 flags into "regular" Preferences?

mstevetodd, 2018-03-22:
I've been investigating the zeroconf broadcast failures, and have seen a marked improvement if I turn IPv6 off. There is a script, zeroconf-preferences.js, that allows the user to disable IPv6 (or IPv4). That script works fine, but seems out of the norm for setting a Preference.
Should we move those flags into the Edit, Preferences window somewhere? If so, where?
Also, should we default IPv6 to OFF? EngineDriver does not have any support for IPv6. I'm not sure about the WiThrottle client.
I'd be interested in others' thoughts on this.

pabender, 2018-03-22:
On 03/21/2018 10:38 PM, SteveT wrote:
>
> I've been investigating the zeroconf broadcast failures, and have seen 
> a marked improvement if I turn IPv6 off. There is a script, 
> zeroconf-preferences.js, that allows the user to disable IPv6 (or 
> IPv4). That script works fine, but seems out of the norm for setting a 
> Preference.
>

Could you elaborate which zeroconf broadcast failures you are talking 
about?

Did I miss a conversation on this somewhere?

Do we have wireshark traces that that show we are not broadcasting the 
information?

> Should we move those flags into the Edit, Preferences window somewhere?
>

I think it could be added as a regular preference, but that seems like 
an unnecessary option to me.

We really should make the broadcast work correctly on both IPv4 and IPv6.

> If so, where?
> Also, should we default IPv6 to OFF? EngineDriver does not have any 
> support for IPv6. I'm not sure about the WiThrottle client.
>

I believe WiThrottle client will use an IPV6 address if it is there, 
though I haven't verified that.

But also remember that all of our other services that advertise via 
JmDNS use the same JmDNS instance (It's a singleton in the library), and 
at least some of those services (such as the web server) can work just 
fine on IPv6 regardless of what operating system is in use.

Paul

mstevetodd, 2018-03-22:
@pabender, I'm mostly going on my experience with the latest builds, and reports from EngineDriver users, but I've also seen several reports on the users group, and issue #4973. 

One complication is that there are a myriad of network configurations out there, so we need a wide range of user feedback. I'd like to be able to easily ask people reporting the problem to disable IPv6 and see if that helps or not, hence this question/issue.

pabender, 2018-03-22:
Sent from my iPad

> On Mar 22, 2018, at 9:01 AM, SteveT <notifications@github.com> wrote:
> 
> @pabender, I'm mostly going on my experience with the latest builds, and reports from EngineDriver users, but I've also seen several reports on the users group, and issue #4973.
> 
> One complication is that there are a myriad of network configurations out there, so we need a wide range of user feedback. I'd like to be able to easily ask people reporting the problem to disable IPv6 and see if that helps or not, hence this question/issue.
> 
This is actually what I am driving at as well.  

If we can show that JMRI does the right thing ( I.e. using wireshark, we show that the advertisements occur ) then any other factors are networking issues beyond our control.

Paul

mstevetodd, 2018-03-22:
@pabender, in my world, explaining to the user "why" something doesn't work isn't very helpful. They prefer things to work. I prefer to look for ways to make things work in less-than-perfect environments.

The current broadcast isn't working in multiple environments. It worked much more reliably at 4.8.

I think being able to more easily toggle the IPv6 flag might prove useful in some of these environments, and may provide useful debugging feedback. 
You feel it is unnecessary.

Anyone else have an opinion on adding the IPv4 and IPv6 flags to Preferences somewhere? Maybe on both the WiThrottle and Web Server Preference pages?

KenC57, 2018-03-22:
Given the number of environments where we have users isolated from the
internet using older wi-fi routers, I think being able to disable IP6 is
useful. While the routers may support it, they may not do a good job of it.
Same for some of the computers, their support may be less than perfect.
While I find it amazing that IP6 didn't grow quicker, I've been doing
networks since the Ethernet I standard was being replaced by Ethernet II and
TCP was the neat cool thing. IP6 has been on the books for 20+ years but
adoption is still slow. I've seen a lot of times with many projects where
turning off IP6 made things work again.
-ken c

pabender, 2018-03-22:
Sent from my iPad

> On Mar 22, 2018, at 11:31 AM, SteveT <notifications@github.com> wrote:
> @pabender, in my world, explaining to the user "why" something doesn't work isn't very helpful. They prefer things to work. I prefer to look for ways to make things work in less-than-perfect environments.
> 
Understanding why is the only way to actually fix the problem however.
> The current broadcast isn't working in multiple environments. It worked much more reliably at 4.8.
> 
Which environments.  Be specific.  
> I think being able to more easily toggle the IPv6 flag might prove useful in some of these environments, and may provide useful debugging feedback.
> You feel it is unnecessary.
> 

I do feel it is unnecessary, but I feel that way because you are addressing the symptom and not the cause by turning off the broadcasts.  In other words, you are only hiding the problem, not fixing it.

Get me traces from wireshark that show what JMRI actually broadcasts on the wire and from there we can figure out how to make it broadcast the right thing.
> Anyone else have an opinion on adding the IPv4 and IPv6 flags to Preferences somewhere? Maybe on both the WiThrottle and Web Server Preference pages?
> 

I will look at this more in depth, but we may not be able to turn this off for individual services.

PAB

bobjacobsen, 2018-03-22:
> In other words, you are only hiding the problem, not fixing it.

If the problem is no longer visible to JMRI users, isn’t that sufficient?  I think I’m missing something in the reasoning here.

BHoffman351, 2018-03-22:
> On Mar 22, 2018, at 1:54 PM, Ken Cameron <notifications@github.com> wrote:
> 
> At least with the startup actions and being able to adjust what runs in
> which order in the startup, a script is easy enough for the user to add. So
> if we had two scripts, one to disable each, they could add it as a step
> before the step starting the servers. It wasn't always that way, but is
> pretty clean to use now.

Yes, but that script should never be set to automatically run as it changes a non-exposed setting in JMRI.  It is a one-and-done.  I agree that it is not a difficult action.

Regards,
Brett

BHoffman351, 2018-03-22:
> On Mar 22, 2018, at 1:54 PM, Ken Cameron <notifications@github.com <mailto:notifications@github.com>> wrote:
> 
> At least with the startup actions and being able to adjust what runs in
> which order in the startup, a script is easy enough for the user to add. So
> if we had two scripts, one to disable each, they could add it as a step
> before the step starting the servers. It wasn't always that way, but is
> pretty clean to use now.

Yes, but that script should never be set to automatically run as it changes a non-exposed setting in JMRI.  It is a one-and-done.  I agree that it is not a difficult action.

Regards,
Brett

pabender, 2018-03-22:
Sent from my iPad

> On Mar 22, 2018, at 12:30 PM, Bob Jacobsen <notifications@github.com> wrote:
> 
> > In other words, you are only hiding the problem, not fixing it.
> 
> If the problem is no longer visible to JMRI users, isn’t that sufficient? I think I’m missing something in the reasoning here.

My concern is does this function in working mixed IPv4/IPv6 environments.

Paul

rhwood, 2018-03-25:
The two largest residential ISPs require that (at least new) routers connected to their networks support IPv6, since they can’t give out additional IPv4 addresses in their networks (that’s Comcast and Verizon). I understand that I outside the US, other ISPs have been doing this for years. Apple requires that iOS apps support IPv6, and automatically preferres an IPv6 route if one exists; I think that masking, hiding, or disabling IPv6 by default will quickly put us in a situation where users may be complaining that WiThrottle or other JMRI services are not working *because* IPv6 is disabled, so we should be leaving both enabled if the underlying OS allows it, and allow users to disable one or the other if it causes problems.

Perhaps the real fix to most of these problems that Engine Driver users are encountering might be to allow Engine Driver to use IPv6 networking if the Android device its running on allows it.

mstevetodd, 2018-03-28:
PR #5173 adds these flags to the WiThrottle Preference's Network pane.
[image]

closed 2018-04-18 by mstevetodd