JMRI/JMRI #5130: How to control ops-mode read option?

bobjacobsen, 2018-03-22:
(Successor to #5128)

Sometimes ops-mode read works:
- Digitrax decoders with transponding installed, connected, etc
- Some LocoNet boards with a LocoNet connection
- NMRA advanced ack with some hardware combination (not sure of details, sorry)

Other times it doesn't.  In particular, some decoders can work or not work depending on other hardware being installed; some boards that might work won't because cables haven't been run.

Yet it's annoying to users to have it look like ops-mode read is available when it's not.  It would be good to have (a set of) options that let's the user specify the situation.

But how to do that?  The available tools are:
- specifications in decoder definitions - we have transpRead to indicate "read with transponding", we could add others as needed - `opsReadWorksIfWiredToLocoNet`?
- perhaps a by-connection preference in the "Advanced Options" section that says "transponding present and/or all boards connected to LocoNet, or if not actually connected you should assume it is" which seems like a rather long description
- Or maybe a different command station type instead?

First, we have to figure out _what_ to do, then we figure out how to code it.

pabender, 2018-03-22:
Sent from my iPad

> On Mar 22, 2018, at 12:34 AM, Bob Jacobsen <notifications@github.com> wrote:
> 
> (Successor to #5128)
> 
> Sometimes ops-mode read works:
> 
> Digitrax decoders with transponding installed, connected, etc
> Some LocoNet boards with a LocoNet connection
> NMRA advanced ack with some hardware combination (not sure of details, sorry)

The required hardware for railcom is:

1) A railcom cutout device ( some boosters have this option available )
2) A railcom detector

We support some systems that have both of these built in ( such as the Roco Z21 ).

For Loconet systems, this device provides the detection separately.
https://www.digikeijs.com/dr5088rc-digidetect.html

There is an interesting twist to this.  Lenz sells a device that provides a local railcom display.  It can show CV values when you read them, but it doesn’t send them on to any other device. ( for a long time, this was the only railcom detector available... )

ClockLoop, 2018-03-22:
@bobjacobsen 

> perhaps a by-connection preference in the "Advanced Options" section that says "transponding present and/or all boards connected to LocoNet, or if not actually connected you should assume it is" which seems like a rather long description

The connection preference lets you choose the pr3 under digitrax, even though the dcs240 is not a pr3, and its firmware is not updatable by pr3 firmware so its technically different, but I understand the communcation protocol etc is the same so this is why a dcs240 user must select pr3 as the device.

So in that same thought pattern, you choose the dcs240 as a sub option once you have selected the pr3.
Is it possible to have another drop down box that lets you add other equiptment that is known to be compatible with the current selected connection and command station?

Probably just make things too hard on programming part.

Why not just add yet another check box.
[image]

ClockLoop, 2018-03-22:
And this one will save tons of people time while doing their cv reads and writes on the programming track.
[image]

pabender, 2018-03-22:
Sent from my iPhone

> On Mar 22, 2018, at 5:00 PM, ClockLoop <notifications@github.com> wrote:
> 
> And this one will save tons of people time while doing their cv reads and writes on the programming track.
> 
I don't understand what transponding has to do with the programming track.  

They are not related.

ClockLoop, 2018-03-22:
When transponding is ON, and my loco is on the programming track, the reads and writes fail 75% of the time.

When transponding is OFF, and my loco is on the programming track, the reads and writes succeed 100%.

devel-bobm, 2018-03-22:
@ClockLoop, please try to communicate more clearly.  I have had a very hard time following some of your posts...

Please define exactly what you mean by "When transponding is ON" and "When transponding is OFF".  There is NOT a single "transponding on/off" switch, and we need to exactly which one you are talking about.

ClockLoop, 2018-03-22:
Ok, I will try again differently.

When I have a sdxn136ps decoder or a dz123 decoder properly connected to the programming track connectors on a dcs240.

If I make CV61 have a value of 2  (transponding ON) and my loco is on the programming track, the cv reads and writes fail 75% of the time.

If I make CV61 have a value of 0  (transponding OFF) and my loco is on the programming track, the cv reads and writes fail 75% of the time.

pabender, 2018-03-22:
Sent from my iPad

> On Mar 22, 2018, at 5:36 PM, ClockLoop <notifications@github.com> wrote:
> 
> When I have a sdxn136ps decoder or a dz123 decoder properly connected to the programming track connectors on a dcs240.
> 
> If I make CV61 have a value of 2 (transponding ON) and my loco is on the programming track, the cv reads and writes fail 75% of the time.
> 
> If I make CV61 have a value of 0 (transponding OFF) and my loco is on the programming track, the cv reads and writes fail 75% of the time.
> 

The options we are discussing here are on the system level.  They have nothing to do with the CV settings of an individual decoder, which is what you describe above.

In other words, the option you are asking for isn’t going to solve the problem.

If you do not see the same behavior using a Digitrax throttle, there could be a timing issue in our programming sequence, but I would think that would impact other Digitrax systems as well.

ClockLoop, 2018-03-22:
@pabender 

> The options we are discussing here are on the system level. They have nothing to do with the CV settings of an individual decoder, which is what you describe above.

Then problem I did, was I added something to this thread that was unrelated to the topic of the thread, perhaps i should make a new issue thread, but i think others would need to confirm this activity,..  i get alot of face time with this hardware, but im only one persons opinion, and one who cant code in your language, so i can't operate on your level, or communicate on it.   

If I could, id just modify my copy and be done with it.

devel-bobm, 2018-03-22:
@bobjacobsen,

"Yet it's annoying to users to have it look like ops-mode read is available when it's not. It would be good to have (a set of) options that let's the user specify the situation.

But how to do that? The available tools are:

- specifications in decoder definitions - we have transpRead to indicate "read with transponding", we could add others as needed - opsReadWorksIfWiredToLocoNet?
- perhaps a by-connection preference in the "Advanced Options" section that says "transponding present and/or all boards connected to LocoNet, or if not actually connected you should assume it is" which seems like a rather long description
- Or maybe a different command station type instead?"

My comments:

With respect to "Specification in decoder defintion":  For one class of devices, it makes a great deal of sense for the decoder definintion to categorically state that "ops mode programming reads" will always work.  For LocoNet-based connections, that is for devices which MUST be connected via LocoNet and which CANNOT be programmed on the service-mode programming track and which CANNOT be programmed via the DCC track signal in Ops Mode.  This applies to several devices from RR-CirKits and some from Tam Valley Depot.  This is not necessarily a complete list of LocoNet devices to which this applies.

But "Specification in decoder definition" does not make sense to me for the typical "mobile" decoder which has "Ops Mode read" capabilities.  The typical JMRI user does not have the appropriate hardware for "readback", but may well have mobile decoders with two-way capabilities.  And even if he does have the appropriate hardware, there is no guarantee that the mobile decoder is within a track block which can support the "readback".  The decoder definition itself cannot predict that "all the starts will align" to conclusively and correctly allow "Ops Mode Read".

With respect to "defining a new command station type"  To me, defining a new command station type doesn't make any sense.  The feature is not necessarily command-station-dependant; it is more readback-hardware-dependant and programmable-device-readback-capable-dependant and programmable-device-connected-in-readable-location-dependant.  The readback-hardware-dependant part of the equation may or may not relate to command station capabilities.

In my opinion, the ability of a system to perform ops-mode read operations in a given situation can be rather dynamic, and should not be part of a connection preference.  With that in mind, I'd like to offer a couple of ideas:

1) Give the user a role in konwing whether "ops-mode reads" will work, as part of the run-time JMRI programming tool.  If the decoder definition says that the device can support "ops-mode readback", and the JMRI "programmer" tool is configured to use "ops-mode programming", the JMRI programmer could provide a user-configurable checkbox to enable or disable "ops-mode programming reads".  The "read" buttons would be disabled if the user specifies that "ops-mode reads" won't work.  If the decoder definition doesn't specify "ops-mode readback", or if the decoder definition specifically declines "ops-mode readback",  then "read" buttons could be disabled in programmers when using "ops-mode".  Else the "read" buttons would be enabled.  If the readback hardware isn't present or if the device is not currently in a situation where an ops-mode read can complete successfully, and a read is tried, then JMRI behaves exactly as it does currently - a delay waiting for the reply then an error message.

2)  If we want to be a bit more aggressive (from a programming standpoint), we could set up the JMRI "programmer" tool to infer the ability to perform "ops-mode reads".  The first user "ops-mode read" from or "ops-mode write" to the device of a "programming session" could enable/disable the "ops-mode read" capability for the remainder of the session.  If the first operation is an "ops-mode read", then if it gets a positive reply from the device, then allow "ops-mode reads" for the remainder of the session.  If the first operation is an "ops-mode read" and the read fails, then disallow "ops-mode reads" for the remainder of the session.  If the first operation is an "ops-mode write", then after a successful response to the write operation, immediately perform a "ops-mode read" operation to see if the read can be completed successfully, and allow/disallow future "ops-mode read" operations during the session accordingly.

ClockLoop, 2018-03-22:
@pabender 

> If you do not see the same behavior using a Digitrax throttle, there could be a timing issue in our programming sequence, but I would think that would impact other Digitrax systems as well.

Hmm, i will try doing reads and writes with my dt500 throttle and see if its the same transponder interference.
Good suggestion.

pabender, 2018-03-23:
On 03/22/2018 06:12 PM, Bob Milhaupt wrote:
> But "Specification in decoder definition" does not make sense to me for
> the typical "mobile" decoder which has "Ops Mode read" capabilities. The
> typical JMRI user does not have the appropriate [...]

There is also another situation you don't specify here.  What if I have 
a decoder that can perform readback, but not using a protocol supported 
by the current connection.

I own a fairly large number of Digitrax decoders, but I don't have any 
equipment that can handle transponding.  I do own equipment that 
supports Railcom and decoders that can talk via railcom.

In other words, If we can't specify just "Ops Mode read" capabilities in 
the decoder definition.  We have to be more precise.

The precision could be something along the lines of:
Supports CV reads via RailCom
Supports CV reads via Digitrax Transponding
Supports CV reads via LocoNet
etc.

If we specify that level of detail in the decoder definition, then, at 
least for RailCom and Digitrax Transponding, we may already have a way 
to specify readback is available on the system, without making this a 
preference.

The thought proces goes something like this:

If a user adds a RailCom Reporter, then the system supports readback via
RailCom.

If a user adds a Transponding Reporter, then the system supports 
readback via Digitrax Transponding.  (Incidentally, we can't say 
"LocoNet Reporter" here because some LocoNet reporters are Railcom 
reporters (see the link I posted earlier in this thread)).

The user has the option of adding as many reporters as desired.

The Operations Mode Programmer can then check with the ReporterManager 
to determine if appropriate hardware is available to read in Ops Mode.

devel-bobm, 2018-03-23:
@pabender, your comments make sense.

Since I am not particularly familiar with JMRI `Reporters`, I took a look at the "help".  Not much there, except that a `Reporter` contains "arbitrary" data.  With Digitrax Transponding, the known uses are "presence of equipment within a given transponding zone" and "report of CV read value".  For a given decoder, the first use is not necessarily "overridden" by the second use; if the mobile decoder is in a given zone, it likely remains in that zone even though an Ops Mode Read comes thru for that decoder.  The two "uses" are not "mutually-exclusive" and the one "use" does not necessarily imply the other "use".

Presumably Railcom and Railcom-capable mobile decoders have comparable "multiple states" that apply for a given `Reporter` and must co-exist and are not mutually-exclusive.

So how do we "extend" a `Reporter` to contain multiple values?

And how does a `Reporter` manage multiple different non-presence-related reports?  For example, with transponding, performing Ops-mode reads of multiple different CVs?  Does the `Reporter` maintain the state from each read?  How long is that data valid?  Only while the decoder is within the transponding zone?

I also wonder about the relationship between `Reporters` and the `ProgrammerManager`.  Presumably, if the programmer is working with a decoder which is capable of ops-mode reads via a `Reporter`, what is the interaction?  The `ProgrammerManager` informs the `ReporterManager` that it is going to perform an "Ops-Mode Read", and the `ReporterManager` watches the `Reporters` to see if one  "reports" a response to an Ops-Mode Read, and if it happens, sends an event back to the `ProgrammerManager`?  Or the `ProgrammerManager` registers some sort of a `Listener` with the `ReporterManager`, and the `ReporterManager` sends a message to the registered listener when a reply to an Ops-Mode Read happens?

I'm not sure if the LocoNet message issued by a Digitrax Transponding reciever device (i.e. BDL16x, BXPw) provides information on the board number and zone when responding to a Ops Mode CV Read access; this might make it difficult for the read to resolve to a specific `Reporter`.  In such a case, a protocol-specific `ReporterManager` may be necessary to provide the readback information to the programmer.  The only LocoNet log I have in my notes with Transponding Ops-Mode CV read responxes were taken with a BDL168 with BoardID=1, so I cannot determine if that data is embedded in the message.

FYI - Uhlenbrock has implemented a LocoNet-based Railcom device.  Captured from their web site many years ago: "The Uhlenbrock MARCo module (art. 68500) combines Railcom (the EU-Centered Train-ID technology) and Loconet (the US / Digitrax multipurpose bus)."

pabender, 2018-03-23:
On 03/22/2018 10:58 PM, Bob Milhaupt wrote:
> @pabender <https://github.com/pabender>, your comments make sense.
>
> Since I am not particularly familiar with JMRI |Reporters|, I took a
> look at the "help". Not much there, except that a |Reporter| [...]

Reporters return objects for the reports.

Currently the LocoNet reporter returns a String.

The RailCom reporters return jmri.Railcom objects, which support 
retrieving CVs.  The jmri.RailCom interface extends the jmri.IdTag 
interface.

We also have reporters that just return IdTag objects (RFID being the 
best example).

I can't answer any of the Digitrax Transponding related questions you 
ask. I don't have that hardware and I'm not familiar with how it works.

> And how does a |Reporter| manage multiple different non-presence-related
> reports? For example, with transponding, performing Ops-mode reads of
> multiple different CVs? Does the |Reporter| maintain [...]

Since the LocoNet reporters currently just use strings as their reports, 
those are only retained as long as the string is around.  You'll have to 
look at the code for that.

For any reporter that returns an IdTag object, including RailCom 
objects, those objects are beans, so they are stored as beans, and can 
even persist after a shutdown.  One of the pieces of information stored 
by an IdTag is a last seen location.

> I also wonder about the relationship between |Reporters| and the
> |ProgrammerManager|. Presumably, if the programmer is working with a
> decoder which is capable of ops-mode reads via a |Reporter|, [...]

As with any of our bean objects, the Operations Mode Programmer (Not the 
manager, but the particular programmer performing the operation) 
registers as a listener for the reporters.

It is also possible to register a listener for an individual IdTag 
object.  Since RailCom objects are IdTags, that might be the best way to 
do it for them.

The specific programmer is responsible for interpreting the meaning of 
the response encoded in the report from either the reader or the IdTag.

> FYI - Uhlenbrock has implemented a LocoNet-based Railcom device.
> Captured from their web site many years ago: "The Uhlenbrock MARCo
> module (art. 68500) combines Railcom (the EU-Centered Train-ID
> technology) and Loconet (the US / Digitrax multipurpose bus)."

Which means there are at least two LocoNet Railcom readers.  The biggest 
question I  have is now do they encode information, and is it different 
than the way Transponding encodes information.

bobjacobsen, 2018-03-25:
I think we can separate this into parts:

1) LocoNet-connected devices - The default for `transpRead` is "no", which means that transponding is not required for these to be read.  `ops-"yes"` in their decoder definitions (which either say `transpRead="no"` or leave it at default) means that a LocoNet command station can just read from them. All that's OK, no changes. If somebody has one of these and it's not connected, it won't work anyway....

2) Transponding-capable decoders - These have `ops-"yes"` and `transpRead="yes"`, which we'll now interpret as "if transponding is installed, then you can read these in ops mode, if not, not". Then we add a "Transponding present?" option to the Advanced Options of the LocoNet connection that's providing the programmer to say whether transponding really is present. Note that it's up to the user to properly configure the decoders and other transponding hardware such that transponding actually works. If it doesn't, then they should leave that option off.

3) RailCom and NMRA Advanced Ack decoders - I have no idea.  Let's solve that later.

Note that neither 1 nor 2 (I don't know about 3) have anything to do with JMRI Reporters. They're separate from the programmer itself.

pabender, 2018-03-25:
Sent from my iPad

> On Mar 25, 2018, at 1:50 AM, Bob Jacobsen <notifications@github.com> wrote:
> 
> I think we can separate this into parts:
> 
> LocoNet-connected devices - The default for transpRead is "no", which means that transponding is not required for these to be read. ops-"yes" in their decoder definitions (which either say transpRead="no" or leave it at default) means that a LocoNet command station can just read from them. All that's OK, no changes. If somebody has one of these and it's not connected, it won't work anyway....
> 
> Transponding-capable decoders - These have ops-"yes" and transpRead="yes", which we'll now interpret as "if transponding is installed, then you can read these in ops mode, if not, not". Then we add a "Transponding present?" option to the Advanced Options of the LocoNet connection that's providing the programmer to say whether transponding really is present. Note that it's up to the user to properly configure the decoders and other transponding hardware such that transponding actually works. If it doesn't, then they should leave that option off.
> 
> RailCom and NMRA Advanced Ack decoders - I have no idea. Let's solve that later.
> 

3 is really the same as 2, only it is a different set of hardware.  And we do need to deal with this now, because issue https://github.com/JMRI/JMRI/issues/5132 is highly related to all of this.

> Note that neither 1 nor 2 (I don't know about 3) have anything to do with JMRI Reporters. They're separate from the programmer itself.

In my view, 2 and 3 really do have to do with the reporters.  It is the transponding and/or railcom readers that deliver the information to us, just like the other information they deliver.  Defining the reporter hardware gives us the configuration information.

Also, my understanding of the LocoNet based Railcom readers is that they send the same messages that Digitrax uses for Transponding.  I do not have hardware available to confirm that however.

bobjacobsen, 2018-03-25:
> > Note that neither 1 nor 2 (I don't know about 3) have anything to do with JMRI Reporters. They're separate from the programmer itself.

> In my view, 2 and 3 really do have to do with the reporters.

Note the capital "R" in my comment.  You do not need, and you do not want, JMRI Reporters to be involved in LocoNet ops mode read back.  The existing code, which already properly handles the read back hardware, operates without them so they're not needed.  They're not trivial to set up, so having them be involved in this is undesirable.

For LocoNet, the only thing that's needed is an option that makes the command station smart enough to not wait to hear from that hardware when it's not expected to be there.

pabender, 2018-03-25:
Sent from my iPad
> On Mar 25, 2018, at 2:39 AM, Bob Jacobsen <notifications@github.com> wrote:
> 
> Note that neither 1 nor 2 (I don't know about 3) have anything to do with JMRI Reporters. They're separate from the programmer itself.
> 
> In my view, 2 and 3 really do have to do with the reporters.
> 
> Note the capital "R" in my comment. You do not need, and you do not want, JMRI Reporters to be involved in LocoNet ops mode read back. The existing code, which already properly handles the read back hardware, operates without them so they're not needed. They're not trivial to set up, so having them be involved in this is undesirable.
> 
> For LocoNet, the only thing that's needed is an option that makes the command station smart enough to not wait to hear from that hardware when it's not expected to be there.
> 

Ok, but there are two issues that need to be addressed still:

1) We need to define a mechanism by which we are going to enable readback for RailCom decoders.  

On Loconet, everything I can find states the existing hardware acts like a BDL16x setup for transponding, so the LocoNet message processing for these should be in place, but this is not transponding, and we need to denote that.

On non-Loconet systems, the JMRI Reporters provide the CV feedback information ( if available, it is encoded in the reported objects ).

2) How do we handle the situation where we have a stand-alone Loconet with transponding installed if the user wants Transponding with a non-Loconet command station?

Your solution assumes the programmer and the feedback will come from the same system.  I don’t see how to implement feedback from another connection unless we get the JMRI Reporters involved.

devel-bobm, 2018-03-25:
@pabender,

Assuming that there is a real use-case with a "Railcom" device providing information to a LocoNet connection as a result of a non-LocoNet command station performing Ops-mode read operations, we have a systematic problem, because the programming operation "spans" two connections.  Having one connection (sometimes) provide status for another connection's requests is pretty far off the way things are normally structured in JMRI.  

This may just be the first of many such cases where cross-connection reporting makes sense.  An architectured solution may make sense.  

Is it the `Reporter` functionality?  I don't know.  But if it turns out that `Reporter` objects will be the "catch-all" for such cross-connection command/response information transfer, then we ought to architect a flexible, robust architecture for such cross-connection transfer.  Sooner or later, someone will come up with an n-way communication transfer which really makes sense and where n is much greater than 2.

bobjacobsen, 2018-03-25:
> How do we handle the situation where we have a stand-alone Loconet with transponding installed if the user wants Transponding with a non-Loconet command station?

I don't believe that case exists in practice.  LocoNet transponding is synchronized to the track signal at the microsecond level via the RailSync signal in the LocoNet cable. Empirically, transponder won't work with NCE command stations, for example, due to the length of the transmitted bits.

As to more general cases for RailCom: Maybe a more general solution is needed.  I don't know enough to create one now based on the single case of transponding, which seems to have very specific constraints built in.  The Reporter mechanism might be part of some cross-system approach, but until we have a larger number of examples of RailCom in use, I don't see how.  I would instead expect that system vendors would make on-main programming work for their command station, throttle and RailCom gear, and we'd pick up from there without having to solve an N^2 cross-connect problem.

devel-bobm, 2018-03-25:
@bobjacobsen, I believe that it would be possible to use Digitrax Transponding with an NCE command station, although I've never heard of anyone doing the "proof-of-concept" work. 

 It is very reasonable to drive the "RailSync" wires on a "Standalone LocoNet" from an NCE "booster net"; it's already documented in the JMRI help pages [RailSync from an NCE Command Station](http://jmri.org/help/en/html/hardware/loconet/StandaloneLocoNet.shtml#rsfromncecs).  I believe that a "Stanadlone LocoNet", with such a RailSync signal, connected to Digitrax transponding hardware, would stand a moderate chance of working.

Would there be a bit-level timing issue?  I don't know, because I've never worked with at the bit-level with NCE and cannot pretend to know of any significant differences in the bit-level timings.

bobjacobsen, 2018-03-25:
> I believe that it would be possible to use Digitrax Transponding with an NCE command station, although I've never heard of anyone doing the "proof-of-concept" work.

That's what "empirically" means: I did it, I have the scope photos to document it, and it doesn't work. 

Digitrax 1 bits (at least from then-current DCS100 boosters) are longer than NCE 1 bits.  IIRC, the NMRA would occasionally argue they were out-of-spec with respect to NMRA S9.1, although I remember that (a) as usual, the language in the document was miserably unclear and (b) Digitrax didn't care.

I'll look around to see if I can find any of that info.

pabender, 2018-03-25:
Sent from my iPhone

> On Mar 25, 2018, at 11:37 AM, Bob Milhaupt <notifications@github.com> wrote:
> 
> @pabender,
> 
> Assuming that there is a real use-case with a "Railcom" device providing information to a LocoNet connection as a result of a non-LocoNet command station performing Ops-mode read operations, we have a systematic problem, because the programming operation "spans" two connections. Having one connection (sometimes) provide status for another connection's requests is pretty far off the way things are normally structured in JMRI.
> 

Actually, that is something we do already.  It is common for users to have feedback for turnouts off a bus that isn't controlling them, for example.

In the bigger scheme of things, it is perfectly reasonable for 

Lenz announced, but has never actually released, a USB device that acts as a RailCom gateway 
> Is it the Reporter functionality? I don't know. But if it turns out that Reporter objects will be the "catch-all" for such cross-connection
> 
It is not the reporters that handle the information, but the objects they return.

In the case of RailCom objects, the objects can store CV values associated with the address, along with other information.

pabender, 2018-03-25:
Sent from my iPad

> On Mar 25, 2018, at 11:56 AM, Bob Jacobsen <notifications@github.com> wrote:
> How do we handle the situation where we have a stand-alone Loconet with transponding installed if the user wants Transponding with a non-Loconet command station?
> 
> I don't believe that case exists in practice. LocoNet transponding is synchronized to the track signal at the microsecond level via the RailSync signal in the LocoNet cable. Empirically, transponder won't work with NCE command stations, for example, due to the length of the transmitted bits.
> 

Digitrax implies Trandponding should work with other DCC systems:
http://www.digitrax.com/tsd/KB963/transponding-with-other-dcc-systems/

While there may be specific cases where it doesn’t work, but there may be other cases where it does.  I think our solution, both for Transponding and RailCom, should be more general.

We have a use case where a single system ( the Roco Z21 ) can talk both  Loconet and XPressNet and has built in railcom support ( which is actually independent of both the Loconet and XPressNet subsystems ).  I don’t have the LocoNet hardware available to do anything with transponding on this system.
> As to more general cases for RailCom: Maybe a more general solution is needed. I don't know enough to create one now based on the single case of transponding, which seems to have very specific [...]
Many of the manufacturers do provide the railcom feedback through the command station connection.

Others have chosen to implement a separate bus for RailCom information ( examples include Tams ( see page 11 of http://tams-online.de/WebRoot/Store11/Shops/642f1858-c39b-4b7d-af86-f6a1feaca0e4/MediaGallery/15_Download/Anleitungen_GB/45/RCD-2_2013_07_EN.pdf ) and Lenz ( though the Lenz hardware is vaporware at this point... ))

There are certainly examples of people using the LocoNet RailCom devices with non LocoNet systems ( https://www.locgeek.com/2013/01/hands-on-uhlenbrock-marco-railcomloconet/  ( note that this is one place you can see RailCom information presented on Loconet as BDl168 messages ))

bobjacobsen, 2018-03-25:
So I'm out on this.  I'll fix the actual issue with real hardware that started in Issue #5128, see PR #5152 for the 1st half.  I'll leave the rest of the bike shed to others.

KenC57, 2018-03-25:
Sometime this year there should be a board for LCC that has eight zones for
RailCom. As I understand its development and the TCS LCC command station are
intertwined somewhat. So that will throw another vendor into the mix of
things. Once a couple of those get out for testing we should be able to play
with that setup too.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

devel-bobm, 2018-03-25:
@pabender,

I have seen no reporting of the LocoNet messaging used by "Railcom-over-Loconet" implementations when an Ops-mode read reply is returned by a Railcom decoder.

I know that the message generated under Digitrax transponding is a 14-byte programming slot report (i.e. <OPC_SL_READ_DATA>, as opposed to the "transponding location report" which is sometimes known as <OPC_MULTI_SENSE>, a 6-byte message.  Digitrax does not use the 6-byte "transponding location" message format, or a variation of same, for Ops-mode read replies.

pabender, 2018-03-25:
Sent from my iPad

> On Mar 25, 2018, at 5:45 PM, Bob Milhaupt <notifications@github.com> wrote:
> 
> @pabender,
> 
> I have seen no reporting of the LocoNet messaging used by "Railcom-over-Loconet" implementations when an Ops-mode read reply is returned by a Railcom decoder.
> 

I haven’t seen any report of that either, and I don’t have the hardware to check.
> I know that the message generated under Digitrax transponding is a 14-byte programming slot report (i.e. <OPC_SL_READ_DATA>, as opposed to the "transponding location report" which is sometimes known [...]

That may very well mean that Loconet is going to need a different solution than everything else.

If we work under the assumption that Loconet RailCom readers mimic Digitrax Transponding hardware, then how do we differentiate between the two?

devel-bobm, 2018-03-25:
@pabender, "If we work under the assumption that Loconet RailCom readers mimic Digitrax Transponding hardware, then how do we differentiate between the two?"

In that case, I don't think we would be able to differentiate between a LocoNet-based RailCom reresponse and a Digitrax Transponding response for transponding Ops-mode CV read reports.

pabender, 2018-03-26:
Ok, so I've been working tonight on #5132, and found that Roco, like Digitrax, made the decision to just send the CV number and the data byte through XPressNet using the same result message as is used for service mode devices.  This isn't a standard XPressNet command, but one that Roco added to the protocol.  Fortunately, the hardware is always present on a Z21 system (the required RailCom reader is built into the command station/booster box.)

Looking at the other Systems we support that provide RailCom based reporters:

Dcc4Pc Reporters definitely decode CV and value information from RailCom.  The Dcc4Pc Ops Mode Programmer DOES get CV feedback information by accessing jmri.RailCom objects.  The Dcc4Pc RailCom boards are separate hardware from the command station, and can connect via a separate USB interface (see http://www.dcc4pc.co.uk/our_products.html ).

Ecos Reporters currently treat RailCom data as only an IdTag, not as a RailCom object, so I don't know how the Ecos treats CVs.  Is there anyone currently active developing for Ecos?  

( @kevin-dickerson wrote this code originally, both for the Ecos and for Dcc4Pc )

devel-bobm, 2018-03-26:
@pabender, It might be worthwhile to send an inquiry to the developer's list about any ongoing ECOS+RailCom development.

closed 2021-04-14 by stale[bot]