JMRI/JMRI #5132: Z21 and DecoderPro

wimorrison, 2018-03-23:
I have loaded decoderpro and it appears to connect with the Z21 however I cannot get it to read anything from my locos.

I am trying to program them on main as I have railcom fully enabled throughouyt and it works correctly, just I cant seem to do what I wanted with decoderpro.

Does it only work on the programming track (I don't have one as I have never needed one) - bit of apin if I have to create one :(

any suggestions?

Thanks

wimorrison, 2018-03-23:
I have worked out that programming on track doesn't work with Z21, or at least I cannot make it work as I want namely reading CV values that are set already and then changing them. This is what Railcom allows but DecoderPro doesn't seem to like Railcom and doesn't seem to support it :(

I have also found that when setting custom speed curves on the programme track Railcom gets switched off by decoder pro which caused me a lot of head scratching trying to work out why I had lost my feedback data.

pabender, 2018-03-23:
Let me look into this.  I have the equipment at home to make check into what is happening.

pabender, 2018-03-26:
I have started to look at the first issue, and that is caused by the simple fact that we have not been handling the input from the command station in Ops Mode.

It turns out the command station returns the results in the same manner whether we are talking about programming in service mode or ops mode.

pabender, 2018-03-28:
@wimorrison,

I've committed the code required to get the RailCom data to JMRI.

This has been tested on a Z21, but when you get a chance, please let me know if the code works for you.  You can download a pre-compiled version of the program with the update here:

http://jmri.tagadab.com/jenkins/job/Development/job/Packages/

pabender, 2018-03-28:
@wimorrison, one other question, which decoder or decoder(s) are you seeing the RailCom data shut off on when you program?  I want to see if I can replicate that behavior.

wimorrison, 2018-03-31:
thanks for sorting this out - seems to work though reading the custom speed curve seems to be slower than it was before, but at least it s working and not knocking off the Railcom in CV29 when it writes back :)

I use Lenz Silver Mini decoders.

pabender, 2018-04-01:
Sent from my iPhone

> On Mar 31, 2018, at 6:15 PM, wimorrison <notifications@github.com> wrote:
> 
> thanks for sorting this out - seems to work though reading the custom speed curve seems to be slower than it was before, but at least it s working and not knocking off the Railcom in CV29 when it writes back :)
> 
This is because before the reads just returned a constant value.  Now JMRI actually waits for a response.
> I use Lenz Silver Mini decoders.
> 
Ok.  I actually have some of those ( and some gold minis ).  I actually tested all this with a silver mini.

pabender, 2018-04-01:
Closed via #5157

closed 2018-04-01 by pabender