JMRI/JMRI #5135: build.xml javac encoding

jszinger, 2018-03-23:
build.xml does not set the encoding for javac.  This causes compilation to fail in a non-UTF-8 environment.  E.g.
`LANG=C ant package-linux`
results in errors such as
`[javac] /builddir/build/BUILD/JMRI-4.11.4/java/src/jmri/util/usb/RailDriverMenuItem.java:620: error: unmappable character for encoding ASCII`
`[javac]                             //log.info("????????????    speed: " + speed);`
The line in question has some non-ascii bullet characters.

The attached patch should explicitly declare UTF-8 instead of relying on the environment.  I've tested it to the point of building ant package-linux for 4.114.

[jmri-build-xml-patch.txt](https://github.com/JMRI/JMRI/files/1842724/jmri-build-xml-patch.txt)

silverailscolo, 2018-03-24:
Apart from ``ant`` per se, using non-ascii chars in code (comments) will also come up in IntelliJ IDE. In the past we have systematically handled this by replacing them with plain ascii substitutes. Wish to check whether proposal also works in Ij.

bobjacobsen, 2018-03-24:
I'm not sure of the right approach here.  Should we have java files with UTF coding? There are pluses and minuses there that I haven't fully understood.

I scanned (with macOS `file`) for UTF content and found some, but not too much:
```
JMRI/ % find java/src -type f -exec file --mime {} \; | grep utf
java/src/jmri/util/usb/RailDriverMenuItem.java: text/x-java; charset=utf-8
java/src/jmri/web/server/Bundle_fr.properties: text/plain; charset=utf-8
java/src/jmri/jmrit/operations/rollingstock/engines/JmritOperationsEnginesBundle_fr.properties: text/plain; charset=utf-8
java/src/jmri/jmrit/operations/setup/JmritOperationsSetupBundle_fr.properties: text/plain; charset=utf-8
java/src/jmri/jmrit/operations/automation/JmritOperationsAutomationBundle_fr.properties: text/plain; charset=utf-8
java/src/jmri/profile/Bundle_fr.properties: text/html; charset=utf-8
java/src/jmri/jmrix/can/cbus/CbusBundle_de.properties: text/plain; charset=utf-8
java/src/jmri/jmrix/mrc/MrcBundle_fr.properties: text/plain; charset=utf-8
java/src/jmri/jmrix/dccpp/DCCppBundle_de.properties: text/plain; charset=utf-8
java/src/jmri/jmrix/loconet/locostats/swing/LocoStatsBundle_fr.properties: text/plain; charset=utf-8
java/src/apps/AppsConfigBundle_fr.properties: text/html; charset=utf-8

% find java/test -type f -exec file --mime {} \; | grep utf
java/test/jmri/jmrit/conditional/BundleTest.java: text/x-java; charset=utf-8
java/test/jmri/jmrit/display/verify/AA1UPtest.xml: text/xml; charset=utf-8
java/test/jmri/jmrit/decoderdefn/fail/FactResetDuplicateI18N.xml: text/xml; charset=utf-8
java/test/jmri/jmrit/decoderdefn/fail/FactResetLabelDuplicateDefault.xml: text/xml; charset=utf-8
java/test/jmri/jmrit/symbolicprog/tabbedframe/pass/DecoderWithQualifier.xml: text/xml; charset=utf-8
```
I manually removed some test XML files from list; those are OK as UTF; the .properties files probably need to be fixed as [Java 8 doesn't correctly handle UTF in properties](http://jmri.org/help/en/html/doc/Technical/I8N.shtml). On the other hand, with proper settings Java itself is fine with UTF in raw strings, comments, etc.

So I think our choices are to (1) stick with us-ascii coding and add a test to check for it or (2) switch to allowing it in .java files, along with associated changes to the build structures.

jszinger, 2018-03-26:
Thanks for looking at this.  I've also notices that the encoding affects building the javadoc, and pom.xml has the line `<project.build.sourceEncoding>UTF-8</project.build.sourceEncoding>`.

rhwood, 2020-05-12:
This appears to have been fixed since javac calls in build.xml appear to specify the UTF-8 encoding.

closed 2020-05-12 by rhwood