JMRI/JMRI #5137: Updated Expanded Basic Signal System

gpmccartney, 2018-03-23:
[basicexpanded.zip](https://github.com/JMRI/JMRI/files/1842941/basicexpanded.zip)
[NW-1981.zip](https://github.com/JMRI/JMRI/files/1842949/NW-1981.zip)

I have made an update to the basic expanded signal system I submitted last month.  Since it was not committed to the last build I am going to close that issue.  I had also updated and fixed an issue with the NW system so I have included it again.

bobjacobsen, 2018-03-25:
Is "basicexpanded" really similar enough to "basic" that we should just combine them?

I'm going to split these into two separate PRs.

bobjacobsen, 2018-03-25:
The NW-1981 part of this is now #5148, thanks again.

One question:  As I'm color blind, I can't really tell, but did the color of some of the display .gifs change?
[image]

If so, is the shape still coding the color? (horizontal for red, vertical for green, diagonal for yellow) That's the only way I can tell them apart e.g. on a panel.

KenC57, 2018-03-25:
I took a look at the file. I think one advantage of keeping them apart is
the basic expanded has flashing aspects while basic does not. Being separate
would keep the side effect of flashing that some systems have out of the
picture in a simple way for the users. I did not check to see if the
expanded matches for the existing basic or if it would change that too.
-ken c

rhwood, 2018-03-25:
There is a slight change in the color of the yellow in the changed signal images.

bobjacobsen, 2018-03-25:
Looking at aspects.xml in basic (B) and basic expanded (E), I see:
 - (E) has added DCC aspect definitions
 - Unlit is changed from Step in (B) to Normal speed in (E)
 - (E) has a Restricting aspect defined
 - (E) has significantly reworked the diverging aspects

I don't see any breaking changes for existing signal configurations, but I may have overlooked something. So it seems generally better.

@KenC57 Are the flashing appearances required, in the sense that users have to use them? Or are they either a feature of specific masts and/or something that can be disabled and leave the signals still working?

It's (currently) really hard to switch from one signal system definition to another.  Switching from "basic" to "extended basic" will be hard for somebody who's outgrown basic. So it seems better to have a introductory set that works as well (even if not true to some specific prototype) as possible.

bobjacobsen, 2018-03-26:
NM 1981 part now merged in #5158.  Waiting on some direction re basic vs expanded basic for the other half.

gpmccartney, 2018-05-31:
Sorry fellows.  I somehow completely missed this discussion.  I checked back multiple times and never saw anything, so I figured there just really wasn't any interest.  I created the expanded basic for 2 reasons.  First when I looked at the existing basic system, in my opinion there were some issues with certain mappings.  Rather than change that and break something someone had already set up, I just created a new system.  The second reason was there was a desire by one person on the group to have a simple system that had three heads.  That way they could have a signal they could set up a situation that had three possible routes.  So the three head signal had a medium and slow route.  Calling it expanded basic may not be the best idea.   Perhaps it should just be called "The Generic Signal System."

Again sorry for the delay.  

Greg McCartney

bobjacobsen, 2018-06-03:
@KenC57 Could you comment on whether somebody using this would hit the flashing appearances if he didn't intend to?  I.e. if he set up basic trackwork in a basic way, would the more advanced aspects just be there for future use?  See comment above about how it's hard for people to migrate to a different system later.

gpmccartney, 2018-06-04:
I can answer that Bob and it would be yes.  When I added the restricting aspect to the signal system, red over yellow, it took the place of what had been diverging approach.  Flashing the yellow was the only upgrade I could think of barring using lunar.  I opted to not use lunar since that could cause a person to need a signal head containing 4 lights.  I'm not even sure if such a head exists in model form.  So a train diverging into a siding in which the next signal is either stop or restricting would get a flashing yellow.  

The flashing yellow would be on the lower head on a two head signal, and the bottom head on a 3 head signal if the block speed was set at "slow."  

The reality is the two systems are not compatible and really wouldn't be an upgrade.  I should not have called it basic expanded.  I am more in favor of calling it "Generic Signal System."  

Also I guess I did not follow proper protocol in creating the icons in that I didn't make the light shapes in vertical, horizontal, or diagonal fashion.  I can go back and revise them so that they follow suit.  

This will also require another fix to the NW system.  I just checked and since the new icons I created were stripped out of the system and the old set used, there are incorrect icons displayed for the Wabash.

Greg McCartney

bobjacobsen, 2020-07-25:
Already merged.

closed 2020-07-25 by bobjacobsen