JMRI/JMRI #5139: Updating "moment" module

bobjacobsen, 2018-03-23:
GitHub is notifying us:
```
We found a potential security vulnerability in one of your dependencies.
A dependency defined in app/package-lock.json has known security vulnerabilities and should be updated.
```
with additional information:
```
Moderate severity
The moment module before 2.19.3 for Node.js is prone to a regular expression denial of service via a crafted date str...

package-lock.json update suggested:
moment ~> 2.19.3
Always verify the validity and compatibility of suggestions with your codebase.
```
The underlying issue is [CVE-2017-18214](https://nvd.nist.gov/vuln/detail/CVE-2017-18214).

Should we update that? How do we update that? (It's not in a part of the code I've worked on)

rhwood, 2018-03-24:
This is updated by editing `web/app/package.json` and then running the command `npm update` from the `web/app` folder.

rhwood, 2018-03-24:
I had been experimenting with creating a [Cockpit Project](http://cockpit-project.org)-like user interface for controlling JMRI, and that is what this the moment warning is coming from.

PR #5145 (2018-03-25): Update web app dependencies.
Update JavaScript dependencies for the web app accessible from `http://jmri-server:12080/app`.

Fixes #5139
closed 2018-03-25 by bobjacobsen