JMRI/JMRI #5141: Zone goes high after reporter makes it report, resulting in a dropped report.

ClockLoop, 2018-03-24:
The bxp88 zone sensors are kinda slow, it take about 1.5 seconds for the zone to go high or low, but the reporter works instantaneously, so if the zone goes high after the report, the report is dropped or ignored.

[image]

ClockLoop, 2018-03-25:
I didn't think to try it, but  i suppose i could just issue the same sequence of commands and see if it does it again.  JMRI even provides a way to send multiple commands with a delay, so i will see if I can replicate it with commands.   :)

devel-bobm, 2018-03-25:
Pardon my ignorance, but what is a "high" zone?  

So far as I know, the terminology is that "zone" is the term used to represent a "transponding zone" and "Detection Section" is the term used to represent the block in which occupancy is detected.  This is historical, based on BDL16x detection with RX4 transponding, and is a basic assumption of JMRI.  

And critical to this is that a "transponding zone" does not necessarily have ANYTHING to do with any "detection section".  A "transponding zone" might include all or part of a "detection section", but it does not need to include any "detection section".  It might even include multiple "detection section"s!

ClockLoop, 2018-03-25:
Ok, the bxp88 ties the detection section and transponding zones together.   They are not like the bdl with rx4 where you get to choose a more unique config.

I will change the title to match.

closed 2021-04-14 by stale[bot]