JMRI/JMRI #5143: Loconet Stationary Interrogate - Turnout Sensor Report (Feedback) handling issue

n3ix, 2018-03-24:
Problem Overview - When JMRI receives a Loconet INPUT LEVELS Turnout Sensor Report for a turnout not yet in the Turnout Table it creates the Turnout but leaves the State set to UNKNOWN, ignoring the feedback information in the Turnout Sensor Report.

Background
When JMRI starts it performs the Stationary Interrogate sequence on Loconet connections.  The Interrogate sequence is also performed by Digitrax Command Stations whenever track power is turned On.  Loconet Stationary decoders respond to the Interrogate messages with Turnout Sensor Reports (Feedback) for each physical turnout.  There are 2 forms of the Turnout Sensor Report message that the decoder could use - OUTPUT LEVELS and INPUT LEVELS.  The results in JMRI are different depending on which form of the Turnout Sensor Report it receives.  Consider the follow cases:

1.  Panel file not yet loaded and the Turnout is not yet in the turnout table.  JMRI receives a Output Levels Turnout Sensor Report.
Result:  The Turnout is created, the feedback mode is set to MONITORING, and the State is CORRECTLY set per the data in the Output Levels Turnout Sensor Report.

2.  Panel file not yet loaded and the Turnout is not yet in the turnout table.  JMRI receives an Input Levels Turnout Sensor Report.
Result:  The Turnout is created, the feedback mode is set to MONITORING, and the State is set to UNKNOWN.

3.  The Turnout is already in the turnout table with feedback mode set to MONITORING.  JMRI receives a Output Levels Turnout Sensor Report.
Result:  The Turnout State is CORRECTLY set per the data in the Output Levels Turnout Sensor Report.

4.  The Turnout is already in the turnout table with feedback mode set to MONITORING.  JMRI receives an Input Levels Turnout Sensor Report.
Result:  The Turnout State is NOT CHANGED.

5.  Panel file has been loaded and the Turnout is already in the turnout table with feedback mode set to EXACT.  JMRI receives a Output Levels Turnout Sensor Report.
Result:  The Turnout State is NOT CHANGED.

6.  Panel file has been loaded and the Turnout is already in the turnout table with feedback mode set to EXACT.  JMRI receives an Input Levels Turnout Sensor Report.
Result:  The Turnout State is CORRECTLY set per the data in the Input Levels Turnout Sensor Report.

Suppose a user has configured their physical Loconet Stationary Decoder to provide EXACT (INPUT LEVEL) feedback.  

A) If the Decoder responds with an INPUT LEVEL feedback message during Interrogation, JMRI will not use the feedback to set the State in case (2) above.   

B) If the Decoder responds with an OUTPUT LEVEL feedback message during Interrogation, JMRI will not use the feedback to set the state in case (5) above.

So how to ensure that JMRI properly initializes the Turnout state during Interrogation when the physical stationary decoder is configured to provide EXACT feedback?  To handle this scenario I took the approach in the QuadLN decoders of sending both the OUTPUT LEVEL and INPUT LEVEL forms of the feedback message during Interrogation.  This covers both case (1) and case (6) above and JMRI always initializes the Turnout state correctly.  

The downside of this solution is that Interrogation results in two feedback messages per turnout instead of one.  In almost all cases this has not been a problem.  However on very large club layouts the high number of responses to Interrogation can already be problematic due to collisions.  In this environment doubling the number of feedback messages just makes things worse and increases the chances that some Turnouts will end up in the UNKNOWN state.  Since UNKNOWN Turnouts cause problems in layout operation, other measures are then required to resolve all the UNKNOWN Turnouts (and to resolve all the UNKNOWN Sensors, since excess collisions can also result in missed Sensor feedback messages during Interrogation).

What I'd like to propose is a change in the way JMRI handles case (2).  When the Turnout does not yet exist in the Turnout Table and JMRI sees an INPUT LEVELS feedback message, can JMRI initialize the Turnout State per the feedback message?  If so, the decoder could just send the EXACT feedback message during Interrogation. 

Thanks for your consideration.

bobjacobsen, 2018-03-25:
Thanks for laying this out in such clear detail.

One more case to consider:  The inputs are being used for something _other_ than EXACT feedback, e.g. occupancy sensors.  If the message comes in before the panel is loaded, won't the Turnout status end up in (perhaps) a wrong state?

devel-bobm, 2018-03-25:
@n3ix,

On first glance, this seems a reasonable request.

I would want to think about the details you've provided much more carefully (than I can after 20+ hours up-time) before doing any coding, though!

Perhaps I should not type while drowsy, and haven't fully comprehended the intial comments, but @bobjacobsen, presumably "occupancy sensor" or other "non-turnout sensor" usage would be in yet another variant of the message - the case which specifies the <SN2> "I" bit as 0 for "aux" input usage.  Presumably, when "I" is 0, the message would not be used to create a new Turnout.

bobjacobsen, 2018-03-25:
> presumably "occupancy sensor" or other "non-turnout sensor" usage would be in yet another variant of the message - the case which specifies the "I" bit as 0 for "aux" input usage. Presumably, when "I" is 0, the message would not be used to create a new Turnout.

Perhaps. It might depend on hardware involved and how it's configured.

devel-bobm, 2018-03-25:
If we cannot rely on the state of the "I" bit to determine whether the report is definitively "sensor" or "turnout", then wouldn't we need to create a "Turnout" for every report that comes in with "I" = 0 (i.e. "aux" input usage)?  And also create a "Sensor" object for that same address?

Seems to me that it would be more sane (i.e. easier) to simply take the "I" bit at "face-value", and create (if necessary) a "Sensor" for the address when "I" is 0, and create (if necessary) a "Turnout" for the address when "I" = 1.  And let the user determine when his hardware is being misleading about the "thingy" type, and let the user resolve those misleading assumptions.

n3ix, 2018-03-25:
It sounds like there is believed to be a case where a board is sending a Turnout Position Feedback report (B1)  for pure sensor inputs rather than using a General Sensor Input report (B2)?  Wow, that is really surprising.  Is this a Digitrax device?  I just pulled out an old DS54 manual and Table VI shows that each input be configured individually for various reporting message types including "Sensor Input message".   So wouldn't the user be expected to configure this device for "Sensor Input message" if that was what they wanted?

devel-bobm, 2018-03-25:
@n3ix - never underestimate the confusion of the DS54 user.  I've seen a lot of silly DS54 programming over the years, and have committed some of that sillyness myself.

And the diy microcontroller community is coding up some interesting spins on LocoNet sensors, too.

In the grand scheme of things, LocoNet "turnout feedback report" messages are really just a different kind of "general purpose input" message, with a different address space, where it is -implied- that there is a related physical turnout.  That implication is not always correct.

A good example is a device which provides "Turnout position feedback" for the points of a manually-controlled turnout which does not have any mechanism for LocoNet control of those points.  I have done this myself as part of a CTC implementation, long ago, using a DS54 input but not the associated DS54 output.

My biggest concern for JMRI and "inferring creation of `Turnout` and `Sensor` obects" is that LocoNet spec and DS54 document allows for several different ways to report turnout position, and they imply different LocoNet messaging.  And JMRI then adds the ability for a "general purpose" sensor to provide turnout position feedback.  

From the JMRI standpoint, it can be difficult to know when a given LocoNet message content is implying a turnout output state or a turnout position feedback (not the same thing!) or a general sensor that is providing point position feedback or a general sensor that is being used for something unrelated to turnout position.  For the first two, the address and other info encoded in the message can provide all that JMRI needs.  

But with the last two, JMRI cannot really infer the correct interpretation.  The LocoNet "Sensor" address does not necessarily have anything to do with a LocoNet Turnout address.  For example, two-sensor "exact" feedback.  The Sensor may or may not be related to a physical Turnout, but, when the turnouts/sensors are not pre-configured, JMRI has no way of knowing.  So JMRI cannot infer much of anything from a LocoNet "Sensor" message other than there is a sensor by a given address and it has a state.  Assuming anything about a Turnout from that message is misleading at best.

n3ix, 2018-03-25:
@devel-bobm  I understand that we can't infer Turnout info from a Sensor report and that implies JMRI will not be able to discover turnouts using "one-sensor" or "two-sensor" feedback.  But dealing just with Turnout Feedback reports for a moment, JMRI is already using these to discover turnouts.  Seems to me that the issue here is that JMRI is treating a Turnout Feedback OUTPUT report differently than a Turnout Feedback INPUT report during discovery.  For an OUTPUT report JMRI initializes the turnout State and for an INPUT report JMRI leaves the state at UNKNOWN.   I don't get that.  If the logic is that someone somewhere has hijacked the Turnout Feedback INPUT report so JMRI is better off not initializing the turnout state, why wouldn't the same logic apply to the Turnout Feedback OUTPUT report?  Couldn't someone somewhere have hijacked the Turnout Feedback OUTPUT report for some other purpose as well?    

I get that people use Loconet messages for other purposes.  Doing it myself in the latest code update in fact - using Request State of Switch (BC) message to provide a way to directly query the state of a particular Turnout and Sensor (as another crutch for Stationary Interrogate issues on very large club layouts).  Just wonder if maybe JMRI could default to treating both forms of the Turnout Feedback reports at face value rather than treating OUTPUT and INPUT forms different, and then handling any known hijacks via user-selected options?

devel-bobm, 2018-03-25:
@n3ix,

I agree with the principle that "JMRI could default to treating both forms of the Turnout Feedback reports at face value".  No need for feature-creep...

n3ix, 2018-03-25:
@devel-bobm I am specifically referring to the OPC_SW_REP message, I.e. ";Turnout  SENSOR state REPORT "as titled by Digitrax.  In the 6 cases I detailed above, case (1) OUTPUT Levels refers to <SN2> =<0,1,I,L- A10,A9,A8,A7> and case (2) INPUT Levels refers to <SN2> =<0,0,C,T- A10,A9,A8,A7>.  Case (2) is used to report Exact feedback.   JMRI creates a Turnout when it sees either the OUTPUT Levels or the INPUT levels form of this message.  The difference is what JMRI does with the State of the new turnout.  When JMRI sees an OUTPUT Levels report for an undefined Turnout, it creates a new Turnout _and_ sets the State of the new Turnout per the state info in the OUTPUT Levels report.  When JMRI sees an INPUT Levels report for an undefined Turnout, it creates a new Turnout but does not set the State of the new Turnout per the state info in the INPUT Levels report.

For any turnout, whether newly created or existing, in my testing the only time JMRI will use the State information from an INPUT Levels form of the OPC_SW_REP is when the Turnout feedback method has already been set to EXACT.

devel-bobm, 2018-03-26:
@n3ix,  I've been looking at this in a number of different ways today.  I have a number of thoughts ideas.

First things first, I find a discrepancy between your most-recent statements that "case (1) OUTPUT Levels refers to =<0,1,I,L- A10,A9,A8,A7>" and "case (2) INPUT Levels refers to =<0,0,C,T- A10,A9,A8,A7>".

From LocoNet Personal Edition Spec 1.0, I see the opposite.  I see that "**INPUT** Levels refers to =<0,1,I,L- A10,A9,A8,A7>" and "**OUTPUT** Levels refers to =<0,0,C,T- A10,A9,A8,A7>"

Is this an inaccuracy of the LocoNet PE spec or is it a typo in your post?

devel-bobm, 2018-03-26:
@n3ix,

Would this JMRI `Turnout` initialization problem be significantly reduced by ensuring that the JMRI-initiated "interrogation" does not occur until after the Panel is loaded?

n3ix, 2018-03-26:
@devel-bobm  Yes I reversed the <SN2> values for the 2 forms, my apologies.

_If_ there was a way to guarantee that the panel file had been loaded before JMRI does the Interrogate, then all the Turnouts should be defined and then yes they would initialize properly.  This can create other issues however.  On one large layout with the panel file loaded first I believe the signal logic began triggering as turnouts and sensors were updated during Interrogation.  So now we had JMRI creating lots of signal traffic to add to the large amount of Interrogation traffic.

I'd still like to keep the "simple" solution on the table: when JMRI needs to create a new turnout in response to a OPC_SW_REP, JMRI should set the State of the new turnout per the state in the OPC_SW_REP regardless of whether it is an OUTPUT Level or INPUT Level OPC_SW_REP.

devel-bobm, 2018-03-26:
@n3ix,
Seems like we're peeling an onion, and finding new layers to a complex set of problems...

devel-bobm, 2018-03-26:
@n3ix,
The problem is summarized by the following experiment:

I start JMRI with a LocoNet Simulator connection (only), without any panels loaded, and use the "Send LocoNet Packet" tool to send eight unique <OPC_SW_REP> messages to encode the eight combinations of "input/output" mode, "C/I" bit and "L/T" bit.
[image]
I open the Turnout Table, and get the following:
[image]

Your concern is the `Turnouts` with "Unknown" values in the Turnout Table.

Why do we get "Unknown"?

It is important to note that the Turnout Table shows only the `commandedState` of the `Turnout`, and not the `knownState` of the `Turnout`.  

But JMRI (currently) treats the "input" mode messages as affecting only the `knownState` of the `Turnout`, if and only if the "feedback" mode of the turnout specifies either "Indirect" or "Exact" feedback.  The code _never_ updates the `commandedState` based on "input" mode messages.  Since the `Turnout` wasn't present before the message was received, it was "created-on-the-fly" with a default feedback mode of "Monitoring", which only looks at the `commandedState`.  So the Turnout Table shows "Unknown".

devel-bobm, 2018-03-26:
A New experiment reveals a potential solution: simply change the creation of `LnTurnout` objects to assume "Indirect" feedback instead of "Monitoring" feedback.

When I make that change, and send the same eight LocoNet messages:

[image]

I get the following Turnout Table:

[image]

This looks like a SIGNIFICANT improvement for this specific set of conditions, where `Turnout`s are created as a result of receipt of <OPC_SW_REP> messages of various types.

Of course this probably "squeezes the baloon" and makes for a problem once the panel is loaded, for any case of a traffic-created `Turnout` with "Indirect" feedback that gets changed to "Monitoring" feedback when the panel is loaded, since the `commandedState` hasn't been updated yet...

If this is deemed to be a "viable" option going forward, I think it might make sense to apply it as a "connection preference" rather than a global default, and default the preference to "Monitoring" to retain current behavior.  Users "in-the-know" would set the preference to "Indirect", if appropriate for their situation.

devel-bobm, 2018-03-26:
@n3ix,
Do you have a JMRI panel and a corresponding LocoNet Log which shows this problem?  If so, please attach to this issue, or e-mail me off-list.

Or a test-setup where you can reproduce the problem?  If so, perhaps you can run with a modified JMRI build which implements this "workaround".  Do you have a JMRI code "build environment" where you could make the change and run the code?  Or would you need a pre-built package created?

devel-bobm, 2018-03-26:
It should be noted that various bits of JMRI code, and jython scripts, may use a `Turnout` object's `commandedState` and its `knownState` in a wide variety of ways, so this experimental solution may just "mask" other potential operational problems.

KenC57, 2018-03-26:
Bob M,
I would say a given connection should have a default create mode, that I
believe is in place. In the case of these specific commands causing the
creation of the turnout, it sounds like that implies a specific feedback
type. If so, for LocoNet, then it should create with whichever feedback type
makes sense for that specific turnout. I would not say that it should ever
change an existing turnout feedback type. It may want to log a warning
(once) that the message received implies the feedback type is wrong if there
is a difference.
-ken c

devel-bobm, 2018-03-26:
@kenc57, The current head-of-master c'tor for `loconet.LnTurnout` ALWAYS defaults to "Monitoring" feedback.  The problem in the specific case brought by @n3ix is that the `Turnout` object is being created **before** the panel loads, so JMRI **does not know** the "feedback type which makes sense for that specific turnout".  

My experiment shows that being able to change that default feedback type used when creating a `Turnout` (in a connection-specific way, _not_ a system-specific way) _may_ resolve some of the `Turnout`-creation-specific issues pointed out in the original post in this issue by @n3ix .  

It does not, however, identify any other "knock-on" effects on things like Logix and Signalling and Jython scripting.  Those effects could possibly be debilitating.

Key to the whole issue is the difference between `commandedState` and `knownState`.  What I'm proposing addresses the `knownState` for the particular LocoNet messages which @n3ix is dealing with.  My proposal **does not** address the `commandedState` part of the equation.  _The effects of that are unknown to me_.  If we can "get away" with ignoring the `commandedState` part of the equation, then the proposed fix can provide a low-impact-to-the-user solution.

If we must address the `commandedState` part of the equation, then we will get involved in non-trivial changes to existing logic.  I am wary of making such changes as the "knock-on effects" can be sigificant and much more difficult to predict.

devel-bobm, 2018-03-26:
@KenC57, I meant to note that when a panel is loaded, if the `Turnout` already exists because of the LocoNet traffic, I _assume_ that the existing `Turnout` will have its "Feedback" mode "overridden" by what's in the panel.  That is a potential knock-on effect that needs to be reviewed.

n3ix, 2018-03-26:
@devel-bobm  Here is the Loconet Monitor and the Turnout table after a JMRI restart with no Panel loaded.  LT50 and LT52 reported with EXACT feedback, the other 3 turnouts used OUTPUT Levels.  Is this what you were looking for?  Hopefully the images come through.

[image]

[image]

n3ix, 2018-03-26:
@devel-bobm   It doesn't look like your test of using Indirect as the feedback setting handled the "Aux Input is Closed" feedback response (your test case LT16).  That is a valid EXACT feedback message for a Thrown turnout.  

Also noted the treatment of LT17.  "Switch Input is Thrown" EXACT feedback indicates that a turnout has just left the Closed position and is moving (towards the Thrown position).  So the inferred command state should probably be "Thrown".

In response to Interrogation, I only send "Aux Input is Closed" or "Switch Input is Closed" EXACT feedback messages.  This avoids the ambiguity issue with the "...Input is Thrown" messages.

devel-bobm, 2018-03-26:
@n3ix,
If I start a LocoNet Simulator run with my modified `LnTurnout` default feedback mode, and issue the <OPC_SW_REP> messages for turnouts LT50 and LT52, I get:

[image]

This is an improvement for LT52 ("switch-input" mode), but no help for LT50 ("aux-input" mode).  Presumably LT50 is an example of "exact" (two-sensor) feedback; the JMRI code for "aux-input" messages is to ONLY update the `knownState` when feedback type is "exact".

So I changed the default feedback upon creation to "Exact".  Then I sent the 8 basic LocoNet messages again, and got:
[image]

If it is known that all `Turnout` objects use "Exact" feedback, then this would seem like a reasonable solution.  

This finding supports the idea of a user-selectable default feedback type for the connection - so long as it includes all feedback types supported for the connection.  But if there are any `Turnout` objects which use feedback that is different than the default setting, there will be some chaos.  And if there are any `Turnout` objects that control outputs other than physical switch points, where "feedback" may not be appropriate, this may also cause some issues.

Additional experiments with LocoNet-created `Turntout` objects and subsequent panel loading shows that changing the `Turnout` object's feedback type can cause the Turnout Table-displayed state to change from known values to unknown values.  Presumably this problem relates to whether or not the `knownState` or `commandedState` has been initialized, and whether the feedback mode relies on `knownState` or `commandedState`.

This leads me to believe that the actual `Turnout` feedback mode **needs to be established** before the `Turnout` state is updated based on <OPC_SW_REP>.  Furthermore, the original problem suggests that `Turnout` state **needs to be established** before Logix and Signalling logic do their thing.

devel-bobm, 2018-03-26:
The previous experiments make me think that we need some sort of "gatekeeper" to ensure that the `Signalling` and `Logix` cannot do their thing before the `Turnout` state is known, and we cannot update the `Turnout` state (via interrogation) until we know the actual feedback type of each `Turnout` object.  

In other words, if we load the panel's `Turnout` and `Sensor` objects, but delay loading Logix and Signalling objects, then "interrogate" to update the actual `Turnout` and `Sensor` state values, then load the Logix and Signalling objects, we would be in a much better position to run trains.

This seems like a **non-trivial** change to JMRI functionality.  And it may justify another github "Issue".

devel-bobm, 2018-03-26:
> The previous experiments make me think that we need some sort of "gatekeeper" to ensure that the `Signalling` and `Logix` cannot do their thing before the `Turnout` state is known, and we cannot [...]

And so I re-worked my experiment for "exact" feedback - results are posted.

It is important to note that this experiment is only effective if every turnout implements the same feedback mode, and the "default" is set for that feedback mode.  If there are multiple "feedback" modes implemented, things can fall apart quickly, once Logix and/or Signalling begin analysis based on incorrect `knownState`.

KenC57, 2018-03-26:
Many times Logix expects that 'unknown' is a state for sensors/turnouts.
Granted it is as a startup state. If your layout has these possibilities,
you can code accordingly to act until (sometimes quite a bit later) these
statuses become valid. Example on LocoNet, the startup poll can take 10-20
seconds before all eight commands get processed to report status.

Key I'd see is that in your one example showing "Unexpected value: 0", that
should have mapped to unknown (on inconsistent), if the original code was
not known to be closed/thrown. I'd say delay should not be considered.
-ken c

n3ix, 2018-03-26:
@KenC57 I've been working with a large club with many hundreds of turnouts and many hundreds of sensors.  The traffic level during Interrogation is high enough that it's hard enough to get through without missing one or more turnouts/sensors.  Getting additional loconet traffic from JMRI for Signaling or from Logix during the Interrogation just makes things worse.  The Signaling messages then fill the command station queue and it starts sending Long ACK rejections so JMRI responds by sending retries which generates more traffic.   If JMRI held off with all that until the Interrogation completed it would help.  Maybe they should just delay loading their Panel file for a minute or two - I think that might do the trick?

devel-bobm, 2018-03-26:
I'm thinking that a workaround might be possible, without any JMRI code changes (including the ones implied in my previous posts to this Issue).  It all revolves around a carefully-crafted set of "start-up actions" in the JMRI "Start Up" preferences.

Here's the "start-up action" sequence I'm thinking of:

- Use the "Pause" "start-up action" to provide a layout-specific delay to hold off execution of subsequent "start-up actions" while the JMRI initial "interrogation" and replies have been sent.  Note: more comments related to this, below*.

- "Open file" of an ".XML" that contains **only** the Turnout and Sensor definitions.  Logix and Signalling **must not** be in this ".XML" file.  This establishes the correct layout-specific "feedback" modes for Turnouts, and associates any "random" sensors used as Turnout feedback sensors to their Turnouts.  Note that Turnout/Sensor states will not necessarily be "correct" at this time because `knownState` is only updated when a state-change event happens, and the LocoNet interrogate replies which have already happened were evaluated based on the default `Turnout` feedback mode of "Monitoring".

- "Run script" a custom .py that issues the LocoNet "interrogate" messages, with layout-specific pauses **after** sending each of the 8 "interrogate" messages to give plenty of time for accumulating all the responses to the "interrogate" request.  It is **critical** that the delay(s) be **after** the interrogate request message, to ensure that the replies are allowed to be processed before the next "interrogate" message and before the next "start-up action" begins.

- Load an ".XML" that contains the Logix, SSL, Signal Masts, Signal Heads, Panels, etc.

- Perform any other layout-specific start-up actions like opening a "fast clock" display, opening a "Turnouts Table", etc.

* This process might only get part-way there, since JMRI automatically sends the "interrogate" messages on start-up.  Those initial "interrogate" messages may cause confusion with the newly-created `Turnout` objects.  If so, it might be necessary/useful to be able to "disable" the JMRI automatic "interrogate" messages via a connection preference.

This workaround is predicated upon the idea that the individual "start-up actions" will be sequentially started, and one "start-up action" must complete, in entirety, before the next starts.  If this is not already the case, it seems that the "Pause" "start-up action" is intended to provide that sequencing feature, so it could be used to enforce the needed sequencing, if necessary.

@n3ix, @bobjacobsen, @KenC57, thoughts?

bobjacobsen, 2018-03-26:
I think the conversation has diverged away from trying to address a specific need.

Let’s drop back to this point:

> I'd still like to keep the "simple" solution on the table: when JMRI needs to create a new turnout in response to a OPC_SW_REP, JMRI should set the State of the new turnout per the state in the OPC_SW_REP regardless of whether it is an OUTPUT Level or INPUT Level OPC_SW_REP.

_Except_ for the case of mis-configured hardware, is there any reason to believe that this is not a good thing to just do?

devel-bobm, 2018-03-26:
@bobjacobsen,
>_Except_ for the case of mis-configured hardware, is there any reason to believe that this is not a good thing to just do?

_That is exactly the problem_ - hardware IS mis-configured (well, un-configured, really) in the original use-case presented in this issue. From the initial post (bolding added):

>**Panel file not yet loaded** and the Turnout is not yet in the turnout table. JMRI receives an Input Levels Turnout Sensor Report.
Result: The Turnout is created, the feedback mode is set to MONITORING, and the State is set to UNKNOWN.

Perhaps I've missed something somewhere along the lines, but I think that the issues I present below add up to plenty of reason that this "simple" solution is not "the right thing to do".

Specifics:
Per @n3ix, in the troublesome cases, the .XML hasn't been loaded (or is not completely loaded) at the time that the LocoNet message is seen that causes `Turnout` creation.  Because of this, the `Turnout` object is created with a default (wrong) feedback type.  In my analysis, this is one of the root causes of the problem!

I'm trying to avoid changes to the logic of how we treat `Turnout` objects when <OPC_SW_REP> is received.  I see that as risky, at best, as it is difficult to predict the effect on other JMRI users.

If we go ahead with logic changes, then the changes I see would require being able to tell that the object was just created, and be able to _infer_ the correct feedback type, solely from a single LocoNet message.  There are feedback modes which can be uniquely inferred from some interrogate reply messages.  But there are also interrogate reply messages which can be used by either of two "feedback" modes, so it is impossible to guess which one applies.  The "guessing game" is as likely to be wrong as right.  What difficulties will this cause?

The logic changes are also entangled with the `knownState` versus `commandedState` problem.  Since the `knownState` and `commandedState` values are set at the event occurence (in this case the reception of a LocoNet message), and are "feedback-mode-dependant",  if you have guessed "feedback" mode incorrectly, you are stuck with the "bad" state data until you have changed the "feedback" mode (i.e. XML evaluation finally gets to the turnout) and the a subsequent event occurs which causes re-evaluation. 

A secondary problem, also reported by @n3ix, is that when Logix and Signalling and SSL are loaded while the "interrogation" process is ongoing, internal state is constantly changing, and this "floods" LocoNet with messages to drive signal aspects based on those ever-changing intermediate states, which have to compete with the "interrogate" reply messages, which means some get lost due to excessive retries, which means unknown sensor and turnout states which means incomplete states for signal logic, which gives misleading and/or meaningless signal aspects, ...

I'd really like to avoid all of this complexity!

bobjacobsen, 2018-03-26:
I agree that you're bringing up complexity, and I agree that it's good to avoid it.

So I ask again: What's wrong with _changing_the_code_ to do what was originally suggested?  I don't see anything in the post above that addresses that.  It just reiterates that the change isn't present, so the existing code isn't doing the right thing.

What will the suggested change get wrong?

devel-bobm, 2018-03-26:
> What will the suggested change get wrong?

The `currentState` of the `Turnout` object, at minimum.  One cannot conclusively determine the "feedback" mode from certain <OPC_SW_REP> "input mode" messages.  So one cannot determine conclusively the correct `currentState` of the `Turnout` object from that <OPC_SW_REP> message.

bobjacobsen, 2018-03-27:
The feedback mode will eventually get set when a panel file is loaded, and messages after that will be interpreted in terms of that. 

What's the harm of having a `[B1 3F 50 2F] Turnout LT50 () Aux input is Closed (input off)` set the state of the LnTurnout that it creates? 

Line 253 of LnTurnout is checking the 0x40 bit in 3rd message byte, finding it not zero, and routing to line 290 where 0x20 bit is checked.  That's found to be zero, so it goes to code that (finally) checks for EXACT feedback and processes it.  If the feedback isn't EXACT there, it's ignored.

So the question becomes: 
 - Is there ever a time this message will be received from properly configured hardware such skipping the check for EXACT when first creating the turnout will result in a wrong state?

n3ix, 2018-03-28:
@bobjacobsen  Per the DS54 manual, there are two configurations that use the OPC_SW_REP INPUT Levels message.  Digitrax calls these modes "Standard" and "Exact".  Standard uses 1 sensor wired to the "Switch" input that is Closed when the Turnout is Closed.  Exact uses 2 sensors, one wired to the Switch input that is Closed when the Turnout is Closed and one wired to the "Aux" input that is Closed when the Turnout is Thrown.

Here is a list of the possible messages and what I believe they mean in each of the two modes.

1. Switch Input is Closed
Standard Mode - the Turnout is in the Closed position
Exact Mode - the Turnout is in the Closed position

2. Switch Input is Thrown
Standard Mode - the Turnout is in the Thrown position, 
Exact Mode - the Turnout is not in the Closed position

3. Aux Input is Thrown
Standard Mode - not used
Exact Mode - the Turnout is not in the Thrown position

4. Aux Input is Closed
Standard Mode - not used
Exact Mode - the Turnout is in the Thrown position

Proposed JMRI usage
Message 1 has identical meaning in both modes so it can be used.
Message 4 is only sent in Exact mode so it can be used.

Message 3 is only sent in Exact mode.  It is typically sent when the Turnout leaves the Thrown position and is moving towards the Closed position, and signifies that the Turnout is not in either the Closed or Thrown position.   This message could probably be used during Interrogation by inferring that the command state is Closed and the net state is Inconsistent (which is what JMRI sets it to when the turnout is defined in the Table as using Exact feedback).

Message 2.  This message effectively carries the same information in both modes - the turnout is not in the Closed Position.  The difference is that there is only one position sensor in Standard mode so this message is taken to mean that the Turnout is in the Thrown position.  In Exact mode it is typically sent when the Turnout leaves the Closed position and is moving towards the Thrown position, and the meaning is that the Turnout is not in either the Closed or Thrown position.  

From a Interrogation standpoint, I would be happy whether this message is ignored or is treated as the equivalent of Message 4.

Summary.  As best I can tell at least 2 of the 4 possible OPC_SW_REP INPUT Levels messages can be used during Interrogation to initialize the Turnout state when the Turnout in question does not yet exist in JMRI.

n3ix, 2018-04-07:
From my standpoint the info above provides a well defined opportunity to improve the overall system behavior during the JMRI Interrogation at startup.  Does the lack of comments over the past 2 weeks indicates that no code changes are forthcoming?  I ask because there is a firmware update coming for the QuadLN product.  If these changes were going to be made in JMRI before the 4.12 release then I could reflect that in the QuadLN update and reduce the Loconet traffic it generates during Interrogation.  Thanks.

bobjacobsen, 2018-04-07:
I have the code change in a branch, but I haven't been back to hardware to test it.  Are you set up to be able to do that?

n3ix, 2018-04-07:
@bobjacobsen That's great, thanks.  Sure I would be happy to test .  Not sure I can build JMRI at the moment - haven't tried it for a year or so, but if needed I'll work through that.

bobjacobsen, 2018-04-08:
Thanks!  It's in the `bobjacobsen:loconet-turnout-init` branch, also how visible in PR #5214.  I've set that to "WIP" while we give it a bit more testing.

Also, I've added the ` Turnout LT64 () Aux input is Closed (input off).` example of a type-4 message from above to the automated tests. If you have time, getting examples of types 1, 2 and 3 to include in the tests would be nice. 

Thanks again.

bobjacobsen, 2018-04-11:
Closed by #5214

closed 2018-04-11 by bobjacobsen