JMRI/JMRI #5146: NullPointerException on LnPacketizer

LexNetAT, 2018-03-25:
The `XmtHandler` throws a _NullPointerException_ if no _controller_ is used.
The expression `!controller.okToSend()` should be change to the function `isXmtBusy()`:
```java
// input - now send
try {
  if (ostream != null) {
    // if (!controller.okToSend()) {
    if (isXmtBusy()) {
      log.debug("LocoNet port not ready to receive"); // NOI18N
    }
```

devel-bobm, 2018-03-25:
I don't think that is the intention behind this check.

The check was intended, if I recall correctly, to determine if the connection was properly established and was ready to operate.  It was NOT intended as a check to see if a controller was available.

If I recall correctly, there are some LocoNet-based connection types which are not necessarily "okToSend" at start-up, and thus would not be "ok to receive", regardless of whether or not the transmit path was busy.

Based on the above recollection, I do not think this change should be made without careful review by those who understand the connection set-up processes, including a variety of LocoNet-based connection types.

@pabender, @bobjacobsen, can you give some cycles to this, and/or recommend a developer who has appropriate knowledge?

bobjacobsen, 2018-03-25:
@LexNetAT Thanks for looking into this.  

Under what circumstances are you seeing the NullPointerException getting thrown?  There might be a problem in the initialization sequence which should make sure the controller reference is present by here.

pabender, 2018-03-25:
In this case it probably is ok to just call isXmtBusy:

https://github.com/JMRI/JMRI/blob/4aadfc68813602382664bdd5d67b1e6dcac3576c/java/src/jmri/jmrix/loconet/LnPacketizer.java#L119-L126

The default implementation of isOkToSend for Loconet always returns true, so they only reason you wouldn’t be able to forward something to the port is if the controller doesn’t exist.

devel-bobm, 2018-03-25:
@pabender,

I'm not comfortable with the assumption that future implementations will follow the pattern established by current implementations.

pabender, 2018-03-25:
@devel-bobm remember that isOkToSend is used for flow control, so if there is no controller defined, it is always going to be ok to override isOkToSend here ( no controller means we are not connected to anything... )

Pugwash1, 2018-03-25:
@LexNetAT 
What led up to this happening? Was  it during startup or later? I think we need to look at how this happened and not just hide the fact that it did.

LexNetAT, 2018-03-25:
I would like to use some code as a library in a project of mine, and I would prefer to use it without any modification. I want to process and send LocoNet messages without the complete framework and class instances around it.
By the way, the controller instance is used here only at two places, just for the data flow - once in the method correctly with the treatment of zero and another time without. As a developer of many years, I avoid doubling code if a function is already doing the job correctly. I think the function isXmtBusy was added later and therefore the piece of code is not adjusted.
Therefore, just my request.

bobjacobsen, 2018-03-26:
@LexNetAT Thanks for looking into this. I've put a fix in #5160, comments welcome.

bobjacobsen, 2018-03-26:
#5160 has been merged, so I'm closing this.

closed 2018-03-26 by bobjacobsen

LexNetAT, 2018-03-26:
Thanks, that looks promising.