JMRI/JMRI #5147: Track power is turned on automatically if this command is run: "[EF 0E 7B 01 7F 78 5E 47 7C 18 70 11 0A 75]  Request Fast Clock is"

ClockLoop, 2018-03-25:
When I tell jmri to power off the track with the power on/off button.

This command turns track power back on.
20:27:56.555: [EF 0E 7B 01 7F 78 5E 47 7C 18 70 11 0A 75]  Request Fast Clock is  Running, rate is 1:1. Day 24, 20:27. Last set by ID 0x0A 0x11 (1297).
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available.

IS this normal?
Is this because I have a "clock" running on my layout?    
This also causes the layout to power up automatically when using any throttle to power it off.
Can't the clock functions work without layout power?
Do i disable the fast clock just so i can control the power properly?

I don't know the normal functionality of this data sequence, im guessing it contains some command to turn track power on, and is simply a bug.

ClockLoop, 2018-03-25:
I do want to say, thanks for this excellent software, we would all be lost trying to make all this happen if it weren't for jmri and all the code support!  Thank you!

GREAT FUN!  Thank you.
https://www.youtube.com/watch?v=GG6KsRWB-A4

https://www.youtube.com/watch?v=a2pDylzGESI

bobjacobsen, 2018-03-25:
I don't know very much about the LocoNet fast clock support, but it looks like JMRI periodically writes to the "fast clock slot" when the fast clock is set to one of several conditions:
```
    private void setClock() {
        if (setInternal || synchronizeWithInternalClock || correctFastClock) {
```
If you just need a local clock, maybe there's a combination of options on the LocoNet clock frame that'll keep it from writing.

I don't know enough about the messaging to know whether JMRI should be setting that bit differently. If so, that's something that the SlotManager should handle, perhaps by listing to the LocoNet power state.

devel-bobm, 2018-03-25:
I do not believe that this is a JMRI problem, but instead a question of "how the DCS240/DCS210 actually works".  I presume that this is seen with a DCS240, given some of your other issues of late.  How to tell if this not a JMRI problem and is "expected" DCS240/DCS210 behavior?  Read on.

I suspect (but cannot confirm at this time) that this relates to DCS240/DCS210 OpSw43.  When "thrown", this OpSw will "Enable LocoNet update of command station's track status".  "Closed" disables this feature.  Default is "thrown".

I am guessing that this OpSw -means- that the command station will pay attention to the value in the <TRK> byte of the slot write message you provided.  In that message, <TRK> is 0x47, which, in part, says that the DCC packet stream is being actively generated and is not "paused".  Note in your LocoNet Monitor snippet the part which says "Track Status: On/Running;" - so your LocoNet request may actually be making a fast clock request that the command station additionally takes as a request to change the track signal generation state.

If, when track power is "off", and you send the (slightly) different LocoNet message 
`EF 0E 7B 01 7F 78 5E 400 7C 18 70 11 0A 75`
does the behavior change so that the track power remains off?

If you change that OpSw and turn off track power and send the original LocoNet message, does the behavior change?

devel-bobm, 2018-03-25:
@bobjacobsen,

LocoNet's "Fast Clock" is not particularly consistently implemented, especially when bringing 3rd-party fast clocks into the picture.  If there is a "standard" way to implement it, I haven't seen it...

if DCS240/DCS210's "slot write" behavior is notably different from previous generations, as I suspect, this does imply that JMRI will need to "properly" follow track status and make it available to the `LocoNetSlot` level.  I think that, rather than relying on `SlotManager`, I'd rather see `LnPowerManager` follow the track power state, solely based on the idea that this is `LnPowerManager`'s basic purpose in life.  And given some of the historical sillyness in LocoNet power state, I would rather see this sillyness dealt with in one place, rather than two or more.

bobjacobsen, 2018-03-25:
> rather than relying on SlotManager, I'd rather see LnPowerManager follow the track power state, solely based on the idea that this is LnPowerManager's basic purpose in life. 

Agreed.  But when it's time to write the slot, which AFAIK is how the title message gets sent, that comes from (or should come from) the SlotManager and the LocoNetSlot object.  They're the ones who should get that write correct, by either listening to the PowerManager or just consulting it directly at write time.

ClockLoop, 2018-03-25:
@devel-bobm   im guessing you meant...

EF 0E 7B 01 7F 78 5E **40** 7C 18 70 11 0A 75

This works, track power remains off.

ClockLoop, 2018-03-25:
@devel-bobm 

> If you change that OpSw and turn off track power and send the original LocoNet message, does the behavior change?

Yes, the power remains off.

Pugwash1, 2018-03-25:
There are actually 2 issues. 1 - writing the slot because we are using the internal clock without being ask to sync the external. 2 - writing the slot with reading it first.

ClockLoop, 2018-03-25:
Heres how i have it configured....
[image]

When i check the boxes, I see that a command is issued on loconet, but it dosen't change at all.
Perhaps this and change on the clock settings, requires a jmri restart to implement?

Pugwash1, 2018-03-25:
Yes, and if you uncheck "synchronize internal....." It still will., but shouldn't.

devel-bobm, 2018-03-25:
@bobjacobsen 
"Agreed. But when it's time to write the slot, which AFAIK is how the title message gets sent, that comes from (or should come from) the SlotManager and the LocoNetSlot object."  I agree with this statement - `SlotMon` would get it for free.

FYI -  I interpreted the initial message to potentially be independant of JMRI's current fast-clock functionality, as @ClockLoop did not say that "he saw JMRI's fast clock functionality send the message".  Instead he said that he "sent" the message.

From that I extrapolated to think that this may affect non-JMRI fast-clock functionality as well as JMRI functionality.  

That is part of the additional investigation I would want to do with respect to this issue - look at differences in behavior with other command stations and with 3rd-party fast clocks.

devel-bobm, 2018-03-25:
@ClockLoop wrote:
"Yes I have a DCS240.

"@devel-bobm im guessing you meant... EF 0E 7B 01 7F 78 5E **40** 7C 18 70 11 0A 75".

No, I meant **44**.  Sorry!

The lowest two bits reflect the track status.  The other bits do not reflect (main) track status, and are not, so far as I can tell, relevant to this discussion.

devel-bobm, 2018-03-25:
@ClockLoop wrote:
"What generates that command, jmri or the dcs240?"

Any LocoNet agent can generate the "write fast clock slot" message.  JMRI can.  A Digitrax Throttle can.  A third-party "fast-clock" device can.   

I am fairly certain (but would need to investigate) that command stations do not send "EF 0E 7B" "fast clock slot write" messages.  My investigations with command stations earlier than DCS210/DCS240 showed that the command stations would generate only "E7 0E 7B" "fast clock slot report" messages (in response to fast clock slot queries), but would not generate fast clock slot writes.  I assume this is still true with DCS210/DCS240 and do not expect it to behave differently.

devel-bobm, 2018-03-25:
@Pugwash1 wrote "Yes, and if you uncheck "synchronize internal....." It still will., but shouldn't.".

That may be the reason why I've never found the JMRI LocoNet fast clock functionality to be usable with the way we use the fastclock on our layout.  It needs to be run-to-ground (#5149.)

devel-bobm, 2018-03-25:
I find that the DB150 I have available (unknown firmware release) has inconsistent behavior whtn the writing the <TRK> byte of the fast-clock slot with "track bits" that are not consistent with actual track power.

I ensured that DB150 OpSw 43 was "t"hrown (i.e. factory default.  Then I sent the sequence:
1. Write fast-clock slot with <TRK> byte as 0x07.
2. Query the fast-clock slot
3. Write fast-clock slot with <TRK> byte as 0x06.
4. Query the fast-clock slot

If I use a throttle to set track power "on", then send the above sequence, the DB150's "track status" lamp stays lit throughout the sequence.

If I use a throttle to set track power to "off", then send the above sequence, the DB150's "track status" lamp stays lit throughout the sequence.

But if I query the fast-clock slot during the sequence, I find that the fast-clock slot reports the <TRK> byte last written.  And if I query slot 0 (instead of 0x7b) between the slot writes, the <TRK> byte of that slot also follows the <TRK> byte written to slot 0x7b.

The above experiments were repeated with OpSw 43 set "closed", with the same result.

This appears to be another indication that one cannot use the information in the <TRK> byte of a slot report to conclusively indicate track power state.  This is consistent with previous conclusions by JMRI developers that one cannot simply rely on the <TRK> byte of a slot report to determine track power state at JMRI start-up.

This continues the basic theme with LocoNet:  for at least some command stations, JMRI will NOT be able to come up with a clean solution, across all command stations, to the <TRK> byte inaccuracy problem.

What can JMRI do?  Perhaps we can use the command station type to _assume_ that the <TRK> byte is accurate.  Or perhaps we can add an "Advanced Preference" or two to LocoNet connections, to allow the user to specify whether his command station properly reports/honors the <TRK> byte in slot reports/queries.  And perhaps we can add a LocoNet "tool" that helps the user figure out the correct setting(s) of that new Preference setting(s) by issuing a sequence of LocoNet operations and inquiring about the "track status" lamp color/lit-ness after each LocoNet operation.

bobjacobsen, 2018-03-25:
I think PR #5151 addresses the original topic here.

bobjacobsen, 2018-03-26:
@ClockLoop Are you able to build and try the change in PR #5151? If you can test it, I'd appreciate it. It addresses the original problem in this Issue by using the power state from the associated LnPowerManager to set the bit before writing to the clock.  If you're not set up to build and test, no problem, will just merge and we can sort that out later.

ClockLoop, 2018-03-28:
@bobjacobsen 
Perhaps it high time I do it.  
http://jmri.org/help/en/html/doc/Technical/NetBeans.shtml
Looks like you guys provide a good instruction on getting it going...
I'll try to get it going asap.

ClockLoop, 2018-03-28:
Ok, i built the jar file and replaced mine, is that good enough to test with?
If so, then it didn't work, my track power still turns on when the initial command is run and OpSw43 is set to dcs240 default.

[image]

Do I need to build from a different location?

bobjacobsen, 2018-03-28:
Thanks for working on this!  The process for building from a branch (i.e. the one where I've put this change) is a bit different from the standard instructions.  (The following is NetBeans 8.2, but I don't think it varies with versions much)

- Under "Team" -> "Remote" select "Fetch" (not Fetch from Upstream)

[image]

- Fill out the form to specify my repository (After the 1st time you do this, you'll be able to select "bobjacobsen" in the top choice)

[image]

- On the next screen to select a branch, you can either select all of them (there's a button) or find and select just the "fast-clock-power-bit" branch that holds this change:

[image]

- Click "Finish"

At this point, you've pulled my changes onto your local machine, but you're not using them yet.  To have those files appear locally for NetBeans to build:

- Back on the main menu, select "Team" -> "Checkout" -> "Checkout Revision...". On the window you get, click "Select":

[image]

- In the upper left, find the imported branch you want, which is "bobjacobsen/ast-clock-power-bit". Either click to highlight (show) and then click "Select", or just double click on it.  (Don't worry about the upper right or the lower box, they'll be automatically set right)

[image]

- Back on the "Checkout Revision ..." window, make sure the "Checkout as New Branch" is _not_ checked, and then click "Checkout"

Now you've got a copy of my changes as your checked-out source.  You should do a "Clean and Build Project" to make sure that you get a consistent compilation, and then you should be able to run it.

Well, that was all a bit longer than I thought it would be (it's two commands on the command line, which I usually use), but hopefully it's clear enough.  Thanks again for testing this.

Important note:  When you want to get back to the _standard_ JMRI version, from the main repository, you need to check out the "master" branch.  The process is similar, and simpler.  Go to "Team" -> "Checkout" -> "Checkout Revision...", type just "master" (no quotes) in the "Checkout Selected Revision" box, and click "Checkout"

ClockLoop, 2018-03-28:
SUCCESS!  You fixed it.
...we get signal...
...main screen turn on...
...how are you gentlemen, all your base are belong to us.

[image]

closed 2018-04-11 by bobjacobsen