JMRI/JMRI #5149: Fast Clock (LocoNet) internal/external sync problem

devel-bobm, 2018-03-25:
(Issue extracted from #5147)

The JMRI LocoNet-based "Fast Clock Setup" tool provides a checkbox "Synchronize Internal Fast Clock and LocoNet Fast Clock".  That checkbox either does not do anything useful, or does something other than what is expected.

KenC57, 2018-03-25:
Bob M,

Did you check it with the choice of who is the master? The idea as I thought
I understood was that worked with the 'who is in charge' so it would then
keep them in sync one direction or the other. Just setting that checkmark
isn't enough.
-ken c

devel-bobm, 2018-03-25:
@KenC57,

Yes, I have played with the "master" setting in addition to the "synchronize" setting, a few different times over many years, with minimal effort and minimal thought and minimal recordkeeping.  Perhaps I should spend more time working with this, in my copious spare time.

Controlling "who is master" can be a challenge on LocoNet.  So far as I can tell, there is no way to turn off a command station's "fast clock master".

And when one is "not allowed" to change the setting of a 3rd-party LocoNet device's "I am fast-clock master" setting, one must live with the fight between the command station and the "3rd-party fast-clock master".  This is an issue without introducing the JMRI fast clock into the system.

devel-bobm, 2018-03-25:
@KenC57, Your description is a good match to what I remember.

But with a 3rd-party fast-clock wanting to be fast-clock master, and the command station wanting to be fast-clock master, things get unpredictable.  

And with JMRI wanting to be fast-clock master and the command station wanting to be fast-clock master and a 3rd-party fast-clock wanting to be fast-clock master things get out-of-control.  An experiment today of exactly this configuration generated a "LocoNet message storm" every time something tries to assert its fast-clock mastership.  Obviously, step one is to disable the 3rd-party fast-clock as fast-clock master.  For one layout, the owner has not yet allowed me to do that.  And this combination of effects prevents me from even putting a "fast clock" on the screen to monitor the LocoNet "fast time"!

Pugwash1, 2018-03-27:
@KenC57 @devel-bobm I have spent the day testing the idea as explained in the loconet PE. My understanding is that each device that uses time "syncs" with the CS. maintaining its own time between syncs. It does this by listening for 'E7' messages (slot response) for the fast clock slot, if it doesnt see one for some random time between 60 and 90 secs it asks for a slot status for the FC slot. Everyone listening syncs to that and resets their timers, meaning only one device polls per system.
Soooo. I set the fast clock CLK_RATE  to zero to freeze the clock on the CS. with an 'EF' slot write (so I could see errors). Then preempting anyone asking for a slot refresh issued an 'E7' (FC slot response) every 50 secs. The DT402's both stopped asking for the slot as they synced on the 'E7' and everything was happy, including no slot write, so no weird power offs/ons, prog track busy (since we are not writing). The fly in the ointment is that a device when first plugged in issues the slot read request., which generates an 'E7' response from the CS,. so occasionally the throttles, they both saw and reacted to the sync,  had the wrong time till our next 'E7' message.  
I think this could be the behaviour if we are providing the fast clock but not syncing the hard clock. Effectively becoming master, except for the fly.

devel-bobm, 2018-03-27:
Again, I repeat - a 3rd-party fast-clock does not behave in a reasonable way.

I have tried the same scheme as @Pugwash1 suggests, and it fails with that 3rd-party fast-clock.  As such, that scheme was useless to me, since I'm heavily-invested in that 3rd-party fast-clock.

Pugwash1, 2018-03-27:
@devel-bobm bear with me for a moment. Are there instructions for adding a second one of these clocks? And what does the Loconet traffic look like with a CS, the 3rd party, a throttle with clock and no jmri clock look like?
Currently jmri doesn't follow the rules, it's doing a slot read even if it has seen an 'E7' record recently, spoiling everyone's day.
After @bobjacobsen power fix goes thru we can fix this all up.

closed 2021-04-14 by stale[bot]