JMRI/JMRI #5163: 4.11.4 Panel File now fails to load on positionrelative/dopplerfactor audio error

ModelTrainNut, 2018-03-27:
Saving my layout that worked fine on 4.11.3 with only Layout Editor Memory Variable format changes  now fails to load: 
'Error on line 5017: cvc-complex-type.2.4.a: Invalid content was found starting with element 'positionrelative'. One of '{dopplerfactor}' is expected.'
[Ontario Southern Railway.zip](https://github.com/JMRI/JMRI/files/1851846/Ontario.Southern.Railway.zip)

I uninstalled 4.11.4, installed 4.11.3, reloaded a Panel File saved under 4.11.3, changed a Memory Variable font type, size, and position, saved Panels, reloaded, and no error.

Randy

bobjacobsen, 2018-03-27:
That's a bug.  I'll post a fix shortly, that'll be in 4.11.5 in a few weeks, but in the meantime you can use the "jython/TurnOffXmlValidation.py" script to bypass the error. Just set the Preferences -> Startup to run the script _before_ you load your file.

Sorry for the trouble.

ModelTrainNut, 2018-03-27:
No trouble at all.
Thank you for the quick reply!
Randy

PR #5165 (2018-03-28): Fix audio schema validation problem in JMRI 4.11.4
See issue #5163 for background

Changed:
- dopplerfactor attribute in audiosource element no longer required

Closes #5163
closed 2018-03-28 by bobjacobsen