JMRI/JMRI #5164: Memory leak in tests

pabender, 2018-03-27:
As many of you know, I have been spending quite a bit of time hunting down threads that are not terminated properly in tests.

The vast majority of the threads are in the OpenLCB tests, but I have made significant progress on getting those cleared ( at this point, I have cut those by more than half ).

One of my test machines consistently starts throwing GC errors due to running out of heap after about 20 minutes.  I had initially speculated the threading on the OpenLCB tests caused this, but the GC errors occur even when the OpenLCB tests have been disabled.

Does anyone have any suggestions for how to determine what objects are actually stored in memory with memory useage statistics?

bobjacobsen, 2018-03-27:
> On Mar 27, 2018, at 6:34 AM, Paul Bender <notifications@github.com> wrote:
> 
> Does anyone have any suggestions for how to determine what objects are actually stored in memory with memory useage statistics?

My recommendation is jvisualvm.  

For example, when connected to a PanelPro instance, clicking on “Sampler”, then “Memory”, gives:

Click “Heap Dump” to get a navigable snapshot:

Sort and navigate in the “Classes” view (as above), then select a row and switch to “Instances”:

Pick an instance, any instance, and then use the bottom right panel to track back to who’s referencing it. In this case, that required digging a down a bit (JDOM has lots of internal references), but that JDOM Attribute is eventually found in a DecoderFile 

and I can now double click on that and iterate.  

Bob
--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

bobjacobsen, 2018-03-27:
(redo via web to get images) 
My recommendation is jvisualvm.  

For example, when connected to a PanelPro instance, clicking on “Sampler”, then “Memory”, gives:

[image]

Click “Heap Dump” to get a navigable snapshot:

[image]

Sort and navigate in the “Classes” view (as above), then select a row and switch to “Instances”:

[image]

Pick an instance, any instance, and then use the bottom right panel to track back to who’s referencing it. In this case, that required digging a down a bit (JDOM has lots of internal references), but that JDOM Attribute is eventually found in a DecoderFile 

[image]

and I can now double click on that and iterate.

pabender, 2018-03-27:
Thanks Bob, that helps.

It looks like most of the data retained in the tests by size are two things:
1) Images
2) XML data

I haven't dug down far enough to find whether they xml data is decoder files or panel files.

bobjacobsen, 2018-03-27:
I can take a look this weekend at the XML data. There’s been some work on dropping decoder and locomotive XML while DecoderPro is running, but that might not be happening during tests.

pabender, 2018-03-27:
Sent from my iPhone

> On Mar 27, 2018, at 3:35 PM, Bob Jacobsen <notifications@github.com> wrote:
> I can take a look this weekend at the XML data. There’s been some work on dropping decoder and locomotive XML while DecoderPro is running, but that might not be happening during tests.
> 

That would be helpful.

Thanks!

bobjacobsen, 2018-03-28:
Took a quick look.  
 - [ ] There's a 100MB link in a Roster (DecoderPro) pane that I'll PR shortly.  
 - [ ] There's about 600MB leaked in the jmrit.display tests due to static references, mostly in jmrit.display.palette.ItemPalette; that's going to take longer to fix.
 - [ ] There's a bunch in the XML load tests, but I think some of that is due to the statics in the display code so I'd like to try to fix that first.

More to follow.

bobjacobsen, 2018-03-28:
One thing I've learned: JUnit4 keeps strong references to instances of test classes until it's completely done.  So, if you have a member variable that's storing a reference to some object for your test methods, you should set it to null in the `@After` method.

closed 2021-04-14 by stale[bot]