JMRI/JMRI #5180: Reporter table shows user name for both system and user names

KenC57, 2018-03-30:
Saw this in 4.11.4. Was doing testing related to a user with non-standard system names. If I enter both, the user name shows in both columns. If I only enter the system name, that shows ok. A saved panel file has the right system name. Entering a user name after the initial (system name only), causes it to switch to both columns having the user name value displayed.

bobjacobsen, 2018-04-08:
I don't see anything in the code for the table itself that can cause this, but I could easily be overlooking something.  What kind of reporters are you using?  Where are you "entering both" names?

bobjacobsen, 2018-04-08:
I think #5215 (now merged) fixes this particular case.  @KenC57 could you check please?

KenC57, 2018-04-09:
Yes. The system name and user names now stay correct. So that seem to have fixed it.
-ken c

closed 2018-04-09 by KenC57