JMRI/JMRI #5182: jmrit.Editor dispose() needs work

bobjacobsen, 2018-03-31:
While looking for why the jmrit.display (i.e. PanelEditor, LayoutEditor et al) tests retain several 100's of MB of memory, I found that the "dispose" process for all the editor panels is Really Messed Up.

 - The base class is `Editor`, a child of `JmriJFrame`.  It has no `dispose()` method of its own, instead inheriting the one in `JmriJFrame`

 - Instead, `Editor` has a `public void dispose(boolean clear)` method that
   a) handles (most of) the local resources
   b) except if `clear` is false it doesn't drop the icons that the Editor knows about (WT...?)
   c) and then calls `dispose()`

 - This matters because `Editor` also maintains a static list of `Editor` objects called `editors`.   If `dispose(boolean)` isn't called, either because no dispose is called or `dispose()` is called directly e.g. when auto-closing a window, there's still a static reference to the editor object in the `editors` static list and the memory is retained. That's 821 retained heavyweight `Editor` objects retained during testing, 100's of MB of retained memory, just in `jmri.jmrit.display`. This is a large part of the problem in Issue #5164. And that's without even considering the load&store tests!

It's confusing when an argument is added to a standard Swing method, thereby breaking the expected overloading. It's confusing to repurpose a standard method like `dispose()` to sometimes do one other thing, and sometimes not.  And I have no idea why you'd want to have a window (i.e. an `Editor`) retain some of its contents when you're disposing it.  

There are seven references to `dispose(boolean)`. Six are `dispose(true)` and one (in `Editor` itself) is `dispose(false)`.  The right thing to do might be to
 - [ ] Figure out what the `dispose(false)` is actually doing and implement it another way
 - [ ] Convert the `dispose(true)` calls to `dispose()`, and change the `dispose(boolean)` method to an @Override of `dispose()`
 - [ ] See what tests need to be changed to adapter to this, e.g. `jmri.jmrit.display.EditorScaffold`.

But there might be another approach that'll work better. This package (and sub packages) now has very complicated code with lots of statics and casts across classes, and with only weak test coverage of any kind.  I can manually test general Panel Editor use cases, but I don't use the other three editors, and Layout Editor seems to construct `Editor` subclasses in several ways; it might also dispose of them in multiple ways too, so systematic tests (manual or JUnit) are needed.

pabender, 2018-04-01:
The only thing I can see that dispose(Boolean) does when the parameter is false is retain a list of positionable items, but the objects in the list are all built with references to the editor we are disposing of.

What that means is calling the dispose(Boolean) method with false as a parameter isn’t actually valid, because the objects in the list include a reference to an object that no longer fully exists after the call.

That one call with the parameter set to false is in the changeView method.  That method appears to be used when changing only when changing from a PanelEditor to controlPanelEditor view.  As part of the conversion, the list of positionable items is copied from one panel to the other.  The dispose call deletes the old editor.  As part of the copying process, the editor of the positionable items in the list is changed to the new editor.

Unless I am missing something about the way clear works on an ArrayList, I don’t think we gain anything by holding on to the list in this case.

When I get a chance later today, I am going to see if I can build a JUnit test that tests the changeView method and verifies the new panel and the old panel have the same positionable content.

pabender, 2018-04-02:
#5189 adds a test for the changeView method.  This is run against all 4 editors (with the editor scaffold as the target).

We could add tests to all the editors as well, but changeView only really makes sense for PanelEditor and ControlPanelEditor in practice.

pabender, 2018-04-02:
So it turns out that new test i just added shows the reason why the dispose(false) was in the changeView method.

Rather than creating a new array for the new editor, changeView assigned the value of the existing editor's ArrayList reference to the new editor's ArrayList reference.  As a result,  deleting the existing ArrayList caused the references on the new editor to be lost.

I've already implemented a change that clones the ArrayList rather than just assigning the existing value.  That allows the test to pass with dispose(true) instead of dispose(false).

pabender, 2018-04-02:
so #5189 and #5190 take care of the first item on the list.

#5189 starts the process of taking care of the third item.

#5190 starts the process of taking care of the second item.

pabender, 2018-04-02:
#5191 should complete item 2 and 3 from the list.

This probably needs some manual testing, in addition to the automated testing that is running now.

closed 2018-05-18 by bobjacobsen