JMRI/JMRI #5184: NPE with invalid User Files Location causes startup hang Windows

dheap, 2018-04-01:
When Using a Shared Profile, JMRI initially copies any existing settings for that profile. This means things like User Files Location path may be invalid for that node.

An NPE is then causing an indefinite hang, rather than allowing startup to complete and allowing the user to fix.

```
2018-04-01 10:42:39,709 apps.AppsBase                         INFO  - DecoderPro version 4.11.5ish+jenkins+20180331T1848Z+R239224c starts under Java 1.8.0_161 on Windows 7 x86 v6.1 at Sun Apr 01 10:42:39 AEST 2018 [main]
2018-04-01 10:42:42,485 gui3.Apps3                            INFO  - Starting with profile NCE_Simulator.3ee5400a [main]
2018-04-01 10:42:42,829 node.NodeIdentity                     INFO  - Using jmri-685B35CCB3CA-3ee5400a as the JMRI Node identity [main]
2018-04-01 10:42:42,829 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main]
java.lang.NullPointerException
	at jmri.util.FileUtilSupport.setUserFilesPath(FileUtilSupport.java:546)
	at jmri.util.FileUtil.setUserFilesPath(FileUtil.java:358)
	at jmri.implementation.FileLocationsPreferences.initialize(FileLocationsPreferences.java:35)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:264)
	at jmri.implementation.JmriConfigurationManager.lambda$initializeProvider$4(JmriConfigurationManager.java:260)
```
[session.log](https://github.com/JMRI/JMRI/files/1866026/session.log)
[JMRI.zip](https://github.com/JMRI/JMRI/files/1866027/JMRI.zip)
[JMRI Config Profiles.zip](https://github.com/JMRI/JMRI/files/1866028/JMRI.Config.Profiles.zip)

bobjacobsen, 2018-04-02:
This is a bit odd. Working backwards in the trace, the FileUtilSupport line seems to only be able to through a NPE if the `path` argument is passed as null.

That can only happen if FileUtil gets passed a null.

But that comes from FileLocationsPreferences:35, where `path` comes from `FileUtil.getAbsoluteFilename(userFiles)` and `userFiles` is known to be non-null.

That tracks down to FileUtilSupport .getAbsoluteFilename which says it returns @Nonnull. But that must be wrong?  The return value is from `pathFromPortablePath` which (a) is deprecated, (b) definitely _can_ return null, and isn't annotated either way and (c) only sometimes does a log.warn before returning null.
 
I haven't been able to recreate this.  Could you add a log.warn statement before the `return null` at 1562 in FileUtilSupport and see if that fires and shows the file name?

I suspect that name is perfectly valid, and the trouble is further up in the method where it's comparing with things that haven't been set yet.  But I'd like to see that confirmed.

Thanks

bobjacobsen, 2018-04-02:
Oops, sorry, misread the annotations.  `pathFromPortablePath` isn't deprecated, it's marked to ignore deprecations in things it calls. (That's a topic for another day)

dheap, 2018-04-03:
@bobjacobsen Do you still want me to try the log.warn?

The situation that triggered this relates to #5185 where nodeIdentity is not distinguishing between the Mac & Windows partitions. As a result, we could either have profile.properties showing:
`jmri-implementation.user-files=/Ours/Our Documents/Dave/Dropbox/JMRI/`
or:
`jmri-implementation.user-files=home\:Dropbox/JMRI/`

That's another issue, but the problem at hand is that when an invalid `jmri-implementation.user-files` path was found, JMRI locked up forever at the startup logo rather than continuing and allowing the user to fix.

dheap, 2018-04-03:
Have just realised that inserting the log.warn into a Windows build (where I saw the problem) is not feasible for me at present due to #4628.

dheap, 2018-04-17:
Changing the title of this. The issue is not Windows-specific and can occur in many circumstances. I'll document as I get time. There are also other similar errors that cause startup hang.

dheap, 2018-04-17:
To recreate a similar (but not NPE) issue I encountered on my Ubuntu testing on macOS (or probably Unix and Windows as well:

- Create a folder called "Roster Test" on the Desktop.
- Create a New Profile and point User Files Location to "Roster Test".
- Quit JMRI.
- Delete the "Roster Test" folder.
- Create a file (any sort) named "Roster Test" on the Desktop.
- Create a folder called "untitled folder" on the Desktop.
- Restart JMRI.

You'll get into a "Preferences Save" tangle. If you try to quit you are likely to get an error like the screenshot below:
[image]
In this case you can resolve the issue by restoring the original "Roster Test" folder.

But in my Ubuntu case it had inherited a User Files Location of something like "/xxx/yyy/zzz" from the "profile.properties" created by the Mac node and in this case I couldn't easily recreate "/xxx/yyy/zzz" on my Ubuntu, so it's manual editing of "profile.properties" time.

dheap, 2018-04-17:
#5242 prevents recurrence of some of the issues documented above, but doesn't prevent the original situations.

dheap, 2018-05-29:
The problem is:
- When a new Node Identity is created, settings are inherited from profile.xml and profile.properties in the directory above (which are created when the profile is created).
- If an important file path such as User Files Location is bad for the new node (e.g. the profile was created by Windows and the new node is Mac, or the path is invalid because of a user name difference) an exception is thrown and JMRI hangs at the splash screen.
- The correct behaviour should be to catch the exception, log an error and an alert and allow startup to proceed, with User Files set to the usual default.
- Preferences should preferably then open at Locations so the user can set a new path. An overriding path entry is then created in the node-specific profile.properties file.

dheap, 2018-05-29:
Note that bad paths in startup items (profile.xml) are handled more gracefully, with an error message and dialog.

But in the case of a critical location like User Files, we need to reset to a valid default so startup can proceed.

I can't access a computer for some time to do testing, but it appears the invalid path in ../profile-dir/properties.xml is causing a null return.

dheap, 2018-05-29:
See <https://groups.io/g/jmriusers/topic/20400632>.

dheap, 2018-05-29:
I can easily reproduce the issue if I have two computers/partitions accessing a shared profile in Dropbox:
- Definitely in the case of a cross platform share.
- Probably in the case of sharing where the user names are different and not available on both computers.

dheap, 2018-05-29:
I'm wondering if the missing clue to reproducibility is that in the case of the machine the profile is created on, User Files Location may need to be set to somewhere outside both settings: and home: so the path stored in the ../profile-name/properties.proprties key:jmri-implementation.user-files is platform-specific raw format and cannot be converted to a portable path.

bobjacobsen, 2018-05-29:
Do you have an exception stack trace from one of the errors? I can take a look with that as a starting point.

dheap, 2020-05-27:
@rhwood This was the issue I couldn't find a couple of days ago in #8582 conversation.

dheap, 2021-03-30:
This is still unresolved and bites when setting up a new shared profile between platforms as the "valid on platform being created" file paths are used as the default for all new nodes added.

dheap, 2021-05-15:
Still needs addressing.