JMRI/JMRI #5185: nodeIdentity does not distinguish between macOS and Boot Camp partitions

dheap, 2018-04-01:
MacBook Pro with macOS and Boot Camp partitions. Both end up being assigned the same nodeIdentity as they both use the same hardware and hence MAC address.

This causes problems with Shared Profiles. It is not possible to share profiles between these two operating systems as they need node-specific settings. The same would apply to any partitioned multi-boot computer.

I tried manually editing the nodeIdentity for one in "nodeIdentity.xml" but it just resets on next JMRI startup.

This was discovered while researching #5184.

dheap, 2018-04-03:
Another thing I have discovered is that since I have two interfaces (EtherNet & WiFi) enabled (on the Mac boot at least), each with their own distinct MAC address) I end up with two possible node identities. Launches seem to move between using the two node identities, a further complication...

rhwood, 2018-04-03:
Unfortunately, the only value that we can use for hardware uniqueness in Java is the MAC address of the first found Ethernet device—Java provides no other means to accomplish this that is cross platform.

You will have to use multiple profiles with a shared user data space in this use case.

bobjacobsen, 2018-04-03:
Is the only important case here "running different OSs on the same hardware"?  If so, adding an OS term to the unique ID could work.  But decided what's the same vs different OS might be subtle:  I wouldn't want an in-place OS update to make JMRI insist that I'm now on a different machine.

pabender, 2018-04-03:
Finding a way to order the MAC addresses might help @dheap 's second problem.

I agree that adding the OS term might help, but there are use cases where that will fail too.  

I believe we will see the same issue if users are running two virtual machines on a single physical piece of hardware, and those virtual machines will likely have the same MAC address assigned.  There is nothing that prevents a virtual machine from running the same OS as the host.  (I have a virtual machine on one of my linux boxes that pretends to be a raspberry pi, running linux, for example).

rhwood, 2018-04-03:
The current implementation maintains a non-synced file (`settings:node-identies.xml` I think) that retains all MAC addresses previously used for Node identities on a specific host to allow it avoid @dheap's second issue, but something isn't paying attention to that.

We can't order the MAC addresses reliably; the first one ordered in Java by some OSes is the first one bound to an address by the OS (so Java may get the WiFi NIC's MAC address first once, but when the system is on CAT-5 Ethernet, it gets that NIC first), while in other OSes, it might be more deterministic.

dheap, 2018-04-03:
Thinking about this, `settings:node-identies.xml` is unique to every JMRI instance.

If we **once-off** generate a UUID (Version 1, I'd guess) and store that as a `<UUID>` element in `settings:node-identies.xml` we've got something that is unique to that JMRI instance and unaffected by MAC address, OS update, computer name, partition, etc. changes. There are schemes to shorten the UUID to around 20+ URL-safe characters for file naming.

The launch procedure would be to look for a `<UUID>` element in `settings:node-identies.xml` and use that for the nodeIdentity. If not present, generate a `<UUID>` element, store it and use the current nodeIdentity  or any `<formerIdentities>` as a basis for migration to the new UUID-based entity.

dheap, 2018-04-17:
This is resolved by #5242, but the issues in #5184 remain.

closed 2018-04-17 by dheap